esphome/issues #4956: Does not create an empty sketch on esp32

pafftis, 2023-10-07:
### The problem

I just installed esp home.
Created an empty scitch via wizard.
I pressed validation and no errors were found.
I clicked manual download and it gave an error in an empty sketch.

### Which version of ESPHome has the issue?

2023.09.03

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.2

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: test-esp
  friendly_name: Test_esp

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "f/X+3mu28hsHo+JEbacjjcRHeBnhQTZ/rf7w89vUD48="

ota:
  password: "2b50921b9fe45c1c98285a4ef63ce9af"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Esp Fallback Hotspot"
    password: "5egZ5fPFfvdS"

#captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/test-esp.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test-esp (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Removing AsyncTCP-esphome @ 2.0.1
INFO Removing AsyncTCP-esphome @ 2.0.1
Library Manager: AsyncTCP-esphome@2.0.1 has been removed!
INFO AsyncTCP-esphome@2.0.1 has been removed!
Library Manager: Removing ESPAsyncWebServer-esphome @ 3.1.0
INFO Removing ESPAsyncWebServer-esphome @ 3.1.0
Library Manager: Removing dependencies...
INFO Removing dependencies...
Library Manager: ESPAsyncWebServer-esphome@3.1.0 has been removed!
INFO ESPAsyncWebServer-esphome@3.1.0 has been removed!
Library Manager: Removing DNSServer @ 1.1.0
INFO Removing DNSServer @ 1.1.0
Library Manager: DNSServer@1.1.0 has been removed!
INFO DNSServer@1.1.0 has been removed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '/data/cache/platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File "/data/cache/platformio/platforms/espressif32/builder/frameworks/arduino.py", line 41, in <module>
Library Manager: Installing ESPmDNS
Warning! Could not find the package with 'ESPmDNS' requirements for your system 'linux_x86_64'
Library Manager: Installing Update
Warning! Could not find the package with 'Update' requirements for your system 'linux_x86_64'
Dependency Graph
|-- WiFi @ 1.2.7
|-- noise-c @ 0.1.4
Compiling .pioenvs/test-esp/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/test-esp/src/esphome/components/api/api_frame_helper.cpp.o
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:156:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:156:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/test-esp/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [.pioenvs/test-esp/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 7.88 seconds ==========================
```

pafftis, 2023-10-13:
Figured it out. The problem is that xp home cannot download large packages and if the speed drops, then it limits the jump time by 90 seconds, per package. And accordingly, each time it breaks off and swings anew. It would be nice if you implemented the possibility of setting the time, or the possibility of reloading the package.

rekcodocker, 2023-10-25:
I have the same error.
Home Assistant runs on a Pi4. I try to initialize an ESP32 device and I get a similar error. 
Compile failed, files not found.

No idea if it is because of a 'windows download'
It is missing files it needs for the compilation.

```

warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript '/data/cache/platformio/packages/framework-arduinoespressif32/tools/platformio-build.py'
File "/data/cache/platformio/platforms/espressif32/builder/frameworks/arduino.py", line 41, in <module>
Library Manager: Installing FS
Warning! Could not find the package with 'FS' requirements for your system 'linux_aarch64'
Library Manager: Installing Update
Warning! Could not find the package with 'Update' requirements for your system 'linux_aarch64'
Library Manager: Installing ESPmDNS
Warning! Could not find the package with 'ESPmDNS' requirements for your system 'linux_aarch64'
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 1.2.7
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/esp32-tessie1/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esp32-tessie1/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp32-tessie1/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/esp32-tessie1/src/esphome/components/api/api_pb2_service.cpp.o
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/socket/headers.h:156:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-tessie1/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/socket/socket.h:6,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/socket/headers.h:156:10: fatal error: sys/ioctl.h: No such file or directory
 #include <sys/ioctl.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-tessie1/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
src/esphome/core/log.h:14:10: fatal error: esp_err.h: No such file or directory

*****************************************************************
* Looking for esp_err.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

 #include <esp_err.h>
          ^~~~~~~~~~~
compilation terminated.
In file included from src/esphome/components/api/proto.h:4,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
src/esphome/core/log.h:14:10: fatal error: esp_err.h: No such file or directory

*****************************************************************
* Looking for esp_err.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:esp_err.h"
* Web  > https://registry.platformio.org/search?q=header:esp_err.h
*
*****************************************************************

 #include <esp_err.h>
          ^~~~~~~~~~~
compilation terminated.
*** [.pioenvs/esp32-tessie1/src/esphome/components/api/api_pb2.cpp.o] Error 1
*** [.pioenvs/esp32-tessie1/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
========================== [FAILED] Took 3.97 seconds ==========================

```

damien-o, 2024-01-06:
Reinstalling ESPHome solved the problem