esphome/issues #4957: E-Papr 4.2 Color does not update properly

pascal164, 2023-10-08:
### The problem

Hello,

I have a problem with an e-paper display.

The update takes a very long time and approx. 8 minutes even though the update rate is set to 2 minutes.

I also placed the pins correctly and also selected the 4.20in-bV2 display, which also updates normally, but here only a red screen is shown without text or anything else.

The functions "reset_duration" and "update_interval" have already been selected for various updates, without success.

Any Ideas ?

Facilities:

1. Waveshare E-Paper 4.2 Color
2. E32 Driver Board

YAML Code:

esphome:
  name: esphome-web-b32c78
  friendly_name: E-Paper Flur
esp32:
  board: esp32dev
  framework:
    type: arduino

    
captive_portal:

 Enable logging
logger:

 Enable Home Assistant API
api:
  encryption:
    key: "K6rxEwrQr466iWq+aTzAAU2YnjSrgICbAXa+PrxqaFY="

ota:
  password: "200wind200"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

   Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-B32C78"
    password: "raewIXth4CyR"

spi:
  clk_pin: 13
  mosi_pin: 14

color:
  - id: COLOR_RED
    red: 100%
    green: 0%
    blue: 0%
    white: 0%
  - id: COLOR_BLK
    red: 0%
    green: 0%
    blue: 0%
    white: 100%

font:

  - file: 'fonts/Roboto-Medium.ttf'
    id: roboto_med_30
    size: 30
  - file: 'fonts/Roboto-Regular.ttf'
    id: footer_font
    size: 15
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'ä', 'ö', '/','º','µ','³']

  - file: 'fonts/Roboto-Regular.ttf'
    id: main_sensor_unit
    size: 20
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'ä', 'ö', '/','º','µ','³']
       
  - file: 'fonts/Roboto-Medium.ttf'
    id: sub_sensor_font
    size: 30
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'ä', 'ö', '/','º','µ','³']
  
  - file: 'fonts/Roboto-Bold.ttf'
    id: main_sensor_font
    size: 72
  
    
display:
  - platform: waveshare_epaper
    cs_pin: 15
    dc_pin: 27
    busy_pin: 25
    reset_pin: 26
    model: 4.20in
    rotation: 180
    update_interval: 2min
    reset_duration: 200ms
    lambda: |-
      it.print(0, 0, id(footer_font), COLOR_RED ,  "HALLO World!");
       it.print(0, 50, id(footer_font),COLOR_BLK , "HALLO World1!");

`LOG Output
INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/esphome-web-b32c78.yaml...
WARNING GPIO15 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from esphome-web-b32c78.local using esphome API
INFO Successfully connected to esphome-web-b32c78.local
[13:26:37][I][app:102]: ESPHome version 2023.9.3 compiled on Oct  8 2023, 13:16:13
[13:26:37][C][wifi:546]: WiFi:
[13:26:37][C][wifi:382]:   Local MAC: C8:F1:9E:B6:2C:78
[13:26:37][C][wifi:383]:   SSID: [redacted]
[13:26:37][C][wifi:384]:   IP Address: 192.168.178.85
[13:26:37][C][wifi:386]:   BSSID: [redacted]
[13:26:37][C][wifi:387]:   Hostname: 'esphome-web-b32c78'
[13:26:37][C][wifi:389]:   Signal strength: -83 dB ▂▄▆█
[13:26:37][C][wifi:393]:   Channel: 5
[13:26:37][C][wifi:394]:   Subnet: 255.255.255.0
[13:26:37][C][wifi:395]:   Gateway: 192.168.178.1
[13:26:37][C][wifi:396]:   DNS1: 192.168.178.1
[13:26:37][C][wifi:397]:   DNS2: 0.0.0.0
[13:26:37][C][logger:357]: Logger:
[13:26:37][C][logger:358]:   Level: DEBUG
[13:26:37][C][logger:359]:   Log Baud Rate: 115200
[13:26:37][C][logger:361]:   Hardware UART: UART0
[13:26:37][C][spi:067]: SPI bus:
[13:26:37][C][spi:068]:   CLK Pin: GPIO13
[13:26:37][C][spi:069]:   SDI Pin: 
[13:26:37][C][spi:070]:   SDO Pin: GPIO14
[13:26:37][C][spi:072]:   Using HW SPI: SPI
[13:26:37][C][waveshare_epaper:1021]: Waveshare E-Paper
[13:26:37][C][waveshare_epaper:1021]:   Rotations: 180 °
[13:26:37][C][waveshare_epaper:1021]:   Dimensions: 400px x 300px
[13:26:37][C][waveshare_epaper:1022]:   Model: 4.2in
[13:26:37][C][waveshare_epaper:1023]:   Reset Pin: GPIO26
[13:26:37][C][waveshare_epaper:1024]:   DC Pin: GPIO27
[13:26:37][C][waveshare_epaper:1025]:   Busy Pin: GPIO25
[13:26:37][C][waveshare_epaper:1026]:   Update Interval: 120.0s
[13:26:37][C][captive_portal:088]: Captive Portal:
[13:26:37][C][mdns:115]: mDNS:
[13:26:37][C][mdns:116]:   Hostname: esphome-web-b32c78
[13:26:37][C][ota:097]: Over-The-Air Updates:
[13:26:37][C][ota:098]:   Address: esphome-web-b32c78.local:3232
[13:26:37][C][ota:101]:   Using Password.
[13:26:37][C][api:138]: API Server:
[13:26:37][C][api:139]:   Address: esphome-web-b32c78.local:6053
[13:26:37][C][api:141]:   Using noise encryption: YES
**[13:30:45][W][component:204]: Component waveshare_epaper.display took a long time for an operation (0.14 s).
[13:30:45][W][component:205]: Components should block for at most 20-30ms.**

Greetings 

Pascal 
`

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.9.3

### What platform are you using?

ESP32

### Board

ESP 32 Driver Board

### Component causing the issue

E-Paper 4.2

SO5HMI, 2024-02-02:
Do you resolve this problem? I have the same issue

pascal164, 2024-02-06:
> Do you resolve this problem? I have the same issue

Hello,

No, unfortunately not. I then stopped the project

lauraled, 2024-04-30:
Hello, I have the same issue as well, but with the b/w version of the 4.2 inch display. I can post my config as well if necessary. I tried with the latest ESPHome version. Maybe someone has a clue regarding this issue?