esphome/issues #4959: Another one xtensa-lx106-elf-g++: not found

andiaa734, 2023-10-08:
### The problem

Upgraded to ESPHome 2023.9.3

Host is RPi4, 
uname -a: Linux hostname 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

/boot/config.txt
[all]
arm_64bit=1

Switiching to arm_64bit=0 HA will no longer start

Home Assistant 2023.10.1
Supervisor 2023.10.0
Frontend 20231005.0 - latest

Full output from compilation:

INFO ESPHome 2023.9.3
INFO Reading configuration /config/esphome/myDevice.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing myDevice (board: esp12e; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/myDevice /src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/myDevice /src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/myDevice/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
*** [.pioenvs/myDevice/src/esphome/components/api/api_connection.cpp.o] Error 127
Compiling .pioenvs/myDevice/src/esphome/components/api/api_pb2_service.cpp.o
*** [.pioenvs/myDevice/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
*** [.pioenvs/myDevice/src/esphome/components/api/api_pb2.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [.pioenvs/myDevice/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
========================== [FAILED] Took 3.30 seconds ==========================

Read almost all related issues to this error message. Some issues are for older versions of esphome and the possible solutions did not work for me.
So, any other suggestions?

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP8266

### Board

ESP12-F

### Component causing the issue

api

### Example YAML snippet

```yaml
substitutions:
  device_name: "ambilight1"

  
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp12e

wifi:
  ssid: "myawesomewifi"
  password: "mytopsecretpassword"
  domain: .fritz.box
  
#Enable logging
logger:

#Enable Home Assistant API
api:

ota:

output:

  - platform: esp8266_pwm
    pin: GPIO5
    id: red_gpio

    
  - platform: esp8266_pwm
    pin: GPIO16
    id: blue_gpio

  - platform: esp8266_pwm
    pin: GPIO14
    id: green_gpio

  - platform: esp8266_pwm
    pin: GPIO12
    id: warm_white_gpio

  - platform: esp8266_pwm
    pin: GPIO4
    id: cold_white_gpio

light:

  - platform: rgbcct
    name: "${device_name}"
    id: ambilight
    red: red_gpio
    green: green_gpio
    blue: blue_gpio
    warm_white: warm_white_gpio
    cold_white: cold_white_gpio
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2400 K
    # In Kelvin or Mireds
    max_warm_color_temperature: 370 mireds  # Optional defaults to 500 mireds
    max_cold_color_temperature: 153 mireds  # Optional defaults to 153 mireds
    constant_brightness: false  # Optional defaults to false, make white brightness levels combine to max_combined_white_level
    max_combined_white_level: 1  # Optional defaults to 1, reduce if light is unable to support max brightness

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "${device_name}"

web_server:
  port: 80
```

GRiDlockUK, 2023-10-20:
Same issue - cannot update any of my devices

andiaa734, 2023-10-21:
> Same issue - cannot update any of my devices

Update 2023.10.1 solved the problem for me.

GRiDlockUK, 2023-10-24:
Not for me - even 2023.10.3 update did not fix

```
INFO ESPHome 2023.10.3
INFO Reading configuration /config/study-clock.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing study-clock (board: d1_mini; framework: arduino; platform: platformio/espressif8266@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- SPI @ 1.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/study-clock/src/esphome/components/adc/adc_sensor.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/study-clock/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/study-clock/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-lx106-elf-g++: not found
Compiling .pioenvs/study-clock/src/esphome/components/api/api_pb2.cpp.o
*** [.pioenvs/study-clock/src/esphome/components/adc/adc_sensor.cpp.o] Error 127
*** [.pioenvs/study-clock/src/esphome/components/api/api_connection.cpp.o] Error 127
*** [.pioenvs/study-clock/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
sh: 1: xtensa-lx106-elf-g++: not found
*** [.pioenvs/study-clock/src/esphome/components/api/api_pb2.cpp.o] Error 127
========================== [FAILED] Took 3.52 seconds ==========================
```

andiaa734, 2023-10-24:
Have you tried to remove the addon first and then install the latest Version?

stuartf, 2023-10-31:
I'm getting this as well with 2023.10.4, not using the addon, just a fresh docker image with only my yaml files and /etc/localtime added to it:

    INFO ESPHome 2023.10.4
    INFO Reading configuration /config/motion-env.yaml...
    INFO Generating C++ source...
    INFO Compiling app...
    Processing motion-env (board: nodemcu; framework: arduino; platform: platformio/espressif8266@3.2.0)
    --------------------------------------------------------------------------------
    Platform Manager: Installing platformio/espressif8266 @ 3.2.0
    INFO Installing platformio/espressif8266 @ 3.2.0
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Platform Manager: espressif8266@3.2.0 has been installed!
    INFO espressif8266@3.2.0 has been installed!
    Tool Manager: Installing platformio/toolchain-xtensa @ ~2.100300.0
    INFO Installing platformio/toolchain-xtensa @ ~2.100300.0
    Downloading  [####################################]  100%          
    Unpacking  [####################################]  100%          
    Tool Manager: toolchain-xtensa@2.100300.220621 has been installed!
    INFO toolchain-xtensa@2.100300.220621 has been installed!
    Tool Manager: Installing platformio/framework-arduinoespressif8266 @ ~3.30002.0
    INFO Installing platformio/framework-arduinoespressif8266 @ ~3.30002.0
    Downloading  [####################################]  100%          
    Unpacking  [####################################]  100%          
    Tool Manager: framework-arduinoespressif8266@3.30002.0 has been installed!
    INFO framework-arduinoespressif8266@3.30002.0 has been installed!
    Tool Manager: Installing platformio/tool-esptool @ <2
    INFO Installing platformio/tool-esptool @ <2
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Tool Manager: tool-esptool@1.413.0 has been installed!
    INFO tool-esptool@1.413.0 has been installed!
    Tool Manager: Installing platformio/tool-esptoolpy @ ~1.30000.0
    INFO Installing platformio/tool-esptoolpy @ ~1.30000.0
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Tool Manager: tool-esptoolpy@1.30000.201119 has been installed!
    INFO tool-esptoolpy@1.30000.201119 has been installed!
    Tool Manager: Installing platformio/tool-scons @ ~4.40502.0
    INFO Installing platformio/tool-scons @ ~4.40502.0
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Tool Manager: tool-scons@4.40502.0 has been installed!
    INFO tool-scons@4.40502.0 has been installed!
    Library Manager: Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
    INFO Installing esphome/ESPAsyncTCP-esphome @ 2.0.0
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Library Manager: ESPAsyncTCP-esphome@2.0.0 has been installed!
    INFO ESPAsyncTCP-esphome@2.0.0 has been installed!
    Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
    INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.1.0
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Library Manager: ESPAsyncWebServer-esphome@3.1.0 has been installed!
    INFO ESPAsyncWebServer-esphome@3.1.0 has been installed!
    Library Manager: Resolving dependencies...
    INFO Resolving dependencies...
    Library Manager: Installing ottowinter/ESPAsyncTCP-esphome
    INFO Installing ottowinter/ESPAsyncTCP-esphome
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Library Manager: ESPAsyncTCP-esphome@1.2.3 has been installed!
    INFO ESPAsyncTCP-esphome@1.2.3 has been installed!
    Library Manager: Installing esphome/noise-c @ 0.1.4
    INFO Installing esphome/noise-c @ 0.1.4
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Library Manager: noise-c@0.1.4 has been installed!
    INFO noise-c@0.1.4 has been installed!
    Library Manager: Resolving dependencies...
    INFO Resolving dependencies...
    Library Manager: Installing esphome/libsodium @ 1.10018.1
    INFO Installing esphome/libsodium @ 1.10018.1
    Downloading  [####################################]  100%
    Unpacking  [####################################]  100%
    Library Manager: libsodium@1.10018.1 has been installed!
    INFO libsodium@1.10018.1 has been installed!
    HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
    Dependency Graph
    |-- ESPAsyncTCP-esphome @ 2.0.0
    |-- ESPAsyncWebServer-esphome @ 3.1.0
    |-- DNSServer @ 1.1.1
    |-- ESP8266WiFi @ 1.0
    |-- ESP8266mDNS @ 1.2
    |-- noise-c @ 0.1.4
    |-- Wire @ 1.0
    Compiling .pioenvs/motion-env/src/esphome/components/api/api_connection.cpp.o
    sh: 1: xtensa-lx106-elf-g++: not found
    Compiling .pioenvs/motion-env/src/esphome/components/api/api_frame_helper.cpp.o
    sh: 1: xtensa-lx106-elf-g++: not found
    Compiling .pioenvs/motion-env/src/esphome/components/api/api_pb2.cpp.o
    sh: 1: xtensa-lx106-elf-g++: not found
    *** [.pioenvs/motion-env/src/esphome/components/api/api_connection.cpp.o] Error 127
    Compiling .pioenvs/motion-env/src/esphome/components/api/api_pb2_service.cpp.o
    *** [.pioenvs/motion-env/src/esphome/components/api/api_frame_helper.cpp.o] Error 127
    *** [.pioenvs/motion-env/src/esphome/components/api/api_pb2.cpp.o] Error 127
    sh: 1: xtensa-lx106-elf-g++: not found
    *** [.pioenvs/motion-env/src/esphome/components/api/api_pb2_service.cpp.o] Error 127
    ========================= [FAILED] Took 83.64 seconds =========================

ssieb, 2023-10-31:
@stuartf can you come to discord so you can help us investigate what's happening?

sbland, 2024-01-25:
I'm having this issue running esphome via docker on RPi5 64 aarch64

stuartf, 2024-02-01:
@sbland you may want to take a look at https://github.com/esphome/issues/issues/3092

There are workarounds posted there, eventually this should be fixed by a new version of platformio

sbland, 2024-02-01:
Thanks @stuartf in the end I just used a different machine to build and push to my esp device. I'll take a look at #3092 if I get chance.