esphome/issues #4962: can't establish mqtt connection with broker

EuPhobos, 2023-10-09:
### The problem

tls is disabled on the broker, but I get such a log from my esp32
`[I][mqtt:242]: Connecting to MQTT...
[D][esp-idf:000]: E (485482) esp-tls: [sock=49] select() timeout

[D][esp-idf:000]: E (485484) TRANSPORT_BASE: Failed to open a new connection: 32774

[D][esp-idf:000]: E (485494) MQTT_CLIENT: Error transport connect

[E][mqtt.idf:150]: MQTT_EVENT_ERROR
[E][mqtt.idf:152]: Last error code reported from esp-tls: 0x8006
[E][mqtt.idf:153]: Last tls stack error number: 0x0
[E][mqtt.idf:155]: Last captured errno : 0 (Success)
[W][mqtt:331]: MQTT Disconnected: TCP disconnected.`
I don't see a way to disable tls

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP32

### Board

esp wroom 32

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
mqtt:
  broker: 172.17.0.7
  port: 8883
  username: !secret mqtt-user
  password: !secret mqtt-password

esphome:
  name: esp32-multimobile
  friendly_name: esp32-multimobile

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:

api:
  encryption:
    key: "o442IliAqK3c0sInSkSpmd3jjHla6BXWVa08lTu8czg="

ota:
  password: "4e8ccac9f394be9013bdbbce26253046"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "Esp32-Multimobile"
    password: "fJR21C29xOGQ"
  manual_ip:
    static_ip: 172.17.100.121
    gateway: 172.17.100.100
    subnet: 255.255.255.0

captive_portal:

i2c:
  sda: 5
  scl: 4
  frequency: 800kHz

font:
  - file: "gfonts://Roboto"
    id: roboto
    size: 10

text_sensor:
  - platform: mqtt_subscribe
    name: "Data from topic"
    id: mysensor
    topic: notify/ha_128x64

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    lambda: |-
      it.printf(0, 0, id(roboto), "Data: %s", id(mysensor).state.c_str());
```

### Anything in the logs that might be useful for us?

```txt
[I][mqtt:242]: Connecting to MQTT...
[D][esp-idf:000]: E (485482) esp-tls: [sock=49] select() timeout

[D][esp-idf:000]: E (485484) TRANSPORT_BASE: Failed to open a new connection: 32774

[D][esp-idf:000]: E (485494) MQTT_CLIENT: Error transport connect

[E][mqtt.idf:150]: MQTT_EVENT_ERROR
[E][mqtt.idf:152]: Last error code reported from esp-tls: 0x8006
[E][mqtt.idf:153]: Last tls stack error number: 0x0
[E][mqtt.idf:155]: Last captured errno : 0 (Success)
[W][mqtt:331]: MQTT Disconnected: TCP disconnected.
```

ssieb, 2023-10-09:
Why are you using the mqtt TLS port for non-TLS connections?

EuPhobos, 2023-10-09:
Oh, it's my mistake that I didn't look and inserted the edited config HERE, in fact I use 1883, and the logs with errors are still shown.
I tested this later with different ports, because I can't avoid this error.

EuPhobos, 2023-10-13:
i think i use wrong platform and board

closed 2023-10-13 by EuPhobos

sedlacekdavid, 2023-10-25:
I have the same problem and don't know how to solve. I don't want TLS connection. Is there any way to disable it?

[image]

EuPhobos, 2023-10-25:
@sedlacekdavid i just change board and platform
`esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf`
but in my case ESPHome fail to compile esp-idf, this is another problem i meet.