esphome/issues #4971: Compilation failed when using Rtttl as a dependency

kvj, 2023-10-11:
### The problem

I'm working on a custom component, and I'm getting error while using `rtttl` component as a dependency. All other components I'm using do not cause any compilation errors.

Details:

I declare schema as follows:

```python
from esphome.components import ..., rtttl

CONFIG_SCHEMA = (
    cv.Schema({
        cv.GenerateID(): cv.declare_id(_cls),
        cv.GenerateID(CONF_DISPLAY): cv.use_id(nextion.Nextion),
        # ...
        cv.GenerateID(CONF_RTTTL): cv.use_id(rtttl.Rtttl),
    })
    .extend(cv.COMPONENT_SCHEMA)
)
```
then I'm able to get the reference via
```python
await cg.get_variable(config[CONF_RTTTL])
```

but when I'm trying to declare a field or make a method in the c++ code like
```c
esphome::rtttl::Rtttl *rtttl_player;
```
I'm getting the compile error
```
src/esphome/components/nspanel_ng/nspanel_ng.h:18:18: error: 'rtttl' in namespace 'esphome' does not name a type                                                                       
         esphome::rtttl::Rtttl *rtttl_player;                         

```

I'm declaring a bunch of other references e.g. display, switches and everything works as expected.
I've checked generated `main.cpp` and `esphome.h` files they look okay, `rtttl.h` is included (as the necessary component is in yaml). The yaml looks like this:

```yaml

rtttl:
  id: buzzer
  output: buzzer_out

output:
  - platform: ledc
    id: buzzer_out
    pin:
      number: 21
```

I've ruled out most obvious reasons why the namespace isn't visible, and I'm running out of ideas

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

Not relevant

### What platform are you using?

ESP32

### Board

esp32dev / Sonoff NSPanel

### Component causing the issue

rtttl

nielsnl68, 2023-10-11:
This question can be  better asked on discord.

ssieb, 2023-10-11:
Right, issues with custom or external components aren't for github unless it's caused by an actual bug in esphome.

closed 2023-10-11 by ssieb

kvj, 2023-10-11:
> Right, issues with custom or external components aren't for github unless it's caused by an actual bug in esphome.

Hard to say :)
Could be a bug is esphome, could be a bug in internal `rtttl` component