esphome/issues #4974: SML component hangs / reboots  with ESP8266

MichaBub, 2023-10-13:
### The problem

There are no stability issues with ESP32, but with an ESP8266 SML component hangs / freezes / reboots

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.2

### What platform are you using?

ESP8266

### Board

nodemcuv2 with chipset ESP8266MOD

### Component causing the issue

SML

### Example YAML snippet

```yaml
esphome:
  name:  ${device_name}
  friendly_name: ${device_name_readable}
  platform: esp8266
  board: nodemcuv2  

uart:
  id: uart_bus
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

sml:
  id: mysml
  uart_id: uart_bus

sensor:
  - platform: sml
    name: "Total Energy Consumed"
    sml_id: mysml
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
#      - throttle: 20s
  - platform: sml
    name: "Total Energy Delivered"
    sml_id: mysml
    obis_code: "1-0:2.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 3
    device_class: energy
    state_class: total_increasing
[... 251 lines cut ...]
[09:08:09][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:32.7.0 [0x0000000000000942]
[09:08:09][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:52.7.0 [0x0000000000000953]
[09:08:09][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:72.7.0 [0x0000000000000941]
[09:08:09][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:31.7.0 [0x0000000000000073]
[09:08:09][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:51.7.0 [0x000000000000008a]
[09:08:09][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:71.7.0 [0x000000000000010d]
[09:08:10][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:81.7.1 [0x00000000000004b0]
[09:08:10][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:81.7.2 [0x0000000000000967]
[09:08:10][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:81.7.4 [0x0000000000000815]
[09:08:10][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:81.7.15 [0x0000000000000777]
[09:08:10][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:81.7.26 [0x000000000000078b]
[09:08:10][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:14.7.0 [0x00000000000001f3]
[09:08:10][D][sml:072]:   (0a014c4f4700057ddaec) 1-0:0.2.0 [0x3036]
[09:08:10][W][component:204]: Component sml took a long time for an operation (2.07 s).
[09:08:10][W][component:205]: Components should block for at most 20-30ms.
[09:08:12][D][api:102]: Accepted 192.168.1.10
[09:08:12][W][component:204]: Component api took a long time for an operation (0.30 s).
[09:08:12][W][component:205]: Components should block for at most 20-30ms.
```

### Additional information

It seems that the SML component / ESP8266 gets stuck after HomeAssitant connects to ESP
in my case it is e.g.
[09:08:12][D][api:102]: Accepted 192.168.1.10

Skeletitor, 2024-04-08:
@MichaBub Did you find a solution for this? My 8266 hangs as soon as I connect the RX diode

MichaBub, 2024-04-09:
Hey @Skeletitor - yes I found a solution. I did change to a ESP32, than everything works smooth. ESP8266 did not work with SML component for me.

Skeletitor, 2024-04-18:
@MichaBub This trick worked for mee too.

So, my good old D1mini (esp8266) is now retired :)

unaiur, 2025-01-19:
SML component generates too much memory pressure and is not designed for devices with scarce RAM like ESP8266. Currently, it converts each SML message into a syntactic tree, with hundreds of memory allocations per second, and ESP8266 does not have enough RAM memory. 

I have contributed a PR to reduce memory consumed by SML to less than half: #8081. It would be great if somebody can test it with an ESP8266. I do not have any.

unaiur, 2025-02-21:
I have optimized  PR#8081 above to reduce even more memory requirements. Instead of deserializing the SML binary message in a syntax tree that requires ~5.4KB and I optimized previously to consume less than 3KB, now I am extracting obis counters directly from SML binary message, using dynamic memory only for binary message storage (~300 bytes using my electric meter).

Can you test it again? Now it should work fine.

zivillian, 2025-02-22:
I can confirm, that the current PR allows me to enable MQTT, the WebUI and view the logs in the dashboard via esphome API all at the same time.
The component still complains about taking more than 30ms, but I'm totally fine with this

unaiur, 2025-04-13:
Yes, of course. You can download the code from my repo and compile it. The compilation process is exactly the same.

Since nobody with merge privileges is interested in reviewing this code, I will convert into an external component. This way, it will be more easily available.