esphome/issues #4976: Raspi Audio Muse Proto Speaker component not working

s00500, 2023-10-13:
### The problem

Hey all, year of the voice part 4 is super cool!

I could not resist updating to latest beta yesterday and testing this on my muse proto

And it does work nicely, but I go t no sound from the speaker using the config found here:

https://github.com/esphome/firmware/blob/main/voice-assistant/raspiaudio-muse-proto.yaml

So I changed the config to use media_player component instead, this has worked for me in the past, and again it did the trick, this way I get audio out working fine. BUT using media_player requires me to use framework:arduino and therefore I had to disable the IDFs VAD function.

Does anyone have an Idea why the speaker does only work when used via mediaplayer ?

To be clear: I am mainly reporting that the config in the esphome repo does NOT have working sound output on the Muse Proto. The added example yaml is my modified config that works fine (without VAD)

### Which version of ESPHome has the issue?

2023.10.0b1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.2

### What platform are you using?

ESP32

### Board

RaspiAudio Muse Proto

### Component causing the issue

speaker

### Example YAML snippet

```yaml
---
esphome:
  name: muse1
  friendly_name: Muse1
  project:
    name: raspiaudio.muse-proto-voice-assistant
    version: "1.0"
  min_version: 2023.10.0b1
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

logger:

api:
  encryption:
    key: 

ota:
  password: 

#external_components:
#  - source: github://pr#5230
#    components:
#      - esp_adf
#    refresh: 0s

i2s_audio:
  - i2s_lrclk_pin: GPIO25
    i2s_bclk_pin: GPIO5

microphone:
  - platform: i2s_audio
    id: board_microphone
    channel: left
    i2s_din_pin: GPIO35
    adc_type: external
    pdm: false

# speaker:
#   - platform: i2s_audio
#     id: board_speaker
#     dac_type: external
#     i2s_dout_pin: GPIO26
#     mode: mono

#esp_adf:

media_player:
  - platform: i2s_audio
    id: media_out
    name: Muse1
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: mono
    mute_pin:
      number: GPIO21
      inverted: true

voice_assistant:
  id: va
  microphone: board_microphone
  # speaker: board_speaker
  use_wake_word: true
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  # vad_threshold: 3
  on_listening:
    - media_player.toggle: media_out
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: pulse
  on_tts_start:
    - light.turn_on:
        id: led
        blue: 0%
        red: 0%
        green: 100%
        brightness: 100%
        effect: pulse
  on_tts_end:
    - media_player.play_media: !lambda return x;
  on_end:
    - delay: 100ms
    # - wait_until:
    #     not:
    #       speaker.is_playing:
    - script.execute: reset_led
  on_error:
    - light.turn_on:
        id: led
        blue: 0%
        red: 100%
        green: 0%
        brightness: 100%
        effect: none
    - delay: 1s
    - script.execute: reset_led
    - script.wait: reset_led
    - lambda: |-
        if (code == "wake-provider-missing" || code == "wake-engine-missing") {
          id(use_wake_word).turn_off();
        }

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      inverted: true
      mode:
        input: true
        pullup: true
    name: Action
    disabled_by_default: true
    on_click:
      - if:
          condition:
            switch.is_off: use_wake_word
          then:
            - if:
                condition: voice_assistant.is_running
                then:
                  - voice_assistant.stop:
                  - script.execute: reset_led
                else:
                  - voice_assistant.start:
          else:
            - voice_assistant.stop
            - delay: 1s
            - script.execute: reset_led
            - script.wait: reset_led
            - voice_assistant.start_continuous:

light:
  - platform: esp32_rmt_led_strip
    rmt_channel: 0
    name: None
    id: led
    disabled_by_default: true
    pin: GPIO22
    chipset: SK6812
    num_leds: 1
    rgb_order: grb
    effects:
      - pulse:
          transition_length: 250ms
          update_interval: 250ms

script:
  - id: reset_led
    then:
      - if:
          condition:
            switch.is_on: use_wake_word
          then:
            - light.turn_on:
                id: led
                blue: 100%
                red: 100%
                green: 0%
                brightness: 100%
                effect: none
          else:
            - light.turn_off: led

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(va).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
      - script.execute: reset_led
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);
      - script.execute: reset_led
```

s00500, 2023-10-13:
Ok, I actually found the issue, it is the mute pin on GPIO21 that is not set correctly
This works really nice when adding the mute_pin to the config =D

Opened a PR :-)

closed 2023-10-13 by jesserockz

spudje, 2023-10-28:
Sorry, github/release pipeline n00b question. Is this already in 2023.10.3? If so, it still doesn't work for me. I don't hear voice assistant responses.

s00500, 2023-10-30:
(Also there are still open issues here https://github.com/esphome/issues/issues/4985 )

spudje, 2023-10-30:
> @spudje did you update your configuration from master ? or how did you go about testing this ? it is only a config change

OK, let me retry. I assumed it was in the ESPHome package and not just a config change. I'll try the config in this thread.