esphome/issues #4978: DSMR: Errors due to non-standard ID string

Evil-Crab, 2023-10-13:
### The problem

Dutch district heat meter Warmtelink uses a non-standard ID (/NWA-WARMTELINK) in its DSMR messages which breaks DSMR parser.

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10

### What platform are you using?

ESP8266

### Board

d1_mini

### Component causing the issue

dsmr

### Anything in the logs that might be useful for us?

```txt
[19:08:16][E][dsmr:265]: /NWA-WARMTELINK
 ^
Invalid identification string
[19:08:27][E][dsmr:265]: /NWA-WARMTELINK
 ^
Invalid identification string
```

### Additional information

Warmtelink dump:
/NWA-WARMTELINK
 
1-3:0.2.8(50)
0-0:1.0.0(221003002940W)
0-0:96.1.1(ADC3100000064761)
0-0:96.13.1()
0-0:96.13.0()
0-1:24.1.0(004)
0-1:96.1.0(719737592D2C340C)
0-1:24.2.1(221003002940W)(55.952*GJ)
!2388
 
/NWA-WARMTELINK
 
1-3:0.2.8(50)
0-0:1.0.0(221003002951W)
0-0:96.1.1(ADC3100000064761)
0-0:96.13.1()
0-0:96.13.0()
0-1:24.1.0(004)
0-1:96.1.0(719737592D2C340C)
0-1:24.2.1(221003002951W)(55.952*GJ)
!49BD
 
/NWA-WARMTELINK
 
1-3:0.2.8(50)
0-0:1.0.0(221003003002W)
0-0:96.1.1(ADC3100000064761)
0-0:96.13.1()
0-0:96.13.0()
0-1:24.1.0(004)
0-1:96.1.0(719737592D2C340C)
0-1:24.2.1(221003003002W)(55.952*GJ)
!B4C8
 
/NWA-WARMTELINK
 
1-3:0.2.8(50)
0-0:1.0.0(221003003013W)
0-0:96.1.1(ADC3100000064761)
0-0:96.13.1()
0-0:96.13.0()
0-1:24.1.0(004)
0-1:96.1.0(719737592D2C340C)
0-1:24.2.1(221003003013W)(55.952*GJ)
!DEFD
 
/NWA-WARMTELINK
 
1-3:0.2.8(50)
0-0:1.0.0(221003003024W)
0-0:96.1.1(ADC3100000064761)
0-0:96.13.1()
0-0:96.13.0()
0-1:24.1.0(004)
0-1:96.1.0(719737592D2C340C)
0-1:24.2.1(221003003024W)(55.952*GJ)
!9386

Evil-Crab, 2023-10-13:
@glmnet it seems there was MR addressing this issue that was closed without merging https://github.com/glmnet/arduino-dsmr/pull/10

jeroen85, 2023-10-29:
It will also need some further modifications to the DSMR component. It should define the `thermal_mbus_id` and process the `thermal_delivered` value.
I have prepared some (small) changes to the component and will get my Warmtelink meter tomorrow, allowing me to test the changes :-)

Evil-Crab, 2023-12-04:
@glmnet It's also fixed in the original version of the parser https://github.com/matthijskooijman/arduino-dsmr/commit/ad2393756ed11e23ab44327c51111dfb55ff38fc

zaregg, 2024-09-29:
Is there still a possibility that it gets added to the dsmr component?