esphome/issues #4982: Prometheus component fails to compile with esp-idf

dokutan, 2023-10-14:
### The problem

Compiling the prometheus component fails when using esp-idf:
```
Compiling .pioenvs/test/cmock/CMock/src/cmock.o
src/main.cpp:18:1: error: 'prometheus' does not name a type
 prometheus::PrometheusHandler *prometheus_prometheushandler;
 ^~~~~~~~~~
src/main.cpp: In function 'void setup()':
src/main.cpp:138:3: error: 'prometheus_prometheushandler' was not declared in this scope
   prometheus_prometheushandler = new prometheus::PrometheusHandler(web_server_base_webserverbase);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/main.cpp:138:38: error: 'prometheus' does not name a type
   prometheus_prometheushandler = new prometheus::PrometheusHandler(web_server_base_webserverbase);
                                      ^~~~~~~~~~
```

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP32-IDF

### Board

esp32dev

### Component causing the issue

prometheus

### Example YAML snippet

```yaml
esphome:
  name: test

esp32:
  board: esp32dev
  framework:
    type: esp-idf

logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

network:
  enable_ipv6: true

web_server:
  port: 80
  local: true

prometheus:

sensor:
  - platform: uptime
    name: Uptime Sensor
```

jesserockz, 2023-10-15:
This is from leftover code before web_server was compatible with esp-idf.

Please try the above linked PR to test my changes.

```yaml
external_components:
  - source: github://pr#5536
    components: prometheus
    refresh: 0s
```

dokutan, 2023-10-15:
Thanks, that PR fixes the issue for me.

jesserockz, 2023-10-15:
There is definitely a newline printed there in code.

https://github.com/esphome/esphome/blob/3daa2f22a60311db2b77f683a76a102ea0307463/esphome/components/prometheus/prometheus_handler.cpp#L89-L90

jesserockz, 2023-10-15:
I have pushed another commit to my linked PR, please try install again.

closed 2023-10-16 by jesserockz

dokutan, 2023-10-16:
I can confirm that it works now.