esphome/issues #4984: Servo component seems to ignore transition length

tuct, 2023-10-15:
### The problem

When using the servo component, it seems the transition length is ignored / not working.
Regardless of what I set, I see in the logs the reached a new position in the same sec as it was set.
Here is a link to similar reportings:
https://community.home-assistant.io/t/servo-transition-length-2s-not-working/509970/4

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

none

### What platform are you using?

ESP32

### Board

esp32-devkit4

### Component causing the issue

servo

### Example YAML snippet

```yaml
number:
  - platform: template
    id: servo_pos
    min_value: -100
    initial_value: 0
    max_value: 100
    step: 10
    optimistic: true
    set_action:
      then:
        - servo.write:
            id: lid_servo
            level: !lambda 'return x / 100.0;'
        
servo:
  - id: lid_servo
    output: pwm_servo
    transition_length: 50s
    auto_detach_time: 1s
```

### Anything in the logs that might be useful for us?

```txt
https://esphome.io/api/servo_8cpp_source   
Those lines seem to be reached without delay
   if (this->target_value_ == this->current_value_ && this->state_ == STATE_ATTACHED) {
     this->state_ = STATE_TARGET_REACHED;
     this->start_millis_ = millis();  // set current stamp for potential auto_detach_time_ check
     ESP_LOGD(TAG, "Servo reached target");
   }
So i expect something wrong here:
   if (this->target_value_ != this->current_value_ && this->state_ == STATE_ATTACHED) {
     if (this->transition_length_) {
       float new_value;
       float travel_diff = this->target_value_ - this->source_value_;
       uint32_t target_runtime = abs((int) ((travel_diff) * this->transition_length_ * 1.0f / 2.0f));
       uint32_t current_runtime = millis() - this->start_millis_;
       float percentage_run = current_runtime * 1.0f / target_runtime * 1.0f;
       if (percentage_run > 1.0f) {
         percentage_run = 1.0f;
       }
       new_value = this->target_value_ - (1.0f - percentage_run) * (this->target_value_ - this->source_value_);
       this->internal_write(new_value);
     } else {
       this->internal_write(this->target_value_);
     }
   }
```

tuct, 2023-10-15:
[15:40:14][D][switch:016]: 'heater_relais' Turning OFF.
**[15:40:14][D][servo:062]: Servo new target: 0.000000**
[15:40:14][D][light:036]: 'Prusa Enclosure - Fan' Setting:
[15:40:14][D][light:047]:   State: OFF
[15:40:14][D][sensor:094]: 'Humidity': Sending state 34.02000 % with 0 decimals of accuracy
[15:40:14][D][sensor:094]: 'VOC': Sending state 81.00000  with 0 decimals of accuracy
[15:40:14][W][component:204]: Component sen5x.sensor took a long time for an operation (0.13 s).
[15:40:14][W][component:205]: Components should block for at most 20-30ms.
**[15:40:14][D][servo:050]: Servo reached target**
[15:40:14][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:14][W][component:205]: Components should block for at most 20-30ms.
[15:40:14][D][sensor:094]: 'auto_filter_sensor': Sending state 0.00000  with 1 decimals of accuracy
[15:40:15][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:15][W][component:205]: Components should block for at most 20-30ms.
[15:40:15][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:15][W][component:205]: Components should block for at most 20-30ms.
[15:40:15][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:15][W][component:205]: Components should block for at most 20-30ms.
**[15:40:15][D][servo:028]: Servo detached on auto_detach_time**
[15:40:15][D][sensor:094]: 'auto_filter_sensor': Sending state 0.00000  with 1 decimals of accuracy
[15:40:16][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:16][W][component:205]: Components should block for at most 20-30ms.
[15:40:16][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:16][W][component:205]: Components should block for at most 20-30ms.
[15:40:16][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:16][W][component:205]: Components should block for at most 20-30ms.
[15:40:16][D][sensor:094]: 'auto_filter_sensor': Sending state 0.00000  with 1 decimals of accuracy
[15:40:16][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:16][W][component:205]: Components should block for at most 20-30ms.
[15:40:17][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:17][W][component:205]: Components should block for at most 20-30ms.
[15:40:17][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:17][W][component:205]: Components should block for at most 20-30ms.
[15:40:17][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:17][W][component:205]: Components should block for at most 20-30ms.
[15:40:17][D][sensor:094]: 'auto_filter_sensor': Sending state 0.00000  with 1 decimals of accuracy
[15:40:18][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).
[15:40:18][W][component:205]: Components should block for at most 20-30ms.
[15:40:18][D][main:1113]: COOOOOOOOOOOOL NOW
[15:40:18][I][main:226]: AC Active: true
[15:40:18][I][main:227]: Fan switch: true
[15:40:18][I][main:228]: AC Mode: 1
[15:40:18][I][main:229]: AC Action: 2
[15:40:18][I][main:256]: AC: ON, HEATER: OFF, FAN: ON, LID: open!
**[15:40:18][D][servo:062]: Servo new target: 0.500000**
[15:40:18][D][switch:016]: 'heater_relais' Turning OFF.
[15:40:18][D][light:036]: 'Prusa Enclosure - Fan' Setting:
[15:40:18][D][light:047]:   State: ON
[15:40:18][D][light:051]:   Brightness: 100%
[15:40:18][D][climate:378]: 'Enclosure Climate Controller' - Sending state:
[15:40:18][D][climate:381]:   Mode: HEAT_COOL
[15:40:18][D][climate:383]:   Action: COOLING
[15:40:18][D][climate:401]:   Current Temperature: 25.59°C
[15:40:18][D][climate:405]:   Target Temperature: Low: 18.00°C High: 19.00°C
[15:40:18][D][main:1073]: STATEEEEEEEEEEEEE NOW
[15:40:18][I][main:226]: AC Active: true
[15:40:18][I][main:227]: Fan switch: true
[15:40:18][I][main:228]: AC Mode: 1
[15:40:18][I][main:229]: AC Action: 2
[15:40:18][I][main:256]: AC: ON, HEATER: OFF, FAN: ON, LID: open!
**[15:40:18][D][servo:062]: Servo new target: 0.500000**
[15:40:18][D][switch:016]: 'heater_relais' Turning OFF.
[15:40:18][D][light:036]: 'Prusa Enclosure - Fan' Setting:
[15:40:18][D][light:051]:   Brightness: 100%
[15:40:18][W][component:204]: Component thermostat.climate took a long time for an operation (0.11 s).
[15:40:18][W][component:205]: Components should block for at most 20-30ms.
**[15:40:18][D][servo:050]: Servo reached target**
[15:40:18][W][component:204]: Component st7735.display took a long time for an operation (0.09 s).

if-kenn, 2023-12-10:
I am finding this issue as well.

ssieb, 2024-03-24:
We can't reproduce this issue with the current release.  If you still see this, then reopen with config and logs.

closed 2024-03-24 by ssieb