esphome/issues #4986: GPIO 5 does not work when in input pulldown mode

skykingjwc, 2023-10-15:
### The problem

Attempting to use GPIO 5 in input pulldown mode does not work.  The voltage hovers around 2.5.  Pulling it up to 3.3 has no affect, likely because the voltage never crosses the 0.25 (low) and 0.75 (high) voltage thresholds.  The documentation for the ESP32 says it should work so I'm wondering if there is a software issue.  Tried on two ESP32 Devkit boards with the same result.  Pullup mode appears to work fine.

### Which version of ESPHome has the issue?

2023.10.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

binarysensor/gpio

### Example YAML snippet

```yaml
esphome:
    name: mytest

esp32:
  board: esp32dev
  framework:
    type: esp-idf  #also tried with arduino with same result
    version: recommended

logger:

binary_sensor:
  - platform: gpio
    name: Test
    id: Test
    pin:
      number: 5
      mode:
        input: true
        pulldown: true

    filters:
      - delayed_on: 10ms

    # Optional variables:
    on_press:
      - logger.log: "Input clicked"  # should see output in the log when pulling high, but nothing happens
```

### Anything in the logs that might be useful for us?

```txt
Not really,  the problem is that the GPIO never registers the press, so there is no loggable event.
```

### Additional information

Does work if using an external pull down resistor, but this should not be necessary as the internal pulldown resistor should be sufficient if it is activated by software.

ssieb, 2023-10-16:
I can verify this, but I also can't find any reason for it.  It's not even an RTC gpio.

glmnet, 2025-03-28:
possible duplicate of https://github.com/esphome/issues/issues/3722

ssieb, 2025-03-28:
Not a duplicate.  This is a physical pin issue, nothing to do with the binary sensor code.

skykingjwc, 2025-04-11:
I was reviewing the ESP32 docs, and it appears that the internal pull up and pull down resistors on a single GPIO pin can be activated at the same time, which would create a voltage divider.   

Thinking the pull up resistor may be turned on, and that might be what is causing this issue, I changed the code as follows, but unfortunately it had no effect. 

``
binary_sensor:
  - platform: gpio
    name: Test
    id: Test
    pin:
      number: 5
      mode:
        input: true
        pulldown: true
        **pullup: false**
``