esphome/issues #4987: Dashboard window resize closes yaml editor without saving

samuelolteanu, 2023-10-16:
### The problem

Like the title sais, when editing ESPHome yaml on Windows Chrome, I need to be extra careful to not modify window size when editing or flashing because it will revert to device list without saving any work.
This happens of Firefox too but accidentally pressing F12 does not modify window size.

https://github.com/esphome/issues/assets/85267083/4f4e4747-534a-422c-ade6-0819f2c2a282

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.3

### What platform are you using?

ESP8266

### Component causing the issue

Dashboard

### Additional information

Separate issue maybe? Can't edit esphome yaml on android with the dashboard. Cannot select text and error detail popups don't show wich is a bummer because I could get a board with me and connect it via wireguard component but i need a laptop too.

https://github.com/esphome/issues/assets/85267083/44ff71dd-d7c1-4e71-b851-22c62b9f60f9

randybb, 2023-10-16:
Cannot reproduce that on latest chrome and firefox - it never gets reloaded.

The second "issue" is a known one - limitation of the Monaco editor.

jesserockz, 2023-10-18:
I just tried this myself on Chrome and I could not reproduce it at all. No matter how much I changed the size, it stayed in the editor the whole time

samuelolteanu, 2023-10-18:
On remote desktop, it's resolution dependant: not happening at 720p, present at more than 1080p, any scale >= 100%. Tried on 2 different remote windows computers (win10 and server 2022) on 2 different HA installs new + old. Latest Chrome, Firefox, Edge. Tried sfc scannow, created different windows user.
Always present on physical access, no matter the resolution and dpi, even after windows restart.

samuelolteanu, 2023-11-16:
[image]
Quick workaround....

patienttruth, 2024-05-19:
I just had this happen when compiling a config to update with. Not sure if it would be a problem, but glad it didn't happen when the actual OTA update was happening.

JonLaliberte, 2024-11-07:
I have this problem all the time.

I've noticed it often when trying to make quick edits from iPad, but it happens on my computer in Arc (Chromium) and Chrome as well (tested in both today before posting).

It happens when viewing the log as well.

Just now I was compiling for a manual download to upload (to a Tasmota device) and while it was running I opened a new split view which resized the window and poof, gone.

The key is to resize to the point where the dashboard wants to show a different screen size UI.

Screen recording: https://share.cleanshot.com/YRnRcxrh

AJolly, 2025-04-21:
It still ends up doing this

heibertelf, 2025-06-18:
Same problem Microsoft Edge browser

https://github.com/user-attachments/assets/16a044a1-6dda-4139-95a5-70f235690bc0