esphome/issues #4992: Missing the `pip` binary!

killproc84, 2023-10-17:
### The problem

Hello,
I'm trying to port the firmware "https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml" to my AI-Thinker Audio Kit v.2.2 but I get IDF not running.
I'm trying to update pip in container but it doesn't help.
Now I don't know what to do, but it looks to me like a faulty Docker container, or have I missed something?

Thanks.

### Which version of ESPHome has the issue?

2023.10.0b3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

AI-Thinker Audio Kit

### Component causing the issue

esp-idf

### Example YAML snippet

```yaml
esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    version: latest
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.10.0b3
INFO Reading configuration config/hansecho01.yaml...
INFO Updating https://github.com/mulcmu/esphome-es8388-a1s.git@nightdav-patch
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.2.0
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
INFO Updating https://github.com/espressif/esp-dsp@v1.2.0
INFO Compiling app...
Processing hansecho01 (board: esp-wrover-kit; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50100.0 (5.1.0) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Creating a new virtual environment for IDF Python dependencies
AssertionError: Error: Failed to create a proper virtual environment. Missing the `pip` binary!:
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/config/.esphome/platformio/platforms/espressif32/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Util/envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/platformio/builder/tools/piobuild.py", line 342:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 662:
    return method(*args, **kw)
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/config/.esphome/platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/config/.esphome/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1232:
    ensure_python_venv_available()
  File "/config/.esphome/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1208:
    _create_venv(venv_dir)
  File "/config/.esphome/platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1200:
    assert os.path.isfile(
========================== [FAILED] Took 2.74 seconds ==========================
```

jesserockz, 2023-10-18:
I believe the issue here is the incompatible esp-idf and platformio versions.

> WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.

If you want to use `latest` (5.1.0), you need a matching platformio version. 

As this is an "unsupported" setup, I am going to close this issue.

closed 2023-10-18 by jesserockz

killproc84, 2023-10-18:
Hi @jesserockz,
thank you. But inside the Docker Container, i don't control the Version, so i only Control the Version of esp-idf in yaml. "latest" i give on troubleshooting, sorry for this. When i put "recommend" i get the same error;
`esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf
    version: recommended`

`INFO ESPHome 2023.10.0b3
INFO Reading configuration config/hansecho01.yaml...
INFO Updating https://github.com/mulcmu/esphome-es8388-a1s.git@nightdav-patch
INFO Generating C++ source...
INFO Compiling app...
Processing hansecho01 (board: esp-wrover-kit; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40405.230623 (4.4.5) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Creating a new virtual environment for IDF Python dependencies
AssertionError: Error: Failed to create a proper virtual environment. Missing the `pip` binary!:`