esphome/issues #4994: Platformio board_dir location unreachable after removing /config/esphome/.esphome directory?

Goodwillson, 2023-10-17:
### The problem

In previous versions was possible to save custom board configuration into /config/esphome/.esphome/build/[your-esphome-device]/boards directory. This location does not exist anymore. I didnt find path where I could access location which platformio looks for the configuration files by default. I guess it all happens inside plugin container now, which is not accessible from HassOS. I tried to set full path like boards_dir: /config/esphome/boards, but that doesnt work either. How I could make it working again while keeping the boards condiguration outside of yaml file?

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.6.1

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: liligo-t5-47-epaper-plus
  platformio_options:
    board_dir: boards
    #include https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/master/platformio/boards/lilygo-t5-47-plus.json as custom board
    board: lilygo-t5-47-plus

esp32:
  variant: esp32s3
  board: lilygo-t5-47-plus
```

ssieb, 2023-10-17:
Yes, the build directory was moved inside the add-on so it wouldn't cause unnecessary bloat in config backups.
What is the benefit of using that particular board file?

Goodwillson, 2023-10-18:
I wasnt able to convert all options from json file into platformio_options without some warnings during compilation. When I use the provided json file it goes smooth and I can reuse it in other devices without copying code, I guess thats main advantage. In the end I have solved it by copying the json file into docker image (docker cp /config/esphome/boards/lilygo-t5-47-plus.json addon_5c53de3b_esphome:/data/build/weather-epaper-display/boards/) Would be handy if there would be a way how to include board files like other includes, which are fetched from place where yaml configuration file is or even better from location outside of specific project folder, so it could be reused/shared between projects.

nickolay, 2023-12-29:
People seem to have success with setting `esphome.build_path` back to `/config/...`, per these instructions: https://github.com/Januznl/esphome-lilygo-t5-s3-plus

There's also https://github.com/esphome/feature-requests/issues/2370 requesting custom board support in esphome.