esphome/issues #4995: Update 2023.10.0 failed

roby70cr, 2023-10-18:
### The problem

when trying to update to 2023.10.0 I get this error:

>INFO ESPHome 2023.10.0
INFO Reading configuration /config/esphome/nspanel.yaml...
INFO Updating https://github.com/Blackymas/NSPanel_HA_Blueprint@dev
INFO Detected timezone 'Europe/Rome'
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing nspanel (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPmDNS @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
Compiling .pioenvs/nspanel/src/main.cpp.o
Compiling .pioenvs/nspanel/lib64d/WiFi/WiFiGeneric.cpp.o
/data/packages/891b50f5/nspanel_esphome_core.yaml: In lambda function:
/data/packages/891b50f5/nspanel_esphome_core.yaml:1507:73: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
             id(disp1).set_component_text_printf("boot.ip_addr", "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
                                                                         ^
/data/packages/891b50f5/nspanel_esphome_core.yaml:1507:80: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
             id(disp1).set_component_text_printf("boot.ip_addr", "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
Compiling .pioenvs/nspanel/lib64d/WiFi/WiFiMulti.cpp.o
                                                                                ^
/data/packages/891b50f5/nspanel_esphome_core.yaml:1507:87: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
             id(disp1).set_component_text_printf("boot.ip_addr", "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
                                                                                       ^
/data/packages/891b50f5/nspanel_esphome_core.yaml:1507:94: error: no match for 'operator[]' (operand types are 'esphome::network::IPAddress' and 'int')
             id(disp1).set_component_text_printf("boot.ip_addr", "%d.%d.%d.%d", ip[0], ip[1], ip[2], ip[3]);
                                                                                              ^
Compiling .pioenvs/nspanel/lib64d/WiFi/WiFiSTA.cpp.o
Compiling .pioenvs/nspanel/lib64d/WiFi/WiFiScan.cpp.o
Compiling .pioenvs/nspanel/lib64d/WiFi/WiFiServer.cpp.o
*** [.pioenvs/nspanel/src/main.cpp.o] Error 1
========================= [FAILED] Took 32.34 seconds =========================<

### Which version of ESPHome has the issue?

2023.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.3

### What platform are you using?

ESP32

### Board

sonoff nspanel

ssieb, 2023-10-18:
That's a problem with the lambda in your yaml, not a bug in esphome.  IP addresses support IPv6 now, so you can't assume IPv4.
Try `set_component_text("boot.ip_addr", ip.str().c_str());`, but if you have any more questions about this, come ask on the esphome discord.

closed 2023-10-18 by ssieb

roby70cr, 2023-10-18:
thank you, will do, since I did not understand what's it related to and how to solve.