esphome/issues #4996: Optional parameter (default value) in register_service.

andrewjswan, 2023-10-18:
### The problem

How can I declare an optional parameter (default value) in register_service so that not all fields are required to be filled in when calling a service from HA, but default values are applied to the unfilled ones?

```
void CUSTOM::my_proc(int par1, int par2 = 0, int par3 = 0)
register_service(&CUSTOM::my_proc, "my_proc", {"par1", "par2", "par3"});
```

How to make `Par2`, `Par3` optional in the example above? HA requires all parameters to be filled in when calling the service.

### Which version of ESPHome has the issue?

2023.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.3

### What platform are you using?

ESP32

### Component causing the issue

api/custom_api_device

andrewjswan, 2024-02-16:
I haven’t received an answer yet, it’s relevant.

ssieb, 2026-03-24:
The answer is that you can't.  Follow ups in the linked feature request.