esphome/issues #4997: HTTP Request does not support bk72xx board

yaminahacs, 2023-10-18:
### The problem

bk72xx:
  board: generic-bk7231t-qfn32-tuya

This board doesn't support HTTP Request

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Board

bk72xx

davewatson91, 2024-02-16:
Hi all, 
Just checking to see if there is any plans for this one at all? 
Thanks!

bennydiamond, 2024-12-23:
I'm kinda wondering myself as well why is this not supported.
I know mbedtls isn't properly supported on Libretiny but if esp8266 can work without encryption, how come bk72xx platform couldn't either?