esphome/issues #4999: Voice Assistant can't be used without a speaker

andrey-yantsen, 2023-10-18:
### The problem

Yes, I want to do something strange — use a voice assistant without the voice.It should either be forbidden during the YAML validation stage or should be allowed :)

When compiling the firmware with a `voice_assistant` config without a speaker, the following error is displayed when building the firmware:

```
src/esphome/components/voice_assistant/voice_assistant.cpp: In member function 'virtual void esphome::voice_assistant::VoiceAssistant::loop()':
src/esphome/components/voice_assistant/voice_assistant.cpp:320:13: error: 'class esphome::voice_assistant::VoiceAssistant' has no member named 'wait_for_stream_end_'
       this->wait_for_stream_end_ = false;
             ^~~~~~~~~~~~~~~~~~~~
```

### Which version of ESPHome has the issue?

2023.10.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

M5Stack Atom U

### Example YAML snippet

```yaml
voice_assistant:
  id: va
  microphone: atomu_microphone
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  on_listening:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: pulse
  on_tts_start:
    - light.turn_on:
        id: led
        blue: 0%
        red: 0%
        green: 100%
        brightness: 100%
        effect: pulse
  on_error:
    - light.turn_on:
        id: led
        blue: 0%
        red: 100%
        green: 0%
        brightness: 100%
        effect: none
    - delay: 1s
    - script.execute: reset_led
    - script.wait: reset_led
    - lambda: |-
        if (code == "wake-provider-missing" || code == "wake-engine-missing") {
          id(use_wake_word).turn_off();
        }
```

andrey-yantsen, 2023-10-18:
Great timing :D Looks like it's a duplicate of #4998, which was created a minute faster than this issue.

chatziko, 2023-10-18:
Same issue here. Btw using a satellite without a speaker is not strange at all, and does work in esphome beta last time I tried.

WizBangCrash, 2023-10-18:
Same issue here. Using a M5StickC Plus with microphone, but no speaker. Was compiling okay in 2023.10.0b3 tough.

andrey-yantsen, 2023-10-18:
Fixed by esphome/esphome#5558 this morning. Now we wait for the [2023.10.1](https://github.com/esphome/esphome/milestone/300) release.

cvwillegen, 2023-10-18:
Does this fix the compile errors on Atom Echo?

andrey-yantsen, 2023-10-18:
@cvwillegen that PR fixes the following error regardless of the device:
```
src/esphome/components/voice_assistant/voice_assistant.cpp: In member function 'virtual void esphome::voice_assistant::VoiceAssistant::loop()':
src/esphome/components/voice_assistant/voice_assistant.cpp:320:13: error: 'class esphome::voice_assistant::VoiceAssistant' has no member named 'wait_for_stream_end_'
       this->wait_for_stream_end_ = false;
             ^~~~~~~~~~~~~~~~~~~~
```

mattbruman, 2023-10-19:
> Same issue here. Using a M5StickC Plus with microphone, but no speaker. Was compiling okay in 2023.10.0b3 tough.

Can you share your yaml? I want to use the same device. Thanks.

closed 2023-10-19 by andrey-yantsen

WizBangCrash, 2023-10-19:
Hi @mattbruman,
Here is my working config without display support. I'm working on adding the axp192 component so that the display is usable too. Just not quite there yet :-)

```
substitutions:
  name: m5stickc-plus-esp32-ac918c
  friendly_name: M5StickC Plus ESP32 ac918c
# packages:
#   m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "1.0"
  min_version: 2023.10.0b1
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

esp32:
  board: m5stick-c
  framework:
    type: esp-idf

logger:
  level: DEBUG

api:
  encryption:
    key: !secret esphome_key

ota:

dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

wifi:
  ap:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

improv_serial:

script:
  - id: reset_led
    then:
      - if:
          condition:
            switch.is_on: use_wake_word
          then:
            - light.turn_on:
                id: led
                effect: "Listening"
          else:
            - light.turn_off: led

output:
  - platform: ledc
    pin: GPIO2
    id: buzzer_pwm
  - platform: ledc
    pin: GPIO10
    id: led_pwm
    inverted: true

i2s_audio:
  i2s_lrclk_pin: GPIO0

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO34
    adc_type: external
    pdm: true

voice_assistant:
  id: va
  microphone: echo_microphone
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  vad_threshold: 3
  on_listening:
    - light.turn_on:
        id: led
        effect: "Listening"
  on_tts_start:
    - light.turn_on:
        id: led
        brightness: 60%
    - delay: 100ms
    - light.turn_off: led
  on_end:
    - delay: 100ms
    - script.execute: reset_led
  on_error:
    - light.turn_on:
        id: led
        effect: "Error"
    - delay: 2s
    - script.execute: reset_led
    - script.wait: reset_led
    - lambda: |-
        if (code == "wake-provider-missing" || code == "wake-engine-missing") {
          id(use_wake_word).turn_off();
        }

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO37
      inverted: true
    name: Button
    disabled_by_default: true
    entity_category: diagnostic
    id: echo_button
    on_click:
      - if:
          condition:
            switch.is_off: use_wake_word
          then:
            - if:
                condition: voice_assistant.is_running
                then:
                  - voice_assistant.stop:
                  - script.execute: reset_led
                else:
                  - voice_assistant.start:
          else:
            - voice_assistant.stop
            - delay: 1s
            - script.execute: reset_led
            - script.wait: reset_led
            - voice_assistant.start_continuous:
  - platform: gpio
    pin:
      number: GPIO39
      inverted: true
    name: ButtonB
    disabled_by_default: false
    entity_category: diagnostic
    id: buzzer_button
    on_click:
      - rtttl.play: 'siren:d=8,o=5,b=100:d,e,d,e,d,e,d,e'

light:
  - platform: monochromatic
    id: led
    name: None
    disabled_by_default: true
    entity_category: config
    output: led_pwm
    effects:
      - pulse:
          name: "Listening"
          transition_length: 1s
          update_interval: 1s
          min_brightness: 0%
          max_brightness: 30%
      - pulse:
          name: "Error"
          transition_length: 0.25s
          update_interval: 0.25s
          min_brightness: 0%
          max_brightness: 100%

rtttl:
  output: buzzer_pwm

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(va).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
      - script.execute: reset_led
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);
      - script.execute: reset_led
  - platform: output
    name: Buzzer
    output: buzzer_pwm
    on_turn_on:
      then:
      - output.turn_on: buzzer_pwm
      - output.ledc.set_frequency:
          id: buzzer_pwm
          frequency: 2000Hz
      - output.set_level:
          id: buzzer_pwm
          level: 100%
    on_turn_off:
      then:
        - output.turn_off: buzzer_pwm

esp_adf:

external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s
```