esphome/issues #5001: OTA update don't install ESP-32

19leva89, 2023-10-18:
### The problem

I have a device on ESP-32 (ESP32-D0WD-V3). After installing the firmware via ESP-Home, you can update it once over the air (OTA) and it doesn’t want to update anymore, although it says that it was updated and everything is OK (I’m attaching the log). Has anyone encountered this problem and is there a way out of this situation?

```
INFO ESPHome 2023.10.0
INFO Reading configuration /config/esphome/xiaomi-ir.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing xiaomi-ir (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- HeatpumpIR @ 1.0.23
|-- IRremoteESP8266 @ 2.7.12
RAM:   [=         ]  12.7% (used 41500 bytes from 327680 bytes)
Flash: [======    ]  59.2% (used 1086501 bytes from 1835008 bytes)
========================= [SUCCESS] Took 27.67 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of xiaomi-ir.local
INFO  -> 192.168.31.47
INFO Uploading /data/build/xiaomi-ir/.pioenvs/xiaomi-ir/firmware.bin (1092272 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from xiaomi-ir.local using esphome API
INFO Successfully connected to xiaomi-ir.local
[16:18:52][I][app:102]: ESPHome version 2023.9.3 compiled on Oct  6 2023, 20:33:09
```

### Which version of ESPHome has the issue?

2023.10.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.10.3

### What platform are you using?

ESP32

### Board

ESP32-D0WD-V3

donburch888, 2024-02-09:
YES !  I (and others) have this problem regularly with Arlec PC191HA (which uses a WB2S module). 

The firmware DID compile and upload successfully.  

The key is in the first line from the run LOG from the device ... where we find that the device is still running a version of the firmware.  
`[16:18:52][I][app:102]: ESPHome version 2023.9.3 compiled on Oct  6 2023, 20:33:09`

1. First line of the provided log indicates ESPHome 2023.10.0 was compiled; yet ESPHome version 2023.9.3" is being run
2. the 16:18:52 timestamp is not consistent with the "Oct  6, 20:33:09" compile date supplied by the device. 

Conclusion is that the firmware did upload correctly - but the device did not automatically activate the new firmware when it reset.  

A power cycle of the device will hopefully trigger the firmware to activate; though you may need to leave the device unplugged for a couple of hours.

Shogoki, 2024-04-22:
Did anybody come up with a good solution or workaround for this.

I think I am experiencing the same issue with 2 of my Shelly Plus 2PM devices. 
I was able to OTA flash them from the original firmware to Tasmota and then I OTA-flashed a basic config of esphome to it. 
Then I used esphome to OTA-flash another config to it, all without problems. 
The current firmware I have, however seems to stick and I am not able to make any changes and get it to load a new firmware. 
The esphome upload seems to succeed, but it still boots up with the old Firmware.
I created another issue #5720, where I did not realize the reason. Now I know it is because the new Firmware is not loading.

I did try to remove the power from these devices and turn it back on. This does not help, unfortunately. 
Also tried keeping them without power for up to 90 minutes, with no effect.
Really annoying.. Would be super happy if someone has an idea on what to do here.

donburch888, 2024-05-16:
19leva89, did you try power cycling the device after uploading the firmware ?

Did you try powering the device off for 10 minutes after uploading the firmware ?

Not the ideal solution I know, but if your device also behaves this way then possibly the developers can program a power cycle rather than just a reset after a successful firmware upload.

ssieb, 2024-05-16:
Or try uploading twice without power cycling which I already suggested.  Did you try it?

19leva89, 2024-07-04:
> Did you try power cycling the device after uploading the firmware ?
> Did you try powering the device off for 10 minutes after uploading the firmware ?
> 

I tried it - it doesn't help!

> Or try uploading twice without power cycling which I already suggested. Did you try it?

I tried it - it doesn't help!

If I connect to a PC via UART, then everything is flashed without problems