esphome/issues #5009: Pulse Meter Sensor doesn't respond in HA with service set_total_pulses

DJTerentjev, 2023-10-19:
### The problem

When I try to update value of Pulse Meter Sensor with service set_total_pulses it doesn't  update the value of corresponding sensor in Home assistant (like it did before). But sensor got updated with next pulse.

### Which version of ESPHome has the issue?

2023.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2023.10.3

### What platform are you using?

ESP32

### Board

wt32_eth01

### Component causing the issue

Pulse Meter Sensor

### Example YAML snippet

```yaml
service: esphome.wt32_eth01_set_gas_total
data:
  new_gas_total: 1
```

DJTerentjev, 2023-10-19:
I mean, I have to wait next pups to see new value in Home assistant

ssieb, 2023-10-19:
Duplicate of https://github.com/esphome/issues/issues/4941
Fix in https://github.com/esphome/esphome/pull/5475

closed 2023-10-19 by ssieb