esphome/issues #5012: Webserver not starting after changing IP and subnet

Kalekulan, 2023-10-20:
### The problem

So previously, I had 148.64.36.0/255.255.255.0 for the subnet on my router and esphome worked fine with my ESP8266. Now I have changed back to 192.168.0.0/255.255.255.0.

DHCP lease has expired, and it's possible to ping the device on the new IP address. But the web server ain't running (checked opened ports). And it doesn't seem to self-heal. I have a H801 device which isn't easy to reset (to my knowledge).

### Which version of ESPHome has the issue?

2023.7

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

webserver

ssieb, 2023-10-20:
You power cycled it?
Can you get logs?
Can you OTA update it?

Kalekulan, 2023-10-20:
Tried all three.  Nothing works. I went back to the same network config on my router but problem persists.

ssieb, 2023-10-20:
Then that suggests it wasn't the IP change that mattered.  How did you install this device initially?  It would be easier to discuss this on the esphome discord instead.  This isn't really appropriate for github issues.