esphome/issues #5013: ESP Home Entity not available in Deepsleep in latest Version

MatthiasDenkler, 2023-10-20:
### The problem

As the Title says as soon as the ESP goes into deep sleep, the Entitys in Homeassistant are going to "Currently not available" in the latest Version.
When this happens in the Logs I get this Message:

### Which version of ESPHome has the issue?

2023.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2023.10.3

### What platform are you using?

ESP32

### Board

Lolin32 ESP32 Lite 4MB

### Component causing the issue

api?

### Example YAML snippet

```yaml
esphome:
name: air-out
friendly_name: air-out
esp32:
board: esp32dev
framework:
type: arduino
i2c:
sda: GPIO16
scl: GPIO17
# Enable logging
logger:
# Enable Home Assistant API
api:
encryption:
key: "snip"
ota:
password: "snip"
wifi:
ssid: !secret wifi_ssid
password: !secret wifi_password
fast_connect: true
manual_ip:
static_ip: 192.168.128.14
gateway: 192.168.178.1
subnet: 255.255.128.0
dns1: 192.168.178.1
# Enable fallback hotspot (captive portal) in case wifi connection fails
ap:
ssid: "Air-Out Fallback Hotspot"
password: "snip"
captive_portal:
sensor:
- platform: bme280
temperature:
name: "Outside_Temp"
id: bme280_temperature
pressure:
[... 183 lines cut ...]
[22:15:07][D][esp32.preferences:114]: Saving 1 preferences to flash...
[22:15:07][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[22:15:07][I][app:029]: Running through setup()...
[22:15:07][I][i2c.arduino:183]: Performing I2C bus recovery
[22:15:07][C][bme280.sensor:086]: Setting up BME280...
[22:15:07][C][adc:047]: Setting up ADC 'Outside_Battery'...
[22:15:07][C][adc:093]: ADC 'Outside_Battery' setup finished!
[22:15:07][C][wifi:038]: Setting up WiFi...
[22:15:07][C][wifi:051]: Starting WiFi...
[22:15:07][C][wifi:052]:   Local MAC: C8:C9:A3:C9:10:14
[22:15:07][I][wifi:277]: WiFi Connecting to 'Adnexus'...
[22:15:07][D][sensor:094]: 'Outside_Battery': Sending state 3.76600 V with 2 decimals of accuracy
[22:15:07][D][sensor:094]: 'Outside_Dew': Sending state -nan °C with 1 decimals of accuracy
[22:15:07][D][sensor:094]: 'Outside_Temp': Sending state 24.28000 °C with 1 decimals of accuracy
[22:15:07][D][sensor:094]: 'Outside_Pressure': Sending state 962.03943 hPa with 1 decimals of accuracy
[22:15:07][D][sensor:094]: 'Outside_Humidity': Sending state 56.30078 % with 1 decimals of accuracy
[22:15:11][D][esp-idf:000]: E (7551) wifi:
[22:15:11][D][esp-idf:000]: Association refused temporarily, comeback time 1024 mSec
[22:15:11][D][esp-idf:000]: 
[22:15:11]
[22:15:13][D][esp-idf:000]: E (9601) wifi:
[22:15:13][D][esp-idf:000]: Association refused temporarily, comeback time 1024 mSec
[22:15:13][D][esp-idf:000]: 
[22:15:13]
```

MatthiasDenkler, 2023-10-23:
After latest Homeassistant Coreupdate, this works again as intended. But now the measured and sended States dont get displayed anymore. Can this be a Bug of ESPHome or is it more a Homeassistant Problem?

VIMVa, 2023-11-13:
I'm on latest updates of everything but devices in deep sleep still are unavailable regularly.
[image]

jesserockz, 2023-11-14:
With a run_duration of only 30 seconds, and update_interval of 40s for the sensors, its likely that they just dont update in time. The first update they do is at a random time between 0 and the update interval to spread out updates.

The best way to solve this is to use an `on_boot` automation to wait until the api is connected, then force an update on each sensor, delay for a second or 2 to allow network traffic to be sent, then manually enter deep_sleep once that is done.

```yaml
esphome:
  on_boot:
    - wait_until: api.connected
    - component.update: bme280_id # give this id to the parent bme sensor
    - component.update: outside_dew # give this id to the Outside_Dew template sensor
    - component.update: outside_battery # give this id to the adc sensor
    - delay: 1s
    - deep_sleep.enter:
```

And delete the `run_duration` from the deep sleep config.
This will likely let it run for even less than 30 seconds too as it only stays on for maybe 1 or 2 seconds after api is connected.

VIMVa, 2023-11-14:
Will give it a try!
Just a question from an eager-to-learn rookie: does the 'on boot' also run on wake after deep sleep then or only at actual boot after forced reset?

ssieb, 2023-11-14:
A wake up is the same as a reboot.

VIMVa, 2023-11-14:
Thanks for clarifying... but I should have known 🤭.
I already have an on_boot to check if a helper switch is on so that the device stays awake and give me opportunity to install new code wirelessly when needed.

VIMVa, 2023-11-14:
Workaround still failing to keep connected here. Intermittent disconnection still popping up. My code for who would like to check:
`
esphome:
  name: temp-004
  on_boot:
      - wait_until: api.connected
      - component.update: bme280_temp004 # give this id to the parent bme sensor
      - component.update: wifi_signal_db_temp004 # give this id to the Wi-Fi sensor
      - component.update: batterij_temp004 # give this id to the adc sensor
      - delay: 5s
      - script.execute: consider_deep_sleep_temp004

esp8266:
  board: d1_mini

# Enable logging
logger:

api:
  encryption:
    key: "Ucno7qxcEczHMQFf0U4xHuxwxgjP€xxxxxx="

ota:
  password: "7f4715d3751axxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.1.134
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temp-004 Fallback Hotspot"
    password: "OU1xxxxxxxxx"

captive_portal:
    
i2c:
  scan: False

sensor:
  - platform: bme280
    id: bme280_temp004
    temperature:
      name: "Temperatuur"
      filters:
        - offset: 0.4
    pressure:
      name: "Luchtdruk"
      filters:
        - offset: 2.1
    humidity:
      name: "Luchtvocht"
    address: 0x76
    update_interval: 6min

  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db_temp004
    update_interval: 6min
    entity_category: "diagnostic"

  - platform: adc
    id: batterij_temp004
    pin: A0
    name: "Batterij"
    update_interval: 6min  
    unit_of_measurement: "%"
    accuracy_decimals: 0
    icon: mdi:battery-medium
    filters:
      - multiply: 4.5
      - calibrate_polynomial:
         degree: 3
         datapoints:
          - 0.00 -> 0.0
          - 3.30 -> 0.0
          - 3.35 -> 5.0
          - 3.39 -> 10.0
          - 3.44 -> 15.0
          - 3.48 -> 20.0
          - 3.53 -> 25.0
          - 3.57 -> 30.0
          - 3.62 -> 35.0
          - 3.64 -> 40.0
          - 3.69 -> 45.0
          - 3.73 -> 50.0
          - 3.78 -> 55.0
          - 3.82 -> 60.0
          - 3.87 -> 65.0
          - 3.91 -> 70.0
          - 3.96 -> 75.0
          - 4.00 -> 80.0
          - 4.05 -> 85.0
          - 4.09 -> 90.0
          - 4.14 -> 95.0
          - 4.20 -> 100.0
      - lambda: |-
          if (x <= 100) {
            return x;
          } else {
            return 100;
          }

deep_sleep:
  id: deep_sleep_control_temp004
#  run_duration: 20s
  sleep_duration: 15min

# Will only pick up a value if this device is configured in Home Assistant > Integrations
# If the device isn't configured, or Home Assistant is offline, it'll default to false
binary_sensor:
  - platform: homeassistant
    id: prevent_deep_sleep_temp004
    entity_id: input_boolean.prevent_deep_sleep_temp004

script:
  - id: consider_deep_sleep_temp004
    mode: queued
    then:
      - delay: 10s
      - if:
          condition:
           binary_sensor.is_on: prevent_deep_sleep_temp004
          then:
           - logger.log: 'Skipping sleep, per prevent_deep_sleep_temp004'
          else:
           - deep_sleep.enter: deep_sleep_control_temp004
      - script.execute: consider_deep_sleep_temp004

web_server:
  port: 80`

VIMVa, 2023-11-14:
No luck with this workaround here: already 2 disconnections in as many hours like befor. My code for if someone would like to have a check:

```
esphome:
  name: temp-004
  on_boot:
      - wait_until: api.connected
      - component.update: bme280_temp004 # give this id to the parent bme sensor
      - component.update: wifi_signal_db_temp004 # give this id to the Wi-Fi sensor
      - component.update: batterij_temp004 # give this id to the adc sensor
      - delay: 5s
      - script.execute: consider_deep_sleep_temp004

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Ucno7qxcEczHMQFf0Uxxxxxxxx"

ota:
  password: "7f471xxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  manual_ip:
    static_ip: 192.168.1.134
    gateway: 192.168.1.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temp-004 Fallback Hotspot"
    password: "OxxxxxUx1xxx"

captive_portal:
    
i2c:
#  sda: 4
#  scl: 5
  scan: False

sensor:
  - platform: bme280
    id: bme280_temp004
    temperature:
      name: "Temperatuur"
      filters:
        - offset: 0.4
    pressure:
      name: "Luchtdruk"
      filters:
        - offset: 2.1
    humidity:
      name: "Luchtvocht"
    address: 0x76
    update_interval: 6min

  - platform: wifi_signal # Reports the WiFi signal strength/RSSI in dB
    name: "WiFi Signal dB"
    id: wifi_signal_db_temp004
    update_interval: 6min
    entity_category: "diagnostic"

  - platform: adc
    id: batterij_temp004
    pin: A0
    name: "Batterij"
    update_interval: 6min  
    unit_of_measurement: "%"
    accuracy_decimals: 0
    icon: mdi:battery-medium
    filters:
      - multiply: 4.5
      - calibrate_polynomial:
         degree: 3
         datapoints:
          - 0.00 -> 0.0
          - 3.30 -> 0.0
          - 3.35 -> 5.0
          - 3.39 -> 10.0
          - 3.44 -> 15.0
          - 3.48 -> 20.0
          - 3.53 -> 25.0
          - 3.57 -> 30.0
          - 3.62 -> 35.0
          - 3.64 -> 40.0
          - 3.69 -> 45.0
          - 3.73 -> 50.0
          - 3.78 -> 55.0
          - 3.82 -> 60.0
          - 3.87 -> 65.0
          - 3.91 -> 70.0
          - 3.96 -> 75.0
          - 4.00 -> 80.0
          - 4.05 -> 85.0
          - 4.09 -> 90.0
          - 4.14 -> 95.0
          - 4.20 -> 100.0
      - lambda: |-
          if (x <= 100) {
            return x;
          } else {
            return 100;
          }

deep_sleep:
  id: deep_sleep_control_temp004
#  run_duration: 20s
  sleep_duration: 15min

# Will only pick up a value if this device is configured in Home Assistant > Integrations
# If the device isn't configured, or Home Assistant is offline, it'll default to false
binary_sensor:
  - platform: homeassistant
    id: prevent_deep_sleep_temp004
    entity_id: input_boolean.prevent_deep_sleep_temp004

script:
  - id: consider_deep_sleep_temp004
    mode: queued
    then:
      - delay: 10s
      - if:
          condition:
           binary_sensor.is_on: prevent_deep_sleep_temp004
          then:
           - logger.log: 'Skipping sleep, per prevent_deep_sleep_temp004'
          else:
           - deep_sleep.enter: deep_sleep_control_temp004
      - script.execute: consider_deep_sleep_temp004
```

ssieb, 2023-11-14:
I don't know what you mean by "disconnections", but if you want to find out what's going on, then you need to watch the serial logs.

VIMVa, 2023-11-15:
Had hoped that today's version 2023.11.0 would have solved this issue... but alas.
Only with my 2 D1's with deep sleep active.
```
Logger: aioesphomeapi.reconnect_logic
Source: runner.py:188
First occurred: 14 november 2023 om 22:26:55 (184 occurrences)
Last logged: 18:35:57
Can't connect to ESPHome API for temp-004 @ 192.168.1.134: Error connecting to ('192.168.1.134', 6053): [Errno 113] Connect call failed ('192.168.1.134', 6053) (SocketAPIError)
Can't connect to ESPHome API for temp-002 @ 192.168.1.132: Timeout while connecting to ('192.168.1.132', 6053) (SocketAPIError)
Can't connect to ESPHome API for temp-004 @ 192.168.1.134: Timeout while connecting to ('192.168.1.134', 6053) (SocketAPIError)
Can't connect to ESPHome API for temp-002 @ 192.168.1.132: Error connecting to ('192.168.1.132', 6053): [Errno 113] Connect call failed ('192.168.1.132', 6053) (SocketAPIError)
```

VIMVa, 2023-11-16:
TWIMC: [this](https://github.com/esphome/issues/issues/4794#issuecomment-1709004676) is the only workaround that keeps the devices as 'available' during deep sleep for me. Increased the DisconnectRequest to 1000ms. Maybe less will work as well.

ssieb, 2024-01-13:
Was this issue resolved by that change?

closed 2025-06-08 by jesserockz