esphome/issues #5014: Problem with turning on the "heating/cooling" mode when pressing the "SMART" button on the Whirlpool (Galactic) DG11J1-91 IR remote control

TokarevSergey, 2023-10-21:
### The problem

At this stage the situation is like this.

If the air conditioner is turned off.
and press "SMART" on the remote control
[image]

Then the air conditioner turns on
[image]

And in the Home Assistant integration the following happens: the temperature is set to 23 degrees and the “Heating/Cooling” mode does not turn on.
[image]

after clicking any of these buttons
[image]

Then the air conditioner turns off
[image]

And what happens when you integrate Home Assistant is that it turns on.
[image]

And if you continue to press the buttons on the integration board, the mode changes, but the air conditioner does not turn on until you turn it on from the remote control.
[image]

Please help me figure it out. Below is my configuration, maybe I configured something wrong?

esphome:
  name: s08ir
  platform: ESP8266
  board: esp_wroom_02

wifi:
  ssid: "DomWiFi"
  password: "Dom1WiFi2"
  fast_connect: true

logger:
  level: DEBUG

api:
  password: "s08ir"

ota:
  password: "s08ir"

status_led:
  pin:
    number: GPIO4

remote_receiver:
  id: "s08_receiver"
  pin: 
    number: GPIO5
    inverted: true
  dump: all

remote_transmitter:
  id: "s08_transmitter"
  pin: 
    number: GPIO14
    inverted: false
  carrier_duty_percent: 55%
  

binary_sensor:
  - platform: gpio
    name: "s08_button"
    id: "s08_button"
    pin:
      number: GPIO13
      inverted: true
    internal: true

sensor:
  - platform: homeassistant
    id: current_temperature
    entity_id: sensor.ble_temperature_e72e0172f902
    device_class: temperature
    state_class: measurement

climate:
  - platform: whirlpool
    name: "Кондиционер спальня"
    model: DG11J1-91
    id: "konditsioner_spalnya"
    sensor: current_temperature
    receiver_id: "s08_receiver"
    transmitter_id: "s08_transmitter"
    visual:
      min_temperature: 16 °C
      max_temperature: 30 °C
      temperature_step: 1 °C

### Which version of ESPHome has the issue?

2023.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.3

### What platform are you using?

ESP8266

### Board

Universal IR remote control Smart9 S08 based on esp8266

### Component causing the issue

Climate Components / IR Remote Climate / whirlpool Climate / model DG11J1-91  

### Example YAML snippet

```yaml
esphome:
name: s08ir
platform: ESP8266
board: esp_wroom_02

wifi:
ssid: "DomWiFi"
password: "Dom1WiFi2"
fast_connect: true

logger:
level: DEBUG

api:
password: "s08ir"

ota:
password: "s08ir"

status_led:
pin:
number: GPIO4

remote_receiver:
id: "s08_receiver"
pin:
number: GPIO5
inverted: true
dump: all

remote_transmitter:
id: "s08_transmitter"
pin:
number: GPIO14
inverted: false
carrier_duty_percent: 55%

binary_sensor:

platform: gpio
name: "s08_button"
id: "s08_button"
pin:
number: GPIO13
inverted: true
internal: true
sensor:

platform: homeassistant
id: current_temperature
entity_id: sensor.ble_temperature_e72e0172f902
device_class: temperature
state_class: measurement
climate:

platform: whirlpool
name: "Кондиционер спальня"
model: DG11J1-91
id: "konditsioner_spalnya"
sensor: current_temperature
receiver_id: "s08_receiver"
transmitter_id: "s08_transmitter"
visual:
min_temperature: 16 °C
max_temperature: 30 °C
temperature_step: 1 °C
```

### Additional information

I use this information and this device to control my air conditioner as described in the link below:
https://github.com/esphome/issues/issues/4650#

TokarevSergey, 2023-10-21:
Please support.  @prokudin07 Thank you.

prokudin07, 2023-10-21:
I have the same problem

TokarevSergey, 2023-10-21:
@prokudin07  Скажите пожалуйста вы тоже ипользуете Universal IR remote control Smart9 S08 based on esp8266?
Ели да то у меня ещё один вопрос к вам.
У вас тоже Smart9 S08 based on esp8266

prokudin07, 2023-10-21:
Напишите мне в телеграм @prokudinvn