esphome/issues #5015: Fastled is imcompatible with ESP 2023.10.x and devbranch ESP 2023.11.x

nicedevil007, 2023-10-21:
### The problem

Compilation fails since 2023.10 with this error:

**short:**
```python
function 'void esphome::captive_portal::CaptivePortal::start()':
src/esphome/components/captive_portal/captive_portal.cpp:51:49: error: call of overloaded 'IPAddress(esphome::network::IPAddress&)' is ambiguous
   this->dns_server_->start(53, "*", IPAddress(ip));
```

**complete:**
```python
INFO ESPHome 2023.10.1
INFO Reading configuration /config/esphome/esp-flur.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Generating C++ source...
INFO Compiling app...
Processing esp-flur (board: d1_mini; framework: arduino; platform: platformio/espressif8266@2.6.3)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Installing fastled/FastLED @ 3.3.2
INFO Installing fastled/FastLED @ 3.3.2
Unpacking  [####################################]  100%
Library Manager: FastLED@3.3.2 has been installed!
INFO FastLED@3.3.2 has been installed!
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.4
|-- FastLED @ 3.3.2
Compiling .pioenvs/esp-flur/lib58d/FastLED/FastLED.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/bitswap.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/colorpalettes.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/colorutils.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/hsv2rgb.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/lib8tion.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/noise.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/platforms.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/power_mgt.cpp.o
Compiling .pioenvs/esp-flur/lib58d/FastLED/wiring.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/adc/adc_sensor.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/esp8266/core.cpp.o
Compiling .pioenvs/esp-flur/src/esphome/components/esp8266/gpio.cpp.o
src/esphome/components/captive_portal/captive_portal.cpp: In member function 'void esphome::captive_portal::CaptivePortal::start()':
src/esphome/components/captive_portal/captive_portal.cpp:51:49: error: call of overloaded 'IPAddress(esphome::network::IPAddress&)' is ambiguous
   this->dns_server_->start(53, "*", IPAddress(ip));
                                                 ^
src/esphome/components/captive_portal/captive_portal.cpp:51:49: note: candidates are:
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/Udp.h:39:0,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/libraries/ESP8266WiFi/src/WiFiUdp.h:25,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/libraries/DNSServer/src/DNSServer.h:3,
                 from src/esphome/components/captive_portal/captive_portal.h:5,
                 from src/esphome/components/captive_portal/captive_portal.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/IPAddress.h:163:9: note: IPAddress::IPAddress(const ip4_addr&)
         IPAddress(const ipv4_addr& fw_addr)   { setV4(); v4() = fw_addr.addr; }
         ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/IPAddress.h:79:9: note: IPAddress::IPAddress(const IPAddress&)
         IPAddress(const IPAddress& from);
         ^
Compiling .pioenvs/esp-flur/src/esphome/components/esp8266/preferences.cpp.o
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/esp-flur/src/esphome/components/captive_portal/captive_portal.cpp.o] Error 1
========================== [FAILED] Took 5.14 seconds ==========================
```

### Which version of ESPHome has the issue?

2023.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.3

### What platform are you using?

ESP8266

### Board

WemosD1 Mini

### Component causing the issue

Fastled

### Example YAML snippet

```yaml
Here is my config:

esphome:
  name: esp-flur

esp8266:
  board: d1_mini
  framework:
    version: 2.7.4

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 1min
  encryption:
    key: "#####"

ota:
  password: "#####"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-Flur"
    password: "#####"

captive_portal:
    
light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO3
    num_leds: 24
    rgb_order: BRG
    name: "Schuhschrank LED Flur"

sensor:
  - platform: adc
    pin: A0
    name: "Helligkeit Flur"
    update_interval: 15s
    filters:
      - multiply: 3.3
```

### Anything in the logs that might be useful for us?

```txt
See above
```

### Additional information

My hotfix to use 2023.10 is to commentout all parts that belong to fastled, but then no LEDs ofc.

nicedevil007, 2023-10-21:
An other part I saw during the compilation in the web log output is that ESPhome is using **Fastled 3.3.2** instead of the lasted Fastled release **3.6**? Maybe the issue is already fixed?

https://github.com/FastLED/FastLED/releases

ssieb, 2023-10-21:
This issue is invalid because you're overriding the platform.  What is the fastled problem?

nicedevil007, 2023-10-21:
But I have to overwrite it because the documentation of ESPhome is telling me this :/

ssieb, 2023-10-21:
Try without the platform or use neopixelbus instead.

nicedevil007, 2023-10-21:
> Try without the platform or use neopixelbus instead.

works like a charm! no framework rewrite anymore, thank you very much

Bekkie, 2023-10-22:
Please ropen this issue, since the issue is not fixed. 
Please either fix this error in ESPHome or update the documentation
"https://esphome.io/components/light/fastled.html" stating that from 2023.10.x Fastled is not supported anymore.

ssieb, 2023-10-22:
You should really use the rmt one instead anyway.

uneart, 2023-10-29:
@nicedevil007 Short question: How did you ultimately solve this? Did you use RMT, NeoPixelBus or something else?

nicedevil007, 2023-10-30:
> @nicedevil007 Short question: How did you ultimately solve this? Did you use RMT, NeoPixelBus or something else?

I'm using neopixelbus now -> https://github.com/esphome/issues/issues/5015#issuecomment-1773840902

closed 2023-10-31 by jesserockz