esphome/issues #5016: Canbus.send causes compile time error if can_id not provided.

skykingjwc, 2023-10-21:
### The problem

If canbus.send is called without specifying a can_id a compile time error occurs.  According do the docs, if a single canbus is defined then canbus.send should use the id of the single canbus unless overridden.  This config variable is marked as optional in the docs so it should not be necessary if only one canbus is defined

### Which version of ESPHome has the issue?

23.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

canbus

### Example YAML snippet

```yaml
#does not work 
switch:
  - platform: output
    output: myoutput
    id:  myid
    name:  myname
    on_turn_on:
       - canbus.send: "mudOn"

#works
switch:
  - platform: output
    output: myoutput
    id:  myid
    name:  myname
    on_turn_on:
       - canbus.send:
          data: "test"
          can_id: 17  #<--- should not need this
```

### Anything in the logs that might be useful for us?

```txt
Error at compile
File "/esphome/esphome/components/canbus/__init__.py", line 20, in validate_id
    can_id = config[CONF_CAN_ID]
KeyError: 'can_id'
```

ssieb, 2023-10-21:
That looks like an error in the component.  I don't see how that could be optional, so the short form is not possible.  What would it mean to send without an id?

skykingjwc, 2023-10-22:
It was 100% working without can can_id in the sedn action, before the update.  
To answer your question, in the absence of a can_id in the canbus.send action it automatically uses the can_id defined in the canbus component, as long as there is only one canbus component defined.    The docs say can can_in in the send action overrides the can_id of the bus, and if not defined it will use the can_id of the bus.

ssieb, 2023-10-22:
It definitely would have been more helpful if you had included the canbus parts in the yaml example.

ssieb, 2023-10-22:
It looks like this has never worked properly.

ssieb, 2023-10-22:
Please test the PR.

closed 2023-10-23 by jesserockz