esphome/issues #5018: require_framework_version does not support this platform configuration

davewatson91, 2023-10-21:
### The problem

I have an ESPHome device configuration, which is similar to many others I use, however this one is having troubles with a specific require. 

The error is: 
  `Internal Error: require_framework_version does not support this platform configuration
                  platform: bk72xx
                  framework: arduino`

It appears to be caused by the http_request:
  `http_request:
    useragent: esphome/device
    timeout: 10s`

### Which version of ESPHome has the issue?

2023.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.3

### What platform are you using?

BK72XX

### Board

generic-bk7231n-qfn32-tuya

### Component causing the issue

http_request

### Example YAML snippet

```yaml
http_request:
    useragent: esphome/device
    timeout: 10s
```

ssieb, 2023-10-21:
There is no http request support for libretuya devices yet.
Duplicate of https://github.com/esphome/issues/issues/4997

closed 2023-10-21 by ssieb

davewatson91, 2023-10-21:
Thanks @ssieb! Apologies for the duplicate.