esphome/issues #5020: RTL8710BN OTA failures

Weissnix4711, 2023-10-22:
### The problem

OTA fails after uploading.

Works fine once immediately after the device is restarted, but subsequent updates will not work unless the device is disconnected from power entirely.

### Which version of ESPHome has the issue?

2023.10.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

RTL87XX

### Board

generic-rtl8710bn-2mb-788k

### Example YAML snippet

```yaml
substitutions:
  friendly_name: "Teckin Plug 1"
  device_name: "teckin-plug-1"

rtl87xx:
  board: generic-rtl8710bn-2mb-788k

esphome:
  name: "${device_name}"
  friendly_name: "${friendly_name}"

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  password: !secret ota_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name}"
    password: !secret wifi_ap_password

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.10.1
INFO Reading configuration /config/esphome/teckin-plug-1.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing teckin-plug-1 (board: generic-rtl8710bn-2mb-788k; framework: arduino; platform: libretiny @ 1.4.1)
--------------------------------------------------------------------------------
Warning! Non-Git installations are NOT SUPPORTED.
HARDWARE: RTL8710BN 125MHz, 256KB RAM, 788KB Flash
 - framework-arduino-api @ 2022.8.24+sha.237b10a 
 - framework-realtek-amb1 @ 0.0.0+v2022.06.21.sha.c4e44ef 
 - library-flashdb @ 1.2.0+sha.d5c892f 
 - library-freertos @ 8.1.2+sha.776ae6c 
 - library-freertos-port @ 2023.5.23+sha.a917d93 
 - library-lwip @ 2.1.3-amb1+sha.6297b80 
 - library-printf @ 6.1.0+sha.28a79bd 
 - library-uf2ota @ 5.0.0+sha.f955412 
PLATFORM VERSIONS:
 - libretiny @ 1.4.1
 - ltchiptool @ 4.6.0
CUSTOM OPTIONS:
 - fw_name = esphome
 - fw_version = 2023.10.1
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 1.1.0
|-- noise-c @ 0.1.4
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/libretiny/core.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/libretiny/gpio_arduino.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/libretiny/lt_component.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/libretiny/preferences.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/ota/ota_backend_arduino_libretiny.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/socket/lwip_sockets_impl.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/socket/socket.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/web_server_base/web_server_base.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/wifi/wifi_component.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/wifi/wifi_component_libretiny.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/components/wifi/wifi_component_pico_w.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/application.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/component.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/component_iterator.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/controller.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/entity_base.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/helpers.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/log.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/scheduler.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/string_ref.cpp.o
Compiling .pioenvs/teckin-plug-1/src/esphome/core/util.cpp.o
Compiling .pioenvs/teckin-plug-1/src/main.cpp.o
Linking .pioenvs/teckin-plug-1/raw_firmware.elf
|-- Image 1: raw_firmware.ota1.elf
|-- Image 2: raw_firmware.ota2.elf
|   |-- image_ota1.0x00B000.bin
|   |   |-- image_ota1.ram_2.r.bin
|   |   |-- image_ota1.xip_image2.bin
|   |   |-- image_ota1.rdp.bin
|   |-- image_ota2.0x0D0000.bin
|   |   |-- image_ota2.ram_2.r.bin
|   |   |-- image_ota2.xip_image2.bin
|   |   |-- image_ota2.rdp.bin
RAM:   [==        ]  20.3% (used 53100 bytes from 262144 bytes)
Flash: [======    ]  60.0% (used 484156 bytes from 806912 bytes)
Building UF2 OTA image
|-- esphome_2023.10.1_generic-rtl8710bn-2mb-788k_rtl8710bn_lt1.4.1.uf2
|-- firmware.uf2
|-- firmware.bin
========================= [SUCCESS] Took 82.32 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of teckin-plug-1.local
INFO  -> 192.168.1.147
INFO Uploading /data/build/teckin-plug-1/.pioenvs/teckin-plug-1/firmware.uf2 (969216 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
```

ssieb, 2023-10-22:
Have you tried watching the device logs while flashing to see what's happening?

yaminahacs, 2023-11-01:
It looks like the update is successful, it is only the booting issue after updates. Just power cycle the device, I also get those errors sometimes on BK devices

closed 2024-05-13 by Weissnix4711