esphome/issues #5021: Current Based Cover' - Malfunction detected during opening. Current flow detected in close circuit

Brembo427, 2023-10-22:
### The problem

Hey guys just trying to build a simple low voltage smart roller shutter controller using an athom. relay-board-x2 with an INA219 I2C Digital current monitor. have written up a basic current based cover into the YAML. 

when I go to open or close the cover it will run for about a second then stop and throw Malfunction detected during opening. Current flow detected in close circuit, in the logs. every time without fail.

regardless of I do inside the YAML in terms of current limits, times or delays it does the exact same thing. 

Its running a basic 2 wire 12-volt dc motor which drives the roller shutter. if I manually force relays on, I can simulate opening and closing manually. 

please help?

### Which version of ESPHome has the issue?

2023.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10

### What platform are you using?

ESP8266

### Board

athom relay board x2

### Component causing the issue

i2c

### Example YAML snippet

```yaml
substitutions:

  name: "shutter"
packages:
  athom.relay-board-x2: github://athom-tech/athom-configs/athom-relay-board-x2.yaml
 
esphome:
  name: ${name}
  name_add_mac_suffix: false

api:
  encryption:
    key: i9JEJ7IOJFhSyGWdJ57iIDu86AeZ8tWlJ2rGkJUqLDs=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

i2c:
  sda: GPIO12  
  scl: GPIO13

sensor:
   - platform: ina219
     address: 0x40
     shunt_resistance: 0.1 ohm
     current:
      id: "current"
      name: "Current"
     power:
      name: "Power"
     bus_voltage:
      name: "Bus Voltage"
     shunt_voltage:
      name: "Shunt Voltage"
     max_voltage: 32.0V
     max_current: 3.2A
     update_interval: 0.5s

cover:
  - platform: current_based
    name: "Current Based Cover"

    open_sensor: current
    open_moving_current_threshold: 1.7
    open_obstacle_current_threshold: 3.0
    open_duration: 24s
    open_action:
      - switch.turn_on: relay1
      - switch.turn_off: relay2

    close_sensor: current
    close_moving_current_threshold: 1.0
    close_obstacle_current_threshold: 2.5
    close_duration: 24s
    close_action:
      - switch.turn_on: relay2
      - switch.turn_off: relay1

    stop_action:
      - switch.turn_off: relay1
      - switch.turn_off: relay2

    obstacle_rollback: 30%
    start_sensing_delay: 1.0s
```

### Anything in the logs that might be useful for us?

```txt
[I][current_based.cover:069]: 'Current Based Cover' - Malfunction detected during opening. Current flow detected in close circuit

[I][current_based.cover:090]: 'Current Based Cover' - Malfunction detected during closing. Current flow detected in open circuit
```

rogerbusser, 2023-10-22:
Hi,

I actually got the same errors ;-)
The problem is using the same current sensor for opening and closing. I used a similar setup using an H-bridge and a DC motor. What helped for me was disabling the malfunction_detection option of the component.

I found another issue with this component, the max_duration is not working, the condition in the code is never reached, so the motor doesn't stop after the max_duration set point.
I'm trying to setup a development environment to test the change... but somehow I run in all kind of issues. Can someone point me to  guide?

Brembo427, 2023-10-23:
Seems a bit silly to have an open and close sensor. Is that even possible on the same board?? I would think one sensor per motor. in my case it monitors current feeding the 2 relays wired in a interlock / flip flop config.  will try the malfunction defect and post an update.