esphome/issues #5023: Uptime Should Return a Datetime Rather than a Counter

barndawgie, 2023-10-22:
### The problem

Currently, the ESPHome Uptime integration returns a monotonically increasing string. This is a pain for usage in Home Assistant and drives higher storage needs.

As implemented, this would intuitively be a `duration` device class, but [per the documentation](https://developers.home-assistant.io/docs/core/entity/sensor#available-device-classes) durations "Should not update only due to time passing."

Instead, uptime should match the [Home Assistant uptime sensor](https://www.home-assistant.io/integrations/uptime/) which provides the datetime of last boot.

### Which version of ESPHome has the issue?

2023.10.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

m5Stack ATOM

### Component causing the issue

uptime

### Example YAML snippet

```yaml
sensor:
- platform: uptime
  name: ESPHome Uptime
  device_class: duration
  unit_of_measurement: s
  icon: mdi:timer-check
  entity_category: diagnostic
  disabled_by_default: false
  force_update: false
  accuracy_decimals: 0
  state_class: total_increasing
  update_interval: 60s
```

ssieb, 2023-10-22:
There's no way to pass a datetime from esphome to HA other than using a `text_sensor`.  You could do that.
But the point of the uptime sensor updating regularly is to indicate that the device is actually still functioning and how long it has been running.  I don't see how HA would use a datetime value to provide the same capability.

barndawgie, 2023-10-22:
Fair enough - maybe I can create my own sensor based off of this in my HA config....

closed 2023-10-22 by barndawgie