esphome/issues #5026: 2023.10.2 compile issue

hermanops, 2023-10-23:
### The problem

```
|-- Improv @ 1.2.3
Compiling .pioenvs/esp32-bluetooth-proxy-0d77f4/src/esphome/components/api/api_connection.o
Compiling .pioenvs/esp32-bluetooth-proxy-0d77f4/src/esphome/components/api/api_frame_helper.o
Compiling .pioenvs/esp32-bluetooth-proxy-0d77f4/src/esphome/components/api/api_pb2.o
Compiling .pioenvs/esp32-bluetooth-proxy-0d77f4/src/esphome/components/api/api_pb2_service.o
In file included from src/esphome/components/network/util.h:4,
                 from src/esphome/components/api/api_connection.cpp:4:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
Compiling .pioenvs/esp32-bluetooth-proxy-0d77f4/src/esphome/components/api/api_server.o
*** [.pioenvs/esp32-bluetooth-proxy-0d77f4/src/esphome/components/api/api_connection.o] Error 1
In file included from src/esphome/components/network/util.h:4,
                 from src/esphome/components/api/api_server.cpp:4:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
*** [.pioenvs/esp32-bluetooth-proxy-0d77f4/src/esphome/components/api/api_server.o] Error 1
========================= [FAILED] Took 118.55 seconds =========================
```

### Which version of ESPHome has the issue?

2023.10.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP32-IDF

### Board

AZ Delivery ESP32S

### Component causing the issue

bluetooth proxy

### Example YAML snippet

```yaml
trying to update my bluetooth proxy in esphome. Initial install was with Esphome Web
```

KitHubek, 2023-10-23:
the same here. LAN module wt32-eth01 and wifi module esp32

```
Compiling .pioenvs/wt32-eth01-bt-proxy-macaddr/src/esphome/components/api/api_pb2_service.o
In file included from src/esphome/components/network/util.h:4,
                 from src/esphome/components/api/api_connection.cpp:4:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
Compiling .pioenvs/wt32-eth01-bt-proxy-macaddr/src/esphome/components/api/api_server.o
*** [.pioenvs/wt32-eth01-bt-proxy-macaddr/src/esphome/components/api/api_connection.o] Error 1
In file included from src/esphome/components/network/util.h:4,
                 from src/esphome/components/api/api_server.cpp:4:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
*** [.pioenvs/wt32-eth01-bt-proxy-macaddr/src/esphome/components/api/api_server.o] Error 1
========================= [FAILED] Took 28.92 seconds =========================
```

MorganMLGman, 2023-10-23:
Same here on Atom Echo.

```
Compiling .pioenvs/atom-echo-1/src/esphome/components/esp32/preferences.o
In file included from src/esphome/components/network/util.h:4,
                 from src/esphome/components/api/api_connection.cpp:4:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
Compiling .pioenvs/atom-echo-1/src/esphome/components/esp32_rmt_led_strip/led_strip.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/esp_adf/esp_adf.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/homeassistant/time/homeassistant_time.o
In file included from src/esphome/components/network/util.h:4,
                 from src/esphome/components/api/api_server.cpp:4:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
Compiling .pioenvs/atom-echo-1/src/esphome/components/i2s_audio/i2s_audio.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/i2s_audio/microphone/i2s_audio_microphone.o
In file included from src/esphome/components/wifi/wifi_component.h:7,
                 from src/esphome/components/captive_portal/captive_portal.cpp:4:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
Compiling .pioenvs/atom-echo-1/src/esphome/components/i2s_audio/speaker/i2s_audio_speaker.o
Compiling .pioenvs/atom-echo-1/src/esphome/components/improv_base/improv_base.o
*** [.pioenvs/atom-echo-1/src/esphome/components/api/api_connection.o] Error 1
*** [.pioenvs/atom-echo-1/src/esphome/components/captive_portal/captive_portal.o] Error 1
*** [.pioenvs/atom-echo-1/src/esphome/components/api/api_server.o] Error 1
In file included from src/esphome/components/network/util.h:4,
                 from src/esphome/components/improv_base/improv_base.cpp:3:
src/esphome/components/network/ip_address.h: In constructor 'esphome::network::IPAddress::IPAddress(ip4_addr_t*)':
src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'
     ip_addr_.type = IPADDR_TYPE_V4;
              ^~~~
*** [.pioenvs/atom-echo-1/src/esphome/components/improv_base/improv_base.o] Error 1
========================== [FAILED] Took 6.83 seconds ==========================
```

drifterdave, 2023-10-23:
I have the same issue, looks like they are pushing a fix. Waiting for CI tests to complete. [PR #5595](https://github.com/esphome/esphome/pull/5595)

closed 2023-10-23 by jesserockz

vb7445, 2023-10-24:
I have the same issue with ESP-IDF on ESP32 DevKits. Will no longer compile. Error messages complaining about 'type'. 

`src/esphome/components/network/ip_address.h:40:14: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'type'`

ssieb, 2023-10-24:
@vb7445 the fix is in the release that's in progress right now.

vb7445, 2023-10-24:
Whatever the issue was it's fixed in 2023.10.3. Just installed it and now my builds are compiling again. Thanks for the quick fix.

KitHubek, 2023-10-24:
For me update to 2023.10.3 solved the problem. thanks