esphome/issues #5027: Error while compiling after update esphome

proasnet, 2023-10-24:
### The problem

I have had a functioning weather station for about a year, which I always update to a new version of esphome without any problems.
I wanted to do an update today, but the compilation shows the following error.
Only this device has problem with update, other without problems.

### Which version of ESPHome has the issue?

2023.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP32

### Board

LaskaKit Meteo Mini 

### Example YAML snippet

```yaml
substitutions:
  name: weatherstation
  friendly_name: "Weather Station"
  device_description: "Weather Station"

esphome:
  name: ${name}
  comment: ${device_description}

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

api:

ota:
 
web_server:
  port: 80
  auth:
    username: !secret web_server_username
    password: !secret web_server_password

wifi:
  ssid: !secret wifi_ssid_2
  password: !secret wifi_password

  ap:
    ssid: "Weatherstation Fallback Hotspot"
    password: "Jirka8011"

captive_portal:

time:
  - platform: sntp
    id: sntp_time
    servers: "pool.ntp.org"
[... 295 lines cut ...]
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/weatherstation/src/esphome/components/adc/adc_sensor.cpp.o
In file included from src/esphome/components/adc/adc_sensor.h:3,
                 from src/esphome/components/adc/adc_sensor.cpp:1:
src/esphome/core/component.h:3:10: fatal error: string: No such file or directory
 #include <string>
          ^~~~~~~~
compilation terminated.
Compiling .pioenvs/weatherstation/src/esphome/components/api/api_connection.cpp.o
*** [.pioenvs/weatherstation/src/esphome/components/adc/adc_sensor.cpp.o] Error 1
In file included from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
 #include <cstdint>
          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/weatherstation/src/esphome/components/api/api_connection.cpp.o] Error 1
========================= [FAILED] Took 10.56 seconds =========================
```

proasnet, 2023-10-25:
Additional info:
Before try update, esp node working normally. Log:
```
INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/weatherstation.yaml...
INFO Detected timezone 'Europe/Prague'
INFO Starting log output from weatherstation.local using esphome API
INFO Successfully connected to weatherstation.local
[06:41:56][I][app:102]: ESPHome version 2023.9.3 compiled on Oct  5 2023, 19:30:00
[06:41:56][C][wifi:546]: WiFi:
[06:41:56][C][wifi:382]:   Local MAC: A0:76:4E:51:A7:90
[06:41:56][C][wifi:383]:   SSID: [redacted]
[06:41:56][C][wifi:384]:   IP Address: 192.168.0.99
[06:41:56][C][wifi:386]:   BSSID: [redacted]
[06:41:56][C][wifi:387]:   Hostname: 'weatherstation'
[06:41:56][C][wifi:389]:   Signal strength: -63 dB ▂▄▆█
[06:41:56][C][wifi:393]:   Channel: 1
[06:41:56][C][wifi:394]:   Subnet: 255.255.255.0
[06:41:56][C][wifi:395]:   Gateway: 192.168.0.254
[06:41:56][C][wifi:396]:   DNS1: 192.168.0.254
[06:41:56][C][wifi:397]:   DNS2: 195.146.128.62
[06:41:56][D][sensor:094]: 'ADC': Sending state 1.09300 V with 2 decimals of accuracy
[06:41:56][D][resistance:039]: 'Resistance Sensor' - Resistance 15847.8Ω
[06:41:56][C][logger:357]: Logger:
[06:41:56][C][logger:358]:   Level: DEBUG
[06:41:56][C][logger:359]:   Log Baud Rate: 115200
[06:41:56][C][logger:361]:   Hardware UART: UART0
[06:41:56][C][i2c.arduino:053]: I2C Bus:
[06:41:56][C][i2c.arduino:054]:   SDA Pin: GPIO19
[06:41:56][C][i2c.arduino:055]:   SCL Pin: GPIO18
[06:41:56][C][i2c.arduino:056]:   Frequency: 50000 Hz
[06:41:56][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[06:41:56][I][i2c.arduino:069]: Results from i2c bus scan:
[06:41:56][I][i2c.arduino:075]: Found i2c device at address 0x77
[06:41:56][D][sensor:094]: 'Weather Station wind speed average': Sending state nan m/s with 2 decimals of accuracy
[06:41:56][D][sensor:094]: 'Weather Station wind speed average (km/h)': Sending state nan km/h with 2 decimals of accuracy
[06:41:56][C][template.sensor:022]: Template Sensor 'Weather Station wind heading'
[06:41:56][C][template.sensor:022]:   State Class: ''
[06:41:56][C][template.sensor:022]:   Unit of Measurement: '°'
[06:41:56][C][template.sensor:022]:   Accuracy Decimals: 1
[06:41:56][C][template.sensor:023]:   Update Interval: 60.0s
[06:41:56][C][template.text_sensor:020]: Template Sensor 'Weather Station wind cardinal direction'
[06:41:56][C][pulse_meter:082]: Pulse Meter 'Weather Station wind speed'
[06:41:56][C][pulse_meter:082]:   State Class: 'measurement'
[06:41:56][C][pulse_meter:082]:   Unit of Measurement: 'm/s'
[06:41:56][C][pulse_meter:082]:   Accuracy Decimals: 2
[06:41:56][C][pulse_meter:082]:   Icon: 'mdi:weather-windy'
[06:41:56][C][pulse_meter:083]:   Pin: GPIO6
[06:41:56][C][pulse_meter:085]:   Filtering rising edges less than 100 µs apart
[06:41:56][C][pulse_meter:089]:   Assuming 0 pulses/min after not receiving a pulse for 5s
[06:41:56][C][copy.sensor:015]: Copy Sensor 'Weather Station wind speed average'
[06:41:56][C][copy.sensor:015]:   State Class: 'measurement'
[06:41:56][C][copy.sensor:015]:   Unit of Measurement: 'm/s'
[06:41:56][C][copy.sensor:015]:   Accuracy Decimals: 2
[06:41:56][C][copy.sensor:015]:   Icon: 'mdi:weather-windy'
[06:41:56][C][copy.sensor:015]: Copy Sensor 'Weather Station wind speed (km/h)'
[06:41:56][C][copy.sensor:015]:   State Class: 'measurement'
[06:41:56][C][copy.sensor:015]:   Unit of Measurement: 'km/h'
[06:41:56][C][copy.sensor:015]:   Accuracy Decimals: 2
[06:41:56][C][copy.sensor:015]:   Icon: 'mdi:weather-windy'
[06:41:56][C][copy.sensor:015]: Copy Sensor 'Weather Station wind speed average (km/h)'
[06:41:56][C][copy.sensor:015]:   State Class: 'measurement'
[06:41:56][C][copy.sensor:015]:   Unit of Measurement: 'km/h'
[06:41:56][C][copy.sensor:015]:   Accuracy Decimals: 2
[06:41:56][C][copy.sensor:015]:   Icon: 'mdi:weather-windy'
[06:41:56][C][adc:097]: ADC Sensor 'ADC'
[06:41:56][C][adc:097]:   Device Class: 'voltage'
[06:41:56][C][adc:097]:   State Class: 'measurement'
[06:41:56][C][adc:097]:   Unit of Measurement: 'V'
[06:41:56][C][adc:097]:   Accuracy Decimals: 2
[... 50 lines cut ...]
[06:41:56][C][bh1750.sensor:118]:   Accuracy Decimals: 1
[06:41:56][C][bh1750.sensor:119]:   Address: 0x23
[06:41:56][E][bh1750.sensor:121]: Communication with BH1750 failed!
[06:41:56][C][bh1750.sensor:124]:   Update Interval: 60.0s
[06:41:56][C][adc:097]: ADC Sensor 'Weather Station Outside Battery'
[06:41:56][C][adc:097]:   Device Class: 'voltage'
[06:41:56][C][adc:097]:   State Class: 'measurement'
[06:41:56][C][adc:097]:   Unit of Measurement: 'V'
[06:41:56][C][adc:097]:   Accuracy Decimals: 2
[06:41:56][C][adc:107]:   Pin: GPIO1
[06:41:56][C][adc:122]:  Attenuation: 11db
[06:41:56][C][adc:142]:   Update Interval: 60.0s
[06:41:57][C][restart.button:017]: Restart Button 'ESP_Restart'
[06:41:57][C][resistance:010]: Resistance Sensor 'Resistance Sensor'
[06:41:57][C][resistance:010]:   State Class: 'measurement'
[06:41:57][C][resistance:010]:   Unit of Measurement: 'Ω'
[06:41:57][C][resistance:010]:   Accuracy Decimals: 2
[06:41:57][C][resistance:010]:   Icon: 'mdi:flash'
[06:41:57][C][resistance:011]:   Configuration: DOWNSTREAM
[06:41:57][C][resistance:012]:   Resistor: 32000.00Ω
[06:41:57][C][resistance:013]:   Reference Voltage: 3.3V
[06:41:57][C][captive_portal:088]: Captive Portal:
[06:41:57][C][web_server:161]: Web Server:
[06:41:57][C][web_server:162]:   Address: weatherstation.local:80
```

proasnet, 2023-10-25:
Not working, error while compile
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

Not working, error while compile
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf

Working, compiling, but only for test. It is not suitable for the device itself
esp32:
  board: esp-wrover-kit
  framework:
    type: esp-idf

proasnet, 2023-10-26:
Downgrade ESPHome to 2023.9.3 from backup fix problem. For working esp32-c3-devkitm-1 no update to never version of ESPHome.

DavidWAbrahams, 2023-10-28:
Same issue here while trying to set up an esp32-c3fh4 from the Home Assistant addon.

For anyone who doesn't have a convenient backup, the "ESPHome (dev)" home assistant addon works correctly.

Thank you for reporting the issue. You saved this noob a lot of pain.

JeeCee1, 2023-10-30:
Same here, thanks for posting.

closed 2024-03-13 by proasnet