esphome/issues #5029: Impossible to upload TFT file to NSPanel

DorjeDorf, 2023-10-24:
### The problem

I believe this has been worked over recently, but looking at the release log, I was sure it got fixed.

The problem I keep hitting is my inability to upload a tft file to new NSPanels (I bought 4 new ones recently). ESPHome flashes just fine, but the tft panel produces the following log:

`17:17:17[W][nextion:077] Nextion is not connected! 
17:17:17[W][component:214] Component nextion.display took a long time for an operation (0.11 s).
17:17:17[W][component:215] Components should block for at most 20-30ms.
17:17:17[E][uart:015] Reading from UART timed out at byte 0!

17:17:17[W][nextion:077] Nextion is not connected! 
17:17:17[W][component:214] Component nextion.display took a long time for an operation (0.11 s).
17:17:17[W][component:215] Components should block for at most 20-30ms.
17:17:18[E][uart:015] Reading from UART timed out at byte 0!

17:17:18[W][nextion:077] Nextion is not connected! 
17:17:18[W][component:214] Component nextion.display took a long time for an operation (0.11 s).
17:17:18[W][component:215] Components should block for at most 20-30ms.
17:17:18[E][uart:015] Reading from UART timed out at byte 0!`

I have an older NSPanel that flashes and updates just fine each time. I tried using the config for that one along with the TFT that works for it - no luck. 

I have tried Tasmota - it will work fine.
I tried NS Panel Manager - that one also flashes just fine (funny enough, at one point during testing, their upload got interrupted and I had to reflash the original firmware and reflash the esp and tft using the eWeLink app). 

It's just ESPHome at this point - and that's what I want to use. 

This is for the EU version and I've been seeing this same error for weeks now. 

Is there anything more from me to be able to help fix this? 

### Which version of ESPHome has the issue?

2023.09.2-3, 2023.10.1-3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.09-2023.10

### What platform are you using?

ESP32

### Board

esp32dev

### Component causing the issue

nextion

DorjeDorf, 2023-10-28:
I did a lot of testing and am not particularly looking forward to retesting this by reflashing stock, but in the end, I believe this is what helped me with to finally be able to flash my new gen NSPanel. 

https://github.com/esphome/esphome/pull/2956#issuecomment-1197793599

The Exit Reparse button is not present - not in the panel itself when you go to it's UI or in HA, but that command helped. 
To be clear, this does not help with the older gen nspanel, which used to upload TFTs without a hitch, but after recent changes, no longer works.

edwardtfn, 2023-11-04:
I've added an "After boot check" for this case where it reports `Nextion is not connected!`:

```yaml
esphome:
  name: ${device_name}
  min_version: 2023.5.0
  platformio_options:
    build_flags:
      - -Wno-missing-field-initializers
  on_boot:
    priority: 200.0
    then:
      - logger.log: After boot check-up
      - wait_until:
          condition:
            - api.connected:
          timeout: 60s
      - wait_until:
          condition:
            - lambda: !lambda return disp1->is_setup();
          timeout: 20s
      - script.execute: exit_reparse
      - wait_until:
          condition:
            - lambda: !lambda return disp1->is_setup();
          timeout: 20s
      - lambda: |-
          static const char *const TAG = "on_boot";
          if (not disp1->is_setup()) {
            ESP_LOGE(TAG, "No response from Nextion display");
            ESP_LOGD(TAG, "Turn off Nextion");
            screen_power->turn_off();
            delay(1500);
            ESP_LOGD(TAG, "Turn on Nextion");
            screen_power->turn_on();
          }

display:
  - id: disp1
    platform: nextion
    uart_id: tf_uart

switch:
  - name: Screen Power
    platform: gpio
    id: screen_power
    entity_category: config
    pin:
      number: 4
      inverted: true
    restore_mode: ALWAYS_ON
    internal: true

script:
  - id: exit_reparse
    mode: restart
    then:
      - logger.log: "Exit reparse"
      - uart.write:
          id: tf_uart
          data: "DRAKJHSUYDGBNCJHGJKSHBDN"
      - uart.write:
          id: tf_uart
          data: [0xFF, 0xFF, 0xFF]
```
The full code is here: https://github.com/Blackymas/NSPanel_HA_Blueprint/blob/dev/nspanel_esphome_core.yaml

I still working at this, but the results are pretty good so far and it rarely gest to the display power cycle part.
I have the "exit_reparse" in a separated script temporarily, as I'm waiting for https://github.com/esphome/esphome/pull/5484 to be released in production so I can hopefully replace that code by a more readable call.

closed 2023-11-28 by kbx81