esphome/issues #5030: MQTT Component doesn't honor availability topic configuration

climategadgets, 2023-10-25:
### The problem

### Expected Behavior
Providing `availability.topic` for an MQTT Component makes the component advertise its availability on the given topic.

### Actual Behavior
* Providing `availability.topic` makes ESPHome correctly advertise the availability topic in Home Assistant Discovery Packet
* However, it does not make the component publish its availability on the provided topic
* Yet, availability is still published, but only at `${topic_prefix}/status`.

### Which version of ESPHome has the issue?

2023.2.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

Irrelevant, behavior is traceable with observing MQTT topics

### What platform are you using?

ESP32

### Board

LILYGO TTGO T-Relay

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
mqtt:
  broker: $IP_ADDRESS
  topic_prefix: /esphome/${MAC}
  log_topic: /esphome-log/${MAC}

switch:
  - platform: gpio
    name: "t-relay-3-r3"
    id: R3
    pin: GPIO05
    availability:
      topic: /esphome/${MAC}/switch/t-relay-3-r3/status
```

### Anything in the logs that might be useful for us?

```txt
It looks like the availability topic is not picked up somewhere, the boot message states the following (note just two topics):

[19:58:19][C][mqtt.switch:041]: MQTT Switch 't-relay-3-r3': 
[19:58:19][C][mqtt.switch:042]:   State Topic: '/esphome/${MAC}/switch/t-relay-3-r3/state'
[19:58:19][C][mqtt.switch:042]:   Command Topic: '/esphome/${MAC}/switch/t-relay-3-r3/command'

```
```

### Additional information

Here's the fragment of the Home Assistant Discovery Packet rendered with the configuration above - the availability topic is advertised correctly:

```
{
  "name": "t-relay-3-r3",
  "stat_t": "/esphome/${MAC}/switch/t-relay-3-r3/state",
  "cmd_t": "/esphome/${MAC}/switch/t-relay-3-r3/command",
  "avty_t": "/esphome/${MAC}/switch/t-relay-3-r3/status",
```

kahrendt, 2023-11-04:
I looked into the MQTT code to try to figure this one out. An individual component doesn't ever set up a lastwill message. The MQTT protocol only allows one lastwill topic per connection (see [Stack Exchange](https://arduino.stackexchange.com/questions/60309/mqtt-can-lastwill-be-sent-to-multiple-topics)), so in the case of ESPHome, the MQTT connection uses the globally specified lastwill topic.

When you configure a specific MQTT component's availability topic, all it does is point to the specified topic in the discovery message. I'm not sure of a particular use case for this. 

So, unfortunately, we can not set up individual availability/lastwill topics for each component due to limitations with the MQTT protocol. I will try to make a pull request to update the documentation to explain this better.

climategadgets, 2023-11-04:
Totally understood, I ran into the same problem with my project (which has ALL Logical devices on the same connection). There's a few things to note here:
* Since you can't LWT on multiple topics, you need to fix the discovery packet - see the last snippet in the original report, the topic is advertised incorrectly, HA compatible consumers will get confused by that as well;
* For the same reason, the `availability` configuration keyword needs to be disabled, or its semantics changed - say, you can see if there's just one in the whole configuration and allow it, or disallow it altogether;
* On a tangent, it might work if you still allow multiple `availability` topics, but provide a limited TTL with MQTT 5 connections, coupled with a heartbeat update to that topic with period less than TTL - this way, MQTT 5 aware clients might figure out that absence of the availability message is a "negative message". This is a stretch because it goes beyond the original "availability" specification, but it might just work for clients with persistent connection, and it will work with no changes required with clients that make fresh MQTT connections often.

Can of worms, eh? :)

kahrendt, 2023-11-07:
I've been considering this a bit more, and I think we don't want to remove it altogether. For example, you can add MQTT subscribe components (like this [sensor](https://esphome.io/components/sensor/mqtt_subscribe)), and it would potentially make sense for that specific LWT topic to be set to the source's LWT topic rather than the ESPHome device's. There are probably other use cases as well, so we wouldn't want to potentially break anyone's setup. I think adding a clear warning to the documentation should suffice. What are your thoughts?

I haven't looked into MQTT 5 at all, so I am unable to discuss your last point.

climategadgets, 2023-11-07:
Fair enough.

Either way, it would help tremendously if by default the component advertised its availability topic correctly in the discovery packet. I believe this is what the default packet should look like:

```
{
  "name": "t-relay-3-r3",
  "stat_t": "${TOPIC_PREFIX}/switch/t-relay-3-r3/state",
  "cmd_t": "${TOPIC_PREFIX}/switch/t-relay-3-r3/command",
  "avty_t": "${TOPIC_PREFIX}//status",
```

climategadgets, 2024-03-07:
@kahrendt , GitHub wants to sweep this issue under the rug. Could you please consider fixing the metadata as described in my previous comment before it does?