esphome/issues #5033: Muse Luxe TTS not working after 2023.10.2 or higher

monsieurlatte, 2023-10-25:
### The problem

pretty sure it started with 2023.10.2 but may be earlier.  I have a muse luxe I use for random testing, I haven't used it a bit but now when I send a tts cloud to it it does nothing.  Sometimes it will pop, sometimes it will just reboot.  I know the tts cloud service is working since I can play them via homepods, browser, etc.   The muse luxe is working because I can play audio on it from radio browser.  

### Which version of ESPHome has the issue?

2023.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP32

### Board

esp-wrover-kit

### Component causing the issue

not sure

### Example YAML snippet

```yaml
esphome:
  name: museluxe

substitutions:
  name: "raspiaudio-muse-luxe"
  friendly_name: "RaspiAudio Muse Luxe"

button:
  - platform: restart
    name: "Muse Luxe Restart"

esp32:
  board: esp-wrover-kit
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "redacted"

ota:
  password: "redacted"

wifi:
  power_save_mode: none
  ssid: "redacted"
  password:  !secret wifi_password
  #output_power: 20.5dB

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Museluxe Fallback Hotspot"
    password: "redacted"

i2c:
  sda: GPIO18
  scl: GPIO23

#dashboard_import:
#  package_import_url: github://esphome/media-players/raspiaudio-muse-luxe.yaml@main

captive_portal:

web_server:
  port: 80

improv_serial:

external_components:
  - source: github://pr#3552
    components: [es8388]
    refresh: 0s

i2s_audio:
  i2s_lrclk_pin: GPIO25
  i2s_bclk_pin: GPIO5

microphone:
  - platform: i2s_audio
    adc_type: external
    pdm: false
    id: lux_microphone
    i2s_din_pin: GPIO35

media_player:
  - platform: i2s_audio
    name: ${friendly_name}
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: stereo
    mute_pin:
      number: GPIO21
      inverted: true

es8388:

sensor:
 - platform: adc
   pin: GPIO33
   name: ${name} Battery
   icon: "mdi:battery-outline"
   device_class: voltage
   state_class: measurement
   unit_of_measurement: V
   update_interval: 15s
   accuracy_decimals: 3
   attenuation: 11db
   raw: true
   filters:
    - multiply: 0.00173913 # 2300 -> 4, for attenuation 11db, based on Olivier's code
    - exponential_moving_average:
        alpha: 0.2
        send_every: 2
    - delta: 0.002

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO19
      inverted: true
      mode:
        input: true
        pullup: true
    name: ${friendly_name} Volume Up
    on_click:
      - media_player.volume_up:
  - platform: gpio
    pin:
      number: GPIO32
      inverted: true
      mode:
        input: true
        pullup: true
    name: ${friendly_name} Volume Down
    on_click:
      - media_player.volume_down:
  - platform: gpio
    pin:
      number: GPIO12
      inverted: true
      mode:
        input: true
        pullup: true
    name: ${friendly_name} Play Button
#    on_click:
#      - media_player.toggle:
    on_press:
      - voice_assistant.start:
    on_release:
      - voice_assistant.stop:
        
light:
  - platform: fastled_clockless
    name: ${friendly_name}
    pin: GPIO22
    chipset: SK6812
    num_leds: 1
    rgb_order: grb

voice_assistant:
    microphone: lux_microphone
    speaker: luxe_speaker
    
speaker:
  - platform: i2s_audio
    id: luxe_speaker
    dac_type: external
    i2s_dout_pin: GPIO26
    mode: stereo
```

### Anything in the logs that might be useful for us?

```txt
nothing at all, just doesn't play.
```

s00500, 2023-10-30:
This feels a bit familiar to my issue here https://github.com/esphome/issues/issues/4985

Maybe its related ?

monsieurlatte, 2023-10-30:
oh interesting, it could be, will dig into it.

llluis, 2023-11-20:
Doc says you can't use speaker and media_player together:

> speaker (Optional, [ID](https://esphome.io/guides/configuration-types#config-id)): The [speaker](https://esphome.io/components/speaker/) to use to output the response. Cannot be used with media_player below.

> media_player (Optional, [ID](https://esphome.io/guides/configuration-types#config-id)): The [media_player](https://esphome.io/components/media_player/) to use to output the response. Cannot be used with speaker above.

In my case:
When using only speaker, everything works.
When using media_player, it can't play the RAW file produced by piper / voice_assistant component. 
However, the media_player can play wav and mp3 files without any issue by TTS via service call.

So, I'm trying to find out how to make voice_assistant return via piper generate a wav instead of a raw.

closed 2024-01-25 by monsieurlatte