esphome/issues #5034: Error while compiling after update esphome 2023.10, delay function

erumaithi, 2023-10-26:
### The problem

after updating to esphome 2023.10, there is an Error during compiling.  After many hours of troubleshooting, i discovered after removing the delay line it is fixed, but i lost the status LED on my sensibo sky.  

before the update it compiling with no issues

on_state:
         - light.turn_on: 
             id: led
             transition_length: 0.5s
             brightness: 60%
         - **delay: 0.6s**
         - light.turn_off: 
             id: led
             transition_length: 0.5s

### Which version of ESPHome has the issue?

2023.10

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10

### What platform are you using?

ESP8266

### Board

Sensibo Sky

### Example YAML snippet

```yaml
esphome:
  name: sensibo-sky-sittingroom

esp8266: 
  board: esp_wroom_02

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

   # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "SensiboSky sittingroom"
    password: "12345678"

  manual_ip:
    static_ip: 10.0.1.75
    subnet: 255.255.255.0
    gateway: 10.0.1.1

captive_portal:

# Enable logging
logger:

i2c:
  sda: 02
  scl: 14
  scan: true
  id: bus_a

remote_transmitter:
  pin: GPIO15
  carrier_duty_percent: 50%
  

remote_receiver:
  id: rcr
  pin:
   number: 4
   inverted: true
   mode:
    input: true
    pullup: true

climate:
  - platform: fujitsu_general
    name: "SettingRoom AC"
    supports_heat: false
    sensor: temp
    visual:
      temperature_step: 0.1 °C
    receiver_id: rcr
    on_state:
         - light.turn_on: 
             id: led
             transition_length: 0.5s
             brightness: 60%
         - delay: 0.6s
         - light.turn_off: 
             id: led
             transition_length: 0.5s

light:
  - platform: monochromatic
    name: "Sensibo LED"
    id: led
    output: light_output

output:
  - id: light_output
    platform: esp8266_pwm
    pin: 12
```

### Anything in the logs that might be useful for us?

```txt
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/pstl/glue_algorithm_defs.h:13,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/algorithm:74,
                 from src/esphome/core/optional.h:19,
                 from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional: In instantiation of 'struct std::_Bind_helper<false, void (esphome::Action<esphome::climate::Climate&>::*)(esphome::climate::Climate&), esphome::DelayAction<esphome::climate::Climate&>*, esphome::climate::Climate&>':
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:785:5:   required by substitution of 'template<class _Func, class ... _BoundArgs> typename std::_Bind_helper<std::__is_socketlike<_Func>::value, _Func, _BoundArgs ...>::type std::bind(_Func&&, _BoundArgs&& ...) [with _Func = void (esphome::Action<esphome::climate::Climate&>::*)(esphome::climate::Climate&); _BoundArgs = {esphome::DelayAction<esphome::climate::Climate&>*, esphome::climate::Climate&}]'
src/esphome/core/base_automation.h:135:23:   required from 'void esphome::DelayAction<Ts>::play_complex(Ts ...) [with Ts = {esphome::climate::Climate&}]'
src/esphome/core/base_automation.h:134:8:   required from here
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:768:71: error: invalid abstract parameter type 'esphome::climate::Climate'
  768 |       typedef _Bind<__func_type(typename decay<_BoundArgs>::type...)> type;
      |                                                                       ^~~~
In file included from src/esphome/core/controller.h:29,
                 from src/esphome/components/api/api_server.h:8,
                 from src/esphome/components/api/api_connection.h:6,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/components/climate/climate.h:161:7: note:   because the following virtual functions are pure within 'esphome::climate::Climate':
  161 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:256:25: note:     'virtual esphome::climate::ClimateTraits esphome::climate::Climate::traits()'
  256 |   virtual ClimateTraits traits() = 0;
      |                         ^~~~~~
src/esphome/components/climate/climate.h:266:16: note:     'virtual void esphome::climate::Climate::control(const esphome::climate::ClimateCall&)'
  266 |   virtual void control(const ClimateCall &call) = 0;
      |                ^~~~~~~
In file included from src/esphome.h:111,
                 from src/main.cpp:3:
src/esphome/core/base_automation.h: In instantiation of 'void esphome::DelayAction<Ts>::play_complex(Ts ...) [with Ts = {esphome::climate::Climate&}]':
src/esphome/core/base_automation.h:134:8:   required from here
src/esphome/core/base_automation.h:135:23: error: no matching function for call to 'bind(void (esphome::Action<esphome::climate::Climate&>::*)(esphome::climate::Climate&), esphome::DelayAction<esphome::climate::Climate&>*, esphome::climate::Climate&)'
  135 |     auto f = std::bind(&DelayAction<Ts...>::play_next_, this, x...);
      |              ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/pstl/glue_algorithm_defs.h:13,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/algorithm:74,
                 from src/esphome/core/optional.h:19,
                 from src/esphome/components/socket/socket.h:5,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:785:5: note: candidate: 'template<class _Func, class ... _BoundArgs> typename std::_Bind_helper<std::__is_socketlike<_Func>::value, _Func, _BoundArgs ...>::type std::bind(_Func&&, _BoundArgs&& ...)'
  785 |     bind(_Func&& __f, _BoundArgs&&... __args)
      |     ^~~~
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:785:5: note:   substitution of deduced template arguments resulted in errors seen above
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:809:5: note: candidate: 'template<class _Result, class _Func, class ... _BoundArgs> typename std::_Bindres_helper<_Result, _Func, _BoundArgs>::type std::bind(_Func&&, _BoundArgs&& ...)'
  809 |     bind(_Func&& __f, _BoundArgs&&... __args)
      |     ^~~~
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/functional:809:5: note:   template argument deduction/substitution failed:
In file included from src/esphome.h:111,
                 from src/main.cpp:3:
src/esphome/core/base_automation.h:135:23: note:   couldn't deduce template parameter '_Result'
  135 |     auto f = std::bind(&DelayAction<Ts...>::play_next_, this, x...);
      |              ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs/sensibo-sky-sittingroom/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.33 seconds ==========================
```

ssieb, 2023-10-26:
I guess you haven't updated this device for a while.
It's caused by https://github.com/esphome/esphome/pull/5028 which was added in 2023.8.

ssieb, 2023-10-26:
Until this is fixed, you can work around it by moving the actions to a script, which is really the better option anyway.

erumaithi, 2023-10-26:
> Until this is fixed, you can work around it by moving the actions to a script, which is really the better option anyway.

thank you ssieb,  would you mind show me how to do it in script.  i apologize for my lack of knowledge .

ssieb, 2023-10-26:
You have a workaround, but it's still a problem.

hsteinhaus, 2025-04-22:
I guess I still see the same issue in 2025.4.0 when trying to add an `delay`action to a òn_state`trigger of a `climate` component:
```
climate:
  - platform: pid
    id: this_thermostat
    name: "${friendly_name} PID Climate Controller"
    sensor: internal_temperature
    default_target_temperature: 21°C
    heat_output: heat_pwm_out
    control_parameters:
      kp: 0.4
      ki: 0.02
      kd: 220.0
      min_integral: 0.0
      max_integral: 0.3
      derivative_averaging_samples: 18
      starting_integral_term: 0.3
    on_state:
        then:
          - delay: 1s
```

```
In file included from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/bits/unique_ptr.h:36,
                 from /data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/memory:76,
                 from src/esphome/components/socket/socket.h:2,
                 from src/esphome/components/api/api_frame_helper.h:14,
                 from src/esphome/components/api/api_connection.h:5,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/tuple: In instantiation of 'struct std::_Head_base<1, esphome::climate::Climate, false>':
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/tuple:416:12:   required from 'struct std::_Tuple_impl<1, esphome::climate::Climate>'
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/tuple:258:12:   required from 'struct std::_Tuple_impl<0, esphome::DelayAction<esphome::climate::Climate&>*, esphome::climate::Climate>'
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/tuple:981:11:   required from 'class std::tuple<esphome::DelayAction<esphome::climate::Climate&>*, esphome::climate::Climate>'
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/functional:476:29:   required from 'class std::_Bind<void (esphome::Action<esphome::climate::Climate&>::*(esphome::DelayAction<esphome::climate::Climate&>*, esphome::climate::Climate))(esphome::climate::Climate&)>'
src/esphome/core/base_automation.h:159:23:   required from 'void esphome::DelayAction<Ts>::play_complex(Ts ...) [with Ts = {esphome::climate::Climate&}]'
src/esphome/core/base_automation.h:158:8:   required from here
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/12.2.0/tuple:238:13: error: cannot declare field 'std::_Head_base<1, esphome::climate::Climate, false>::_M_head_impl' to be of abstract type 'esphome::climate::Climate'
  238 |       _Head _M_head_impl;
      |             ^~~~~~~~~~~~
Compiling .pioenvs/tempcontrol/lwip/lwip/src/core/tcp_in.c.o
In file included from src/esphome/core/controller.h:29,
                 from src/esphome/components/api/api_server.h:11,
                 from src/esphome/components/api/api_connection.h:8:
src/esphome/components/climate/climate.h:168:7: note:   because the following virtual functions are pure within 'esphome::climate::Climate':
  168 | class Climate : public EntityBase {
      |       ^~~~~~~
src/esphome/components/climate/climate.h:275:25: note:     'virtual esphome::climate::ClimateTraits esphome::climate::Climate::traits()'
  275 |   virtual ClimateTraits traits() = 0;
      |                         ^~~~~~
src/esphome/components/climate/climate.h:285:16: note:     'virtual void esphome::climate::Climate::control(const esphome::climate::ClimateCall&)'
  285 |   virtual void control(const ClimateCall &call) = 0;
      |                ^~~~~~~
Compiling .pioenvs/tempcontrol/lwip/lwip/src/core/tcp_out.c.o
Compiling .pioenvs/tempcontrol/lwip/lwip/src/core/timeouts.c.o
Compiling .pioenvs/tempcontrol/lwip/lwip/src/core/udp.c.o
Compiling .pioenvs/tempcontrol/lwip/lwip/src/core/ipv4/autoip.c.o
Compiling .pioenvs/tempcontrol/lwip/lwip/src/core/ipv4/dhcp.c.o
*** [.pioenvs/tempcontrol/src/main.cpp.o] Error 1
```