esphome/issues #5035: 2023 Mopeka Devices & esp32_ble_tracker

TelecomRob, 2023-10-26:
### The problem

Hello, I'm having issues with the esp32_ble_tracker not picking up all ble devices when scanning. It is pick up other bluetooth devices but not a 2023 Model Mopeka Pro check Universal M1017 Sensor. I'm wondering if Mopeka changes the chipset on the boards or encrypted it as if I run a scan using Arduino Ble it can see the device.

I have poked around other threads and it seems other people have been having issues with the 2023 Model Mopeka Pro check Universal M1017 Sensor in regards to esp32_ble_tracker not seeing them.

Screenshot of CLI:
[image]

### Which version of ESPHome has the issue?

2023.10.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
name: mopeka

esp32:
board: esp32dev
framework:
type: arduino

Enable logging
logger:

Enable Home Assistant API
api:
password: ""

ota:
password: ""

wifi:
ssid: "abc12345"
password: "abc12345"

Enable fallback hotspot (captive portal) in case wifi connection fails
ap:
ssid: "Mopeka Fallback Hotspot"
password: "abc12345"

captive_portal:

Enable Bluetooth Proxy Components
esp32_ble_tracker:
scan_parameters:
interval: 320ms
window: 120ms
duration: 60s
active: true

bluetooth_proxy:
active: true

mqtt:
broker: 192.168.1.94
username: ""
password: ""
client_id: "mopeka"

sensor:

platform: mopeka_pro_check
mac_address: 4c:3c:19:88:3f:4c
tank_type: 20LB_V
temperature:
name: "Propane test temp"
level:
name: "Propane test level"
distance:
name: "Propane test distance"
battery_level:
name: "Propane test battery level"
```

### Additional information

How Arduino BLE see device when scan:
Advertised Device: Name: , Address: 4c:3c:19:88:3f:4c, manufacturer data: 4c0010061b1940e8a878, txPower: 12, rssi: -10

spbrogan, 2023-10-28:
I might suggest a couple of things to try first.  
1. Drop the ble proxy
2. drop the mqtt stuff.  

My guess is the esp32 is not able to handle all the activity generated by the proxy and is having to minimize its ble scanning time and missing stuff.  I have seen this on a number of devices and in busy ble environments.

spbrogan, 2023-10-28:
You may also want to press the green sync button a bunch of times (like 10+) to see if for some reason it is in mfg mode.

TelecomRob, 2023-10-28:
I’ve been doing some digging and it’s looking like the 2023 “universal models” are not supported for now.

runjmc, 2023-10-31:
There looks to be a change in the sensor. I see a unsupported sensor type in the logs.

[15:10:50][I][mopeka_ble:081]: MOPEKA PRO (NRF52) SENSOR FOUND: F5:5A:E4:8A:AA:2C
[15:10:50][E][mopeka_pro_check:058]: Unsupported Sensor Type (0xC)
[15:10:58][I][mopeka_ble:081]: MOPEKA PRO (NRF52) SENSOR FOUND: F5:5A:E4:8A:AA:2C
[15:10:58][E][mopeka_pro_check:058]: Unsupported Sensor Type (0xC)

Let me know if I should open up another issue for this.

TelecomRob, 2023-11-01:
Update: I have officially confirmed that the Universal Mopeka Pro Check is not compatible. I bought a non universal and everything works fine.

TelecomRob, 2023-11-19:
Another update: The core integration for Mopeka has been updated apparently to support the unsupported sensor. When can we see this with esp?

spbrogan, 2023-11-19:
Looks like they just added another sensor type. Seems trivial.   I can add that but will need @TelecomRob or someone else to test it on real hardware.  Is that possible?

TelecomRob, 2023-11-19:
> Looks like they just added another sensor type. Seems trivial. I can add that but will need @TelecomRob or someone else to test it on real hardware. Is that possible?

I can test it.

closed 2023-11-20 by nagyrobi

TelecomRob, 2023-11-20:
> > Looks like they just added another sensor type. Seems trivial. I can add that but will need @TelecomRob or someone else to test it on real hardware. Is that possible?
> 
> I can test it.

I'm having issues with esphome-docs at them moment so I can not test. Can someone else possibly. Cheers
[image]

TelecomRob, 2023-12-10:
I’m unable to install docs on my end of things but running esphome-beta it is not working for me still. Apparently core has updated and it’s working for that type of installation. Reference: https://github.com/home-assistant/core/issues/98323#issuecomment-1848667178

TelecomRob, 2023-12-16:
@spbrogan 
It now works using the current ESP Home Beta version. Both my universal and pro check sensors are detecting. I’ll monitor for stability.
[image]

TelecomRob, 2024-02-20:
> Hey Guys, today I get my Mopeka Pro Check Water Level Sensor. Unfortunately I get the message:
> 
> _[20:26:36][E][mopeka_pro_check:059]: Unsupported Sensor Type (0x5)_
> 
> The "normal" integration of Mopeka is working... Is it possible to integrate this sensor, too? (I could help out if some testing is needed)
> 
> Best regards, Steffen

@spbrogan 

Would this be the same situation as the Pro Check where another sensor Type needs to be added?

spbrogan, 2024-02-24:
@smai86 could you open a new issue/feature request.   Adding the sensor id is easy but i would guess the distance measurement is different for water (or water like).   The calculation I used for propane came from documentation mopeka shared with me and I have been unable to get mopeka to respond to any email since to get updated documentation.   

Does the main integration have support for water?  Is their sensor reading accurate compared to the app? 

thanks
Sean

smai86, 2024-02-26:
@spbrogan 
Yes of course I could open a new feature request. If I could help out with some data, let me know.

The main integration doesn´t work very well... The bluetooth connection is pretty unstable and the reading shows a different distance than the app shows. Pls let me know if I could help u out. Would be pretty cool, to get it with ESPHome like the sensors for butane/propane.

thanks a lot,
Steffen

kevinhoogdorp, 2024-05-08:
> @spbrogan It now works using the current ESP Home Beta version. Both my universal and pro check sensors are detecting. I’ll monitor for stability. ![IMG_3631](https://private-user-images.githubusercontent.com/145357807/290965258-211b1bb2-2ed4-4714-9f83-0f1de11eb239.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTUxNzg1NzQsIm5iZiI6MTcxNTE3ODI3NCwicGF0aCI6Ii8xNDUzNTc4MDcvMjkwOTY1MjU4LTIxMWIxYmIyLTJlZDQtNDcxNC05ZjgzLTBmMWRlMTFlYjIzOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwOFQxNDI0MzRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kYzRkODQyMzFiZDE3MjIzMmYzY2UzZjgwOWEyNTk5ZmRkMzM2NTc5N2Y1OWE1NDQxYjY1YzRlZmRlNGU1Mjg5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.tXWxFk6zPvh34GID_ddQJNQSh-cG3zjbmVkVlBcS_oY)

This worked for me..
I'm running supervisor 2024.05.1 on Ubuntu
I installed the ESPHome Beta and did a restart and the sensor popped up