esphome/issues #5037: build_path ignored

sbrk-r, 2023-10-26:
### The problem

esphome core config is no longer honouring the 'build_path' option exactly, it seems to be prepending it with './.esphome/'

I hadn't updated in quite some time but this used to work, and definitely doesn't work in 2023.9.x and 2023.10.3

using, eg , 
```
esphome:
  build_path: ./out/project0
```
code used to be generated as stated, but now it goes into './.esphome/out/project0'

### Which version of ESPHome has the issue?

2023.10.3

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
esphome:
  name: project0
  build_path: ./out/project0
```

andrewjswan, 2024-01-30:
Some here ...

andrewjswan, 2024-04-04:
In https://github.com/esphome/esphome/pull/6276
Written: 
This build path still can be overridden by individual components over `build_path` option (as described in [doc](https://esphome.io/components/esphome.html)), but introduce default folder that can be specified.

But if you specify a relative path (`./out/project0`), this does not happen.
As far as I understand, the absolute path to build the firmware is hardcoded into the container.
https://github.com/esphome/esphome/blob/0148ebcaa604f929aba06b6722697bb1b6e4c83c/docker/docker_entrypoint.sh#L27

And in the end we get the path: `/build/./out/project0`
https://github.com/esphome/esphome/blob/0148ebcaa604f929aba06b6722697bb1b6e4c83c/esphome/core/config.py#L203-L206

When simultaneously working on firmware in the ESPHome addon and in ESPHome, for example in Windows, this is very inconvenient.

Is it possible to add an option in the addon where to `build the firmware` and where the `/build` directory will be located? 
Or let's say direct it to `/addons/esphome` or `/addon_config/esphome`?

sbrk-r, 2024-04-08:
I guess that is docker specific, on pip install it seems to default to ./.esphome/ rather than build.

andrewjswan, 2024-04-08:
Yes, the problem is in the container settings, and it can be fixed.