esphome/issues #5038: M5Stack Atom Echo no longer detects voice to start wakeword detection after a while

rccoleman, 2023-10-26:
### The problem

There's been quite a bit of discussion about the M5 Atom Echo no longer responding to the programmed wakeword after "a while" (like overnight for me), but I don't see anything here and it seems related to the ESPHome build on this device.  See the tail end of https://github.com/home-assistant/core/issues/101995 for observations and logs from the 2023.10.3 ESPHome build.  In summary, the device seems to get stuck in WAITING_FOR_VAD mode until I either shout directly at it at close range, or sometimes even that's not enough.  Without it transitioning from that state to recording voice, it won't detect the wakeword.  Toggling the "Use wake word" switch in HA always restores normal behavior for me, and the device is otherwise responsive to commands from HA (like Use listen light) even when it's not detecting voice.

### Which version of ESPHome has the issue?

2023.10.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.11.0b0 currently, but also 2023.10.x

### What platform are you using?

ESP32

### Board

M5Stack Atom Echo

### Component causing the issue

Voice Assistant

### Additional information

I'm using the default, unmodified voice assistant configuration for the Atom Echo, as first flashed via <https://www.home-assistant.io/voice_control/thirteen-usd-voice-remote/> and then adopted by ESPHome and clean/build/install with updated ESPHome releases.

kevvar, 2023-10-27:
I'm facing this issue in the same way exactly. Any help will be great

rccoleman, 2023-10-27:
I'm going try to add some debug info tonight to see if it's the voice detection itself or the number of consecutive samples that are required. Unfortunately, it looks like voice detection function itself comes from Espressif and isn't open source.

ScottG489, 2023-10-27:
> voice detection itself or the number of consecutive samples 

So from personal experience I'm more confident that it's due to duration and not number of activations if that's what you mean by this. Due to the glitch I wasn't using the device much or at all and when I did try the voice detection almost never worked unless I "reset" it with the switch as you described.

rccoleman, 2023-10-27:
The initial voice detection process is pretty straightforward -

* Every time through the event loop, when waiting for voice audio, read a sample from the microphone (512 bytes)
* Call `vad_process` with that sample to look for voice
* If that call returns `VAD_SPEECH`, increment the number of "voice" samples until it meets the threshold
* If that call returns something else, decrement that count
* When the voice threshold is met, start the wakeword pipeline

My suspicion is that it's maybe not getting data from the microphone or the `vad_process` function isn't detecting voice in the sample, either leading to the pipeline never being triggered.  One observation is that when it hangs, I don't see any logging at all, in contrast to the regular state transitions that I see in normal operation.  With my additional logging, I also see that it always detects voice in every sample, even in a quiet room, and consistent across two Atoms.  That's undesirable, but would theoretically lead to more activations, not fewer.

kevvar, 2023-10-27:
Correct in my case the same thing happens, I have to talk to it very close to activate the voice reception, in other cases it does not even work, I have to press the button or activate-deactivate the wake word in the device configuration, as you say it probably due to microphone gain issues, in my case I flashed my device from the Home Assistant guides, do you think that flashing the code manually to modify the audio and microphone gain values ​​can solve something?

rccoleman, 2023-10-27:
I'm already making changes to the code and flashing via ESPHome, but I need to narrow down the problem first. I hope it repros with my current build and I get some useful information.

kevvar, 2023-10-27:
Great @rccoleman any news or additional configuration would be great, please share it if it is not a bother.

Neurocranium, 2023-10-27:
> The initial voice detection process is pretty straightforward -
>     * Every time through the event loop, when waiting for voice audio, read a sample from the microphone (512 bytes)
>     * Call [...]

Fully agree to this suspicion. I did experiments with vad_threshold = 0 and while I'm not able to inject my own debug code and get this to run, I certainly can read the code. x-) With the vad_threshold = 0 the whole increment/decrement is logic is basically skipped and each return of `VAD_SPEECH` by `vad_process` should lead to the "VAD detected speech" notification and streaming of the mic data.

Not sure how I could further verify, whether the issue comes from vad_process or maybe not getting data from the mic. @rccoleman If you have any tips how to get my own debug code in and running on the Atom, I happily try this over the weekend.

I actually hope it is something related to reading from the mic. Because if it is solely in vad_process we might have an upstream Espressif SDK issue here, likely no one of us could fix.

rccoleman, 2023-10-27:
I'm using a container installation for ESPHome and just shelled into the container, copied the entire `/esphome` directory into `/config/esphome` and made a link in the container to the new directory in `/config` that I can edit externally.  You can theoretically do the same for the addon.  You could probably do it easier by installing ESPHome [via pip](https://esphome.io/guides/installing_esphome), but I didn't want to mess with my configuration any more than necessary.

Neurocranium, 2023-10-27:
Oh, good point. My setup is container based as well. Then I even could just volume mount my customized `voice_assistant.cpp` onto the right place in the container and would be good to go. Over-designed it in my head too much.

Will try this out over the weekend and share what I find.

rccoleman, 2023-10-27:
Yep, that too :)

Neurocranium, 2023-10-27:
Injected this little piece of debug code
```cpp
    case State::WAITING_FOR_VAD: {
      size_t bytes_read = this->read_microphone_();
      ESP_LOGD(TAG, "Read %d from mic", bytes_read);
      if (bytes_read > 0) {
        vad_state_t vad_state =
            vad_process(this->vad_instance_, this->input_buffer_, SAMPLE_RATE_HZ, VAD_FRAME_LENGTH_MS);
        ESP_LOGD(TAG, "vad_state %d returned", vad_state);
        if (vad_state == VAD_SPEECH) {
```
Flashed it on two Atoms in different rooms under different noise conditions and it works quite nicely
```
[18:45:59][D][voice_assistant:176]: Waiting for speech...
[18:45:59][D][voice_assistant:368]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[18:45:59][D][voice_assistant:182]: Read 512 from mic
[18:45:59][D][voice_assistant:186]: vad_state 1 returned
[18:45:59][D][voice_assistant:191]: VAD detected speech
```
Now it is just wait and see whether we get 0 or less from the mic, or always a vad_state of 0.

rccoleman, 2023-10-27:
Yeah, very similar to mine :).  I've had a lot of trouble keeping logs streaming, so I plugged one of my devices directly into a laptop and am using a serial terminal to watch the logs.  Hopefully it keeps the logs going and I eventually see the failure.

kevvar, 2023-10-27:
Thank you for your comments, guys. I don’t know much about programming other than the basics to program ESPHome devices, so if I want to try these solutions that you give, where should I inject that code?

rccoleman, 2023-10-27:
I suggest waiting until there's something to try first.  This exercise is just aimed at finding what's wrong.

kevvar, 2023-10-27:
Oh got it, thanks!

rccoleman, 2023-10-28:
After updating to the latest openwakeword container (1.8.1), I hadn't seen it fail until I restarted HA.  Afterward, it stopped responding to my wakeword:

```
[D][voice_assistant:392]: Client started, streaming microphone
[D][voice_assistant:371]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[D][voice_assistant:377]: Desired state set to STREAMING_MICROPHONE
[D][voice_assistant:473]: Event Type: 1
[D][voice_assistant:476]: Assist Pipeline running
[D][voice_assistant:473]: Event Type: 9
[D][api.connection:171]: Home Assistant 2023.11.0b1 (192.168.1.159) requested disconnected
[D][voice_assistant:371]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[D][voice_assistant:377]: Desired state set to IDLE
[D][voice_assistant:467]: Signaling stop...
[D][voice_assistant:371]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[D][esp-idf:000]: I (76530541) I2S: DMA queue destroyed

[D][voice_assistant:371]: State changed from STOPPING_MICROPHONE to IDLE
[D][api:102]: Accepted 192.168.1.159
[W][component:214]: Component api took a long time for an operation (0.05 s).
[W][component:215]: Components should block for at most 20-30ms.
[D][api.connection:1032]: Home Assistant 2023.11.0b1 (192.168.1.159): Connected successfully
```
It's weird, because it claims to have reconnected successfully.  Pressing the button gets it started again:

```
[D][binary_sensor:036]: 'Button': Sending state ON
[D][binary_sensor:036]: 'Button': Sending state OFF
[D][light:036]: 'Family Room Assist' Setting:
[D][voice_assistant:371]: State changed from IDLE to START_MICROPHONE
[D][voice_assistant:377]: Desired state set to WAIT_FOR_VAD
[D][voice_assistant:159]: Starting Microphone
[D][voice_assistant:371]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[D][esp-idf:000]: I (78415641) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4
[D][voice_assistant:371]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[D][voice_assistant:176]: Waiting for speech...
[D][voice_assistant:371]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[D][voice_assistant:182]: Read 512 bytes
[D][voice_assistant:187]: VAD detected speech (raw)
[D][voice_assistant:189]: VAD detected speech (counter=0, threshold=3)
[D][voice_assistant:182]: Read 512 bytes
[D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[D][voice_assistant:201]: VAD counter decrement
[D][voice_assistant:182]: Read 512 bytes
[D][voice_assistant:187]: VAD detected speech (raw)
[D][voice_assistant:189]: VAD detected speech (counter=0, threshold=3)
[D][voice_assistant:182]: Read 512 bytes
[D][voice_assistant:187]: VAD detected speech (raw)
[D][voice_assistant:189]: VAD detected speech (counter=1, threshold=3)
[D][voice_assistant:182]: Read 512 bytes
[D][voice_assistant:187]: VAD detected speech (raw)
[D][voice_assistant:189]: VAD detected speech (counter=2, threshold=3)
[D][voice_assistant:182]: Read 512 bytes
[D][voice_assistant:187]: VAD detected speech (raw)
[D][voice_assistant:192]: VAD detected speech
```

So it seems like disconnecting & reconnecting to HA leaves it in `IDLE` mode, even if it's supposed to be looking for the wakeword.

turboc1208, 2023-10-28:
I'm not sure if I'm having this same issue or whether it's just a noise issue, but while trying to debug it, I can't find the troubleshooting files referenced in the docs section on "Tweaking the atom echo configuraiton". I'm running under docker, I have separate containers for HA, Whisper, Piper, Open wake word and esp home.  I have samba installed but I don't think anything is actually using it.  the M5 works sometimes, but I can't find the wav files referenced or the yaml file referenced for tweaking the settings.  Which container should these files be being generated and used by?

rccoleman, 2023-10-28:
Step 2 is referring to the HA configuration.yaml. Step 7 refers to wherever you pointed the ESPHome container to. You can use whatever method you want to access and edit them.

rccoleman, 2023-10-28:
Now my second Atom has gotten into a state where I have to shout at it for it to recognize the wakeword, and my logging shows that it has gone from *always* detecting voice from `vad_process` to almost never detecting it.  Here's a log snippet:

```
[16:37:54][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:187]: VAD detected speech (raw)
[16:37:54][D][voice_assistant:189]: VAD detected speech (counter=0, threshold=3)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:187]: VAD detected speech (raw)
[16:37:54][D][voice_assistant:189]: VAD detected speech (counter=1, threshold=3)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:187]: VAD detected speech (raw)
[16:37:54][D][voice_assistant:189]: VAD detected speech (counter=2, threshold=3)
[16:37:54][D][voice_assistant:182]: Read 512 bytes
[16:37:54][D][voice_assistant:187]: VAD detected speech (raw)
[16:37:54][D][voice_assistant:192]: VAD detected speech
[16:37:54][D][voice_assistant:371]: State changed from WAITING_FOR_VAD to START_PIPELINE
[16:37:54][D][voice_assistant:377]: Desired state set to STREAMING_MICROPHONE
[16:37:55][D][voice_assistant:211]: Requesting start...
[16:37:55][D][voice_assistant:371]: State changed from START_PIPELINE to STARTING_PIPELINE
[16:37:55][D][voice_assistant:392]: Client started, streaming microphone
[16:37:55][D][voice_assistant:371]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[16:37:55][D][voice_assistant:377]: Desired state set to STREAMING_MICROPHONE
[16:37:55][D][voice_assistant:473]: Event Type: 1
[16:37:55][D][voice_assistant:476]: Assist Pipeline running
[16:37:55][D][voice_assistant:473]: Event Type: 9
[16:37:56][D][voice_assistant:473]: Event Type: 10
[16:37:56][D][voice_assistant:482]: Wake word detected
[16:37:56][D][voice_assistant:473]: Event Type: 3
[16:37:56][D][voice_assistant:487]: STT Started
[16:37:56][D][light:036]: 'Master Bedroom Assist' Setting:
[16:37:56][D][light:047]:   State: ON
[16:37:56][D][light:051]:   Brightness: 100%
[16:37:56][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[16:37:56][D][light:109]:   Effect: 'Pulse'
[16:38:11][D][voice_assistant:473]: Event Type: 12
[16:38:11][D][voice_assistant:604]: Unhandled event type: 12
[16:38:11][D][voice_assistant:473]: Event Type: 0
[16:38:11][E][voice_assistant:585]: Error: stt-no-text-recognized - No text recognized
[16:38:11][D][voice_assistant:467]: Signaling stop...
[16:38:11][D][voice_assistant:371]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[16:38:11][D][voice_assistant:377]: Desired state set to IDLE
[16:38:11][D][light:036]: 'Master Bedroom Assist' Setting:
[16:38:11][D][light:051]:   Brightness: 100%
[16:38:11][D][light:059]:   Red: 100%, Green: 0%, Blue: 0%
[16:38:11][D][light:109]:   Effect: 'None'
[16:38:11][D][voice_assistant:371]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[16:38:11][D][esp-idf:000]: I (16702953) I2S: DMA queue destroyed

[16:38:11][D][voice_assistant:473]: Event Type: 2
[16:38:11][D][voice_assistant:555]: Assist Pipeline ended
[16:38:11][D][voice_assistant:371]: State changed from STOPPING_MICROPHONE to IDLE
[16:38:11][D][voice_assistant:371]: State changed from IDLE to START_MICROPHONE
[16:38:11][D][voice_assistant:377]: Desired state set to WAIT_FOR_VAD
[16:38:11][D][voice_assistant:159]: Starting Microphone
[16:38:11][D][voice_assistant:371]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[16:38:11][D][esp-idf:000]: I (16703005) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4

[16:38:11][D][voice_assistant:371]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
[16:38:11][D][voice_assistant:176]: Waiting for speech...
[16:38:11][D][voice_assistant:371]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
[16:38:11][D][voice_assistant:182]: Read 512 bytes
[16:38:11][D][voice_assistant:187]: VAD detected speech (raw)
[16:38:11][D][voice_assistant:189]: VAD detected speech (counter=0, threshold=3)
[16:38:11][D][voice_assistant:182]: Read 512 bytes
[16:38:11][D][voice_assistant:187]: VAD detected speech (raw)
[16:38:11][D][voice_assistant:189]: VAD detected speech (counter=1, threshold=3)
[16:38:11][D][voice_assistant:182]: Read 512 bytes
[16:38:11][D][voice_assistant:187]: VAD detected speech (raw)
[16:38:11][D][voice_assistant:189]: VAD detected speech (counter=2, threshold=3)
[16:38:11][D][light:036]: 'Master Bedroom Assist' Setting:
[16:38:11][D][light:047]:   State: OFF
[16:38:11][D][voice_assistant:182]: Read 512 bytes
[16:38:11][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[... cut ...]
[16:38:11][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
[16:38:11][D][voice_assistant:201]: VAD counter decrement
[16:38:11][D][voice_assistant:182]: Read 512 bytes
[16:38:11][D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
```
I didn't actually give it a command here (hence the `Error: stt-no-text-recognized - No text recognized`), but it took a pretty loud and close range "OK NABU" to get it to even detect voice at all before it processed the wakeword.  I guess I can log some of the packet to see if the contents are interesting.

turboc1208, 2023-10-29:
> Step 2 is referring to the HA configuration.yaml. Step 7 refers to wherever you pointed the ESPHome container to. You can use whatever method you want to access and edit them.

The file in step 7 doesn't seem to be in the esphome container or any of the others.  I tried doing finds for all .yaml files and never saw it.  Exactly where is that configured or could it be pointing somewhere else, turned on another way, or called something else?  I just installed the M5 today, so I'm assuming I'm on the most current version of it's esp code.

rccoleman, 2023-10-29:
I suggest following up on Discord where you've already asked about how to debug general configuration issues.

Neurocranium, 2023-10-29:
My logs and observations are pretty much consistent with what @rccoleman found.

In the hard of hearing state, the Atom is constantly looping over reading from mic and getting a no voice detected from `vad_process` unless shouting at it.
```
INFO ESPHome 2023.10.3
INFO Reading configuration /config/va-atom-m5-arbeitszimmer.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:30][I][app:102]: ESPHome version 2023.10.3 compiled on Oct 27 2023, 19:35:29
[14:49:30][I][app:104]: Project m5stack.atom-echo-voice-assistant version 1.0
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:30][C][logger:416]: Logger:
[14:49:30][C][logger:417]:   Level: DEBUG
[14:49:30][C][logger:418]:   Log Baud Rate: 115200
[14:49:30][C][logger:420]:   Hardware UART: UART0
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:30][C][esp32_rmt_led_strip:171]: ESP32 RMT LED Strip:
[14:49:30][C][esp32_rmt_led_strip:172]:   Pin: 27
[14:49:30][C][esp32_rmt_led_strip:173]:   Channel: 0
[14:49:30][C][esp32_rmt_led_strip:198]:   RGB Order: GRB
[14:49:30][C][esp32_rmt_led_strip:199]:   Max refresh rate: 0
[14:49:30][C][esp32_rmt_led_strip:200]:   Number of LEDs: 1
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:30][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:30][D][voice_assistant:186]: vad_state 0 returned
[14:49:30][D][voice_assistant:203]: Current vad_counter is 0
[14:49:31][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:31][D][voice_assistant:186]: vad_state 0 returned
[14:49:31][D][voice_assistant:203]: Current vad_counter is 0
...
[14:49:37][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:37][D][voice_assistant:186]: vad_state 0 returned
[14:49:37][D][voice_assistant:203]: Current vad_counter is 0
[14:49:37][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:37][D][voice_assistant:186]: vad_state 0 returned
[14:49:37][D][voice_assistant:203]: Current vad_counter is 0
[14:49:37][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:37][D][voice_assistant:186]: vad_state 0 returned
[14:49:37][D][voice_assistant:203]: Current vad_counter is 0
[14:49:37][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:37][D][voice_assistant:186]: vad_state 1 returned
[14:49:37][D][voice_assistant:203]: Current vad_counter is 1
[14:49:37][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:37][D][voice_assistant:186]: vad_state 1 returned
[14:49:37][D][voice_assistant:203]: Current vad_counter is 2
[14:49:37][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:37][D][voice_assistant:186]: vad_state 1 returned
[14:49:37][D][voice_assistant:203]: Current vad_counter is 3
[14:49:37][D][voice_assistant:182]: Read 512 bytes from mic
[14:49:37][D][voice_assistant:186]: vad_state 1 returned
[14:49:37][D][voice_assistant:191]: Current vad_counter is 3
[14:49:37][D][voice_assistant:192]: VAD detected speech
[14:49:37][D][voice_assistant:370]: State changed from WAITING_FOR_VAD to START_PIPELINE
[14:49:37][D][voice_assistant:376]: Desired state set to STREAMING_MICROPHONE
[14:49:37][D][voice_assistant:203]: Current vad_counter is 0
[14:49:37][D][voice_assistant:210]: Requesting start...
[14:49:37][D][voice_assistant:370]: State changed from START_PIPELINE to STARTING_PIPELINE
[14:49:37][D][voice_assistant:391]: Client started, streaming microphone
[14:49:37][D][voice_assistant:370]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
[14:49:37][D][voice_assistant:376]: Desired state set to STREAMING_MICROPHONE
[14:49:37][D][voice_assistant:472]: Event Type: 1
[14:49:37][D][voice_assistant:475]: Assist Pipeline running
[14:49:37][D][voice_assistant:472]: Event Type: 9
[14:49:38][D][voice_assistant:472]: Event Type: 10
[14:49:38][D][voice_assistant:481]: Wake word detected
[14:49:38][D][voice_assistant:472]: Event Type: 3
[14:49:38][D][voice_assistant:486]: STT Started
[14:49:38][D][light:036]: 'M5Stack Atom VA Arbeitszimmer 888db8' Setting:
[14:49:38][D][light:051]:   Brightness: 100%
[14:49:38][D][light:059]:   Red: 0%, Green: 0%, Blue: 100%
[14:49:38][D][light:109]:   Effect: 'Pulse'
```

My second Atom, also either ended up in this state or even in the complete no logs at all state.

rccoleman, 2023-10-29:
I'm currently printing the first 10 bytes of the buffer that's passed to vad_process and also monitoring whether the ring buffer fills up, but I haven't figured out how to trigger the mostly-deaf case yet. I haven't had it completely lock up after adding the logic to restart on API reconnect, though.

rccoleman, 2023-10-30:
I got the deafness to happen again and it seems related or at least coincident to the ring buffer becoming full:

```
[23:40:55][D][voice_assistant:184]: Read 512 bytes, instance=0x3ffd299c
[23:40:55][D][voice_assistant:188]: Buffer=AC4 AD4 ADC AD4 AE8 ADC AC8 AC4 AF8 AE0
[23:40:55][D][voice_assistant:204]: VAD detected NO speech (VAD state = 0)
[23:40:55][D][voice_assistant:121]: RB FULL: 0 avail
[23:40:55][D][voice_assistant:184]: Read 512 bytes, instance=0x3ffd299c
[23:40:55][D][voice_assistant:188]: Buffer=A88 AB0 AC0 A90 A9C AAC AB0 AA8 AA8 AA0
[23:40:55][D][voice_assistant:204]: VAD detected NO speech (VAD state = 0)
[23:40:55][D][voice_assistant:121]: RB FULL: 0 avail
```

I need to look more into how the ring buffer is managed, but it isn't even used for VAD and it seems like copying into it could be skipped when we're in that state.

rccoleman, 2023-11-02:
I later added a counter to see how often that stuck-at-no-voice situation occurs and it's more often that I had thought.  I count whenever it hits that "RB FULL" case, which generally means that it's stuck in a "no voice detected" loop and just filling the ring buffer with silence, recreate the VAD instance, and then display the count when it's looping in the WAITING_FOR_VAD state.  My observations are:

- Sometimes it *starts* in that stuck state and then gets out of it, maybe either because I recreate the VAD instance when the buffer finally fills or because it needs to "warm up" 🤷 .
- If I leave my two Atoms for a while and check on the count periodically, the counter does increase from time to time, but not continuously.  Whenever I check the logs, it appears to be operating normally, but has seemingly recovered from the "bad state" at some point in the past.
- I haven't had either device get into a persistent "deaf" state since adding the config code to restart WW detection when the API reconnects and recreating the VAD instance when the ring buffer fills.  Both are pretty ugly workarounds, but they seem to make a more stable system for me.

WebSpider, 2023-11-03:
I experience the same every morning when I walk into my office. My echo there works just fine for the day, but overnight it fails.

I reset it by pushing the button, which then gives me the following logs stopping the voice_assistant:

```txt
[08:54:14][D][binary_sensor:036]: 'Button': Sending state ON
[08:54:14][D][binary_sensor:036]: 'Button': Sending state OFF
[08:54:14][D][voice_assistant:495]: Signaling stop...
[08:54:14][D][voice_assistant:395]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
[08:54:14][D][voice_assistant:401]: Desired state set to IDLE
[08:54:14][D][voice_assistant:395]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
[08:54:14][D][esp-idf:000]: I (135383664) I2S: DMA queue destroyed

[08:54:14][D][voice_assistant:502]: Event Type: 0
[08:54:14][E][voice_assistant:614]: Error: no_wake_word - No wake word detected
[08:54:14][D][voice_assistant:495]: Signaling stop...
[08:54:14][D][voice_assistant:395]: State changed from STOPPING_MICROPHONE to STOP_MICROPHONE
[08:54:14][D][voice_assistant:401]: Desired state set to IDLE
[08:54:14][D][light:036]: 'M5Stack Atom Echo' Setting:
[08:54:14][D][light:047]:   State: ON
[08:54:14][D][light:051]:   Brightness: 100%
[08:54:14][D][light:059]:   Red: 100%, Green: 0%, Blue: 0%
[08:54:14][W][component:214]: Component api took a long time for an operation (0.05 s).
[08:54:14][W][component:215]: Components should block for at most 20-30ms.
[08:54:14][D][voice_assistant:395]: State changed from STOP_MICROPHONE to IDLE
[08:54:14][D][voice_assistant:502]: Event Type: 2
[08:54:14][D][voice_assistant:584]: Assist Pipeline ended
```

No wake word detected isnt from a lack of trying, I've been yelling at the device loudly :D
Apperently, it's stuck in 'streaming microphone'. I cant seem to find a documented state-machine for the voice_assistant, so I hope my logs help someone else.

imatrixcz, 2023-11-03:
isn't the problem that it puts the firmware something about Home Assistant 10.5 in the basic settings during installation, if I remember correctly?

rccoleman, 2023-11-03:
Well, I just had one device stop logging, and pressing the button seems to try and fail to start the microphone.

```
[08:29:43][D][binary_sensor:036]: 'Button': Sending state ON
[08:29:43][D][binary_sensor:036]: 'Button': Sending state OFF
[08:29:43][D][voice_assistant:382]: State changed from STARTING_MICROPHONE to STOP_MICROPHONE
[08:29:43][D][voice_assistant:388]: Desired state set to IDLE
[08:29:43][D][voice_assistant:382]: State changed from STOP_MICROPHONE to IDLE
[08:29:43][D][voice_assistant:382]: State changed from IDLE to START_MICROPHONE
[08:29:43][D][voice_assistant:388]: Desired state set to WAIT_FOR_VAD
[08:29:43][D][voice_assistant:166]: Starting Microphone
[08:29:43][D][voice_assistant:382]: State changed from START_MICROPHONE to STARTING_MICROPHONE
[08:29:44][D][light:036]: 'Master Bedroom Assist' Setting:
```

From the code, I'm guessing that it tries to start the microphone, but it isn't running when it checks before switching to WAIT_FOR_VAD mode and just stays in STARTING_MICROPHONE state. First time I've seen this particular failure, and I doubt that it's related to the other hacks I have in place.

rccoleman, 2023-11-08:
Part of the problem is that the VAD implementation is *always* detecting voice for me, which moves the state to STREAMING_MICROPHONE (perhaps leading to @WebSpider 's observation) where it listens to silence for a while, then runs the pipeline for a few seconds where it isn't reading from the microphone at all.  If you try to say the wakeword during that time, it isn't listening for it and won't hear it.  @jesserockz pointed to https://github.com/esphome/esphome/pull/5613 as a WIP to replace the VAD engine and I'll play with that to see if it's better.  At least the algorithm is in the source code and isn't just a call to an opaque library.

kevvar, 2023-11-08:
Thanks for every update @rccoleman great work, I hope any fix is released soon

WebSpider, 2023-11-08:
Thanks for mentioning that PR, I'll set it up on my atom echo and report progress in that PR

quadroluxinator, 2023-11-14:
Same problem here, only help is frequent disconnecting from power source.

pdarcos, 2023-11-17:
Just adding that I also have this exact same problem

hanogs, 2023-11-29:
> After updating to the latest openwakeword container (1.8.1), I hadn't seen it fail until I restarted HA. Afterward, it stopped responding to my wakeword:
> 
> ```
> [D][voice_assistant:392]: Client started, streaming microphone
> [D][voice_assistant:371]: State changed from STARTING_PIPELINE to STREAMING_MICROPHONE
> [D][voice_assistant:377]: Desired state set to STREAMING_MICROPHONE
> [D][voice_assistant:473]: Event Type: 1
> [D][voice_assistant:476]: Assist Pipeline running
> [D][voice_assistant:473]: Event Type: 9
> [D][api.connection:171]: Home Assistant 2023.11.0b1 (192.168.1.159) requested disconnected
> [D][voice_assistant:371]: State changed from STREAMING_MICROPHONE to STOP_MICROPHONE
> [D][voice_assistant:377]: Desired state set to IDLE
> [D][voice_assistant:467]: Signaling stop...
> [D][voice_assistant:371]: State changed from STOP_MICROPHONE to STOPPING_MICROPHONE
> [D][esp-idf:000]: I (76530541) I2S: DMA queue destroyed
> 
> [D][voice_assistant:371]: State changed from STOPPING_MICROPHONE to IDLE
> [D][api:102]: Accepted 192.168.1.159
> [W][component:214]: Component api took a long time for an operation (0.05 s).
> [W][component:215]: Components should block for at most 20-30ms.
> [D][api.connection:1032]: Home Assistant 2023.11.0b1 (192.168.1.159): Connected successfully
> ```
> 
> It's weird, because it claims to have reconnected successfully. Pressing the button gets it started again:
> 
> ```
> [D][binary_sensor:036]: 'Button': Sending state ON
> [D][binary_sensor:036]: 'Button': Sending state OFF
> [D][light:036]: 'Family Room Assist' Setting:
> [D][voice_assistant:371]: State changed from IDLE to START_MICROPHONE
> [D][voice_assistant:377]: Desired state set to WAIT_FOR_VAD
> [D][voice_assistant:159]: Starting Microphone
> [D][voice_assistant:371]: State changed from START_MICROPHONE to STARTING_MICROPHONE
> [D][esp-idf:000]: I (78415641) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=4
> [D][voice_assistant:371]: State changed from STARTING_MICROPHONE to WAIT_FOR_VAD
> [D][voice_assistant:176]: Waiting for speech...
> [D][voice_assistant:371]: State changed from WAIT_FOR_VAD to WAITING_FOR_VAD
> [D][voice_assistant:182]: Read 512 bytes
> [D][voice_assistant:187]: VAD detected speech (raw)
> [D][voice_assistant:189]: VAD detected speech (counter=0, threshold=3)
> [D][voice_assistant:182]: Read 512 bytes
> [D][voice_assistant:199]: VAD detected NO speech (VAD state = 0)
> [D][voice_assistant:201]: VAD counter decrement
> [D][voice_assistant:182]: Read 512 bytes
> [D][voice_assistant:187]: VAD detected speech (raw)
> [D][voice_assistant:189]: VAD detected speech (counter=0, threshold=3)
> [D][voice_assistant:182]: Read 512 bytes
> [D][voice_assistant:187]: VAD detected speech (raw)
> [D][voice_assistant:189]: VAD detected speech (counter=1, threshold=3)
> [D][voice_assistant:182]: Read 512 bytes
> [D][voice_assistant:187]: VAD detected speech (raw)
> [D][voice_assistant:189]: VAD detected speech (counter=2, threshold=3)
> [D][voice_assistant:182]: Read 512 bytes
> [D][voice_assistant:187]: VAD detected speech (raw)
> [D][voice_assistant:192]: VAD detected speech
> ```
> 
> So it seems like disconnecting & reconnecting to HA leaves it in `IDLE` mode, even if it's supposed to be looking for the wakeword. The default ESPHome configuration calls this when it wants to start looking for the wakeword (see [here](https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml)):
> 
> ```
>             - voice_assistant.start_continuous:
> ```
> 
> It does that in response to pressing the button, toggling the "Use wake word" switch, and when it initially boots, so perhaps we need to add something to trigger that when the API reconnects. I wonder if part of the issue is that the connection glitches sometimes and just doesn't restart listening, but that doesn't explain the need to shout at it sometimes.
> 
> I see that there's [an FR](https://github.com/esphome/feature-requests/issues/2272) to requesting a trigger for API connect/disconnect, but something like [this](https://community.home-assistant.io/t/send-sensor-value-when-api-connects/601400/7) is currently required?
> 
> I added this to my device config and it now reconnects and works after restarting HA:
> 
> ```
> interval:
>   - interval: 1s
>     then:
>       - if:
>           condition:
>             api.connected:
>           then:
>             - if:
>                 condition:
>                   and:
>                     - switch.is_on: use_wake_word
>                     - not:
>                         - voice_assistant.is_running
>                 then:
>                   - voice_assistant.start_continuous:
> ```
> 
> My other Atom without this change is currently in the "stuck, no logs" state after restarting HA, so I suspect that this is definitely part of the issue.

Where do I put this? In the m5stacks device ymal?

rccoleman, 2023-11-29:
A bunch of changes have been made around reestablishing the connection, so I don't think you need it anymore.

adekloet, 2023-12-01:
After the last ESPHome update the Athom Echo is more responsive and stays in listing mode. This is good, it does bring a new "Issue" to the table. The add-on for openwakeword uses a lot more CPU and some how together with ESPHome on RPI compilation of projects gives problems. Could this be related? Is the Echo loading the openwakeword add-on while listening all the time? Is there a way to optimize this? On an RPI4 4G it seems to bog it down.

CPU load in % of the openwakeword add-on after the update of the athom echo firmware using ESPHome 11.6:

[image]

ScottG489, 2023-12-01:
@adekloet Are you sure that you're comparing things properly? Meaning, maybe your CPU was low before because the device wasn't actually listening properly due to the bug discussed in this ticket.

Maybe someone can correct me on this, but open wake word *does* consume a meaningful amount of CPU I believe. I noticed quite a bit of CPU usage as well even before the update.

locomot1f, 2023-12-23:
Even after all the updates, I'm still experiencing the issue with having to be right next to my Atom Echo for it to recognize my voice. Wondering if this is still an open issue.
I will give those setting a try to see if it help with my issues.

kgstorm, 2023-12-29:
I also often need to scream at the M5 from about a foot away sometime to get it to recognize my voice, or press the button to reset. 
Can anyone confirm that vad_threshold does anything with the M5? And if so, what are the min/max values and the meaning behind them? I'm guessing it's like a squelch, and 0 would always push data to wakeword detection. Is that correct?

rccoleman, 2023-12-29:
`vad_threshold` is just a threshold for a counter for the number of samples that appear to contain voice based on the result of `vad_process`.  A counter increments each time a sample is determined to contain voice and decrements when it doesn't (clamped at zero).  When the counter exceeds `vad_threshold`, voice is "detected" and the pipeline is started.  Reducing `vad_threshold` to 1, for instance, will start the pipeline the first time `vad_process` detects voice.  My experience is that `vad_process` *always* detects voice, so changing the counter doesn't really do anything useful.  If it did, changing `vad_threshold` to a lower number would make it more "sensitive" and maybe also more likely to detect voice when there is none.

WebSpider, 2023-12-29:
I've seen a new interesting issue, trying to update my atom from ESPHome 2023.11 to the latest 2023.12.5

Backstory: It's holiday season, the device is in my office, so has been idle for like a week.

The update failed, due to a timeout.

Opening the logs via ESPHome dashboard, wireless option, made it apparent that the atom was very busy doing *something*, it wasn't even able to print the usual log entries with the wifi state (ip/dns) without being lagged.

I restarted the device, and then the update went through without problems.

Should this occur again, how can I find out what the device is occupied with?

TheSmartGerman, 2024-01-06:
Hi there,
I'll also see the behavior mentioned above. So far my Atom Echo seems to stop after a couple of minutes, or I have to shoud at it.
I'm running the latest stables versions (ESPHome, Phipher, Wisper, ...)  (06.01.2024). Any beta to test?

Not a good soluation. Is it possible to change the sensitivity of the microphone?

axwell, 2024-01-07:
I'm having the same issue i need to constantly reboot it. Otherwise pretty happy with it. Any updates on this ?

kgstorm, 2024-01-08:
> Hi there, I'll also see the behavior mentioned above. So far my Atom Echo seems to stop after a couple of minutes, or I have to shoud at it. I'm running the latest stables versions (ESPHome, [...]

@TheSmartGerman - to change the microphone settings, flash your M5 with this yaml:
https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml
except remove these lines:
dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

That'll give you the exact same firmware, with the ability to tune the mic as you see it. Downside is you wont get notified if/when the maintainers make an update to the yaml.

You can then modify these settings to meet your needs:
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  vad_threshold: 3

You can also follow these instructions to save the audio recordings to hear the result of changing the settings:
https://www.home-assistant.io/voice_control/troubleshooting/#to-tweak-the-assist-audio-configuration-for-your-device

However, I found that with the default settings, my voice was recorded loud and clear.  So I think there is some other issue going on. 
When I reset the M5, it can often hear me easily for several consecutive commands. But if I randomly walk into a room and expect it to work, it rarely does.

kgstorm, 2024-01-08:
I also had 3 of these things running, and am wondering if that is just too much processing for a RPI4. I'm going to try running just a single M5 and see if it improves.

rccoleman, 2024-01-08:
> @TheSmartGerman - to change the microphone settings, flash your M5 with this yaml: https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml except remove these lines: [...]

I don't think you need to replicate the package contents to change a value like that.  I just added a section in my YAML configuration that overrides the values in the package:

```
voice_assistant:
  vad_threshold: 1
  volume_multiplier: 4.0
```

llluis, 2024-02-17:
> Part of the problem is that the VAD implementation is _always_ detecting voice for me, which moves the state to STREAMING_MICROPHONE (perhaps leading to @WebSpider 's observation) where it listens to silence for a while, then runs the pipeline for a few seconds where it isn't reading from the microphone at all. If you try to say the wakeword during that time, it isn't listening for it and won't hear it. @jesserockz pointed to [esphome/esphome#5613](https://github.com/esphome/esphome/pull/5613) as a WIP to replace the VAD engine and I'll play with that to see if it's better. At least the algorithm is in the source code and isn't just a call to an opaque library.

I stumbled on this issue while searching about the VAD implementation on esphome voice_assistant and I have found the same bug: it's always detecting voice. I wrote a new application with just the mic and VAD process and it was pretty clear it was always detecting voice, no matter the input (absolute silence!).

I changed the input buffer read from mic and passed to the vad_process (from 32 to 64) and it solved in my case. 

Maybe you could try it as it's a very simple test.

https://github.com/esphome/esphome/blob/27a3a081c3aa50c7591565cc9eb45b70e484e023/esphome/components/voice_assistant/voice_assistant.cpp#L19

mrtnkhl, 2024-02-18:
Seeing the same issue. Running three Atom Echo's connecting to a Home Assistant Yellow. After a while all Atom Echos start blinking red once every couple seconds and the HA instance on the Yellow becomes unresponsive. Does not even react to reboot request via web interface. Only way to get it working again is to pull power from the Yellow and all Atom Echos. Then the setup works for a random period of time before the same issue reoccurs.

ESPHome 2023.12.9
HA Core 2024.2.2
Supervisor 2024.01.1
HA OS 11.5

Nothing relevant in the logs when the Yellow starts to freeze.

kgstorm, 2024-02-18:
Just another datapoint:
I switched my openwakeword from ‘hey rhasppie’ to ‘ok nabu’ and it seems to work much better now. 
other changes that may have contributed: I recently transitioned my system from an SD card to an SSD and changed TTS from Google cloud to piper

zolakt, 2024-02-20:
I have a similar issue. The wake word works a few times, then it just stops (usually after a few minutes). Disabling and enabling wake word detection "fixes it".

I've read this whole thread, but honestly can't find a valid fix or a workaround (maybe I just missed it). Some change suggestions, but nothing definitive. So, is there a fix or a workaround for this?

Currently, I'm using the stupidest possible workaround... but it helps. I have a NodeRed automation running every 2 minutes (unless an assist is already in progress) which just toggles the wake word switch off and on.

For what it's worth, initially I've flashed the firmware from the webinstaller, as per instructions in that 13$ voice assistant post. But I've noticed that the firmware version was stuck at the same version for a few months. So, I flashed it again, manually, using the yaml esphome firmware. Currently it's running the 2023.12.9 firmware. It might have helped a little bit (not sure honestly), but it definitely didn't solve the problem.

vtolstov, 2024-02-21:
in my case my atom-echo does not detect any wakewords and assist_in_progress is always off

Neurocranium, 2024-02-25:
> I changed the input buffer read from mic and passed to the vad_process (from 32 to 64) and it solved in my case.

I tried this tweak as well and it truly seems to increase the reliability of the vad_process. Now the system is actually waiting for voice. Maybe even becoming a bit too insensitive, so I plan to do some experiments with different buffer length. Unfortunately, right now I have also a bunch of other unrelated issues going on with voice assistant on the Atom. :-( So, might be a longer process overall.

super-qua, 2024-02-29:
From what I can see in my logs, the m5 and Piper lose the connection when this problem occurs. 
I don't know what causes them to lose the connection though.

```
Logger: homeassistant
Source: components/esphome/voice_assistant.py:354
First occurred: 17:38:56 (2 occurrences)
Last logged: 17:39:41

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 1190, in sendto
    self._sock.sendto(data, addr)
    ^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'sendto'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/voice_assistant.py", line 354, in _send_tts
    self.transport.sendto(chunk, self.remote_addr)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 1200, in sendto
    self._fatal_error(
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 867, in _fatal_error
    self._loop.call_exception_handler({
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'call_exception_handler'
```
---- 
```
ERROR:asyncio:Task exception was never retrieved
future: <Task finished name='Task-32907' coro=<AsyncEventHandler.run() done, defined at /usr/local/lib/python3.11/dist-packages/wyoming/server.py:28> exception=ConnectionResetError('Connection lost')>
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/wyoming/server.py", line 35, in run
    if not (await self.handle_event(event)):
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/wyoming_piper/handler.py", line 104, in handle_event
    await self.write_event(
  File "/usr/local/lib/python3.11/dist-packages/wyoming/server.py", line 26, in write_event
    await async_write_event(event, self.writer)
  File "/usr/local/lib/python3.11/dist-packages/wyoming/event.py", line 114, in async_write_event
    await writer.drain()
  File "/usr/lib/python3.11/asyncio/streams.py", line 378, in drain
    await self._protocol._drain_helper()
  File "/usr/lib/python3.11/asyncio/streams.py", line 167, in _drain_helper
    raise ConnectionResetError('Connection lost')
ConnectionResetError: Connection lost
```

michelcve, 2024-03-01:
I have a similar problem, but the M5 echo seems to have crashed because of problems with the playback buffer. This is from the logging of the device:

```
ice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
[D][voice_assistant:347]: Speaker buffer full, trying again next loop
```

It makes sense, since I sometimes notice that the playback of the response has gaps in it.

mrtnkhl, 2024-03-02:
> Seeing the same issue. Running three Atom Echo's connecting to a Home Assistant Yellow. After a while all Atom Echos start blinking red once every couple seconds and the HA instance on the Yellow becomes unresponsive. Does not even react to reboot request via web interface. Only way to get it working again is to pull power from the Yellow and all Atom Echos. Then the setup works for a random period of time before the same issue reoccurs.
> 
> ESPHome 2023.12.9 HA Core 2024.2.2 Supervisor 2024.01.1 HA OS 11.5 openWakeWord 1.8.2
> 
> Nothing relevant in the logs when the Yellow starts to freeze. Deactivating openWakeWord add on and Wyoming integration leads to stable HA Yellow. No other voice assistants than the Atom Echos mentioned above are connected to the Yellow.

Issue persists on Home Assistant Yellow with
- HA Yellow
- 2 Atom Echo running ESPHome 2024.2.1
- HAOS 12.0
- Home Assistant Core 2024.2.5
- Supervisor 2024.02.1
- openWakeWord 1.10.0
- Frontend 20240207.1

After voice assistant commands work over a period of a few hours from enabling openWakeWord and Wyoming,
- HA UI becomes unresponsive, cannot even reboot through the UI (dialogue does not load)
- integrations that require external network connectivity (e.g. BMW Connected Drive, Buienradar, Electricity Maps) do not work anymore (gray on dashboard)
- UI starts to fail, e.g. cannot open Settings > About anymore (Error while loading page info.)

Purely a guess but maybe something is going wrong on the network level on the Yellow/HAOS, too many open connections initiated from the M5 Atoms make the network unresponsive and stall the kernel.

skyw33, 2024-05-21:
I did 2 things to get my atom echo to work.  I don't know what exactly the root cause is but maybe this will help someone?

I run esphome using docker.  First, I removed the ESPHOME_DASHBOARD_USE_PING=true environmental parameter for ESPHome running on Docker.  ESPHOME_DASHBOARD_USE_PING disables mDNS.

Once I did this, the esphome dashboard showed a 2nd new instance of the atom echo device and the new instance had an "adopt" button.  I first copied the yaml from the old instance and saved it to a text file.  Then I removed the old instance and adopted the new instance using the auto-generated yaml.

You shouldn't need to use it, but this is what the new yaml looks like for reference:
```
substitutions:
  name: atom-echo-kitchen-asdf
  friendly_name: Atom Echo Kitchen asdf
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: asdfasdfasdfasdfasdf
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

And this is what the old file that I replaced looked like:

```
---
substitutions:
  name: "atom-echo-kitchen"
  friendly_name: "Atom Echo Kitchen"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
  name_add_mac_suffix: true
  project:
    name: m5stack.atom-echo-voice-assistant
    version: "1.0"
  min_version: 2023.10.0
  on_boot:
    - priority: -100
      then:
        - wait_until: api.connected
        - delay: 1s
        - if:
            condition:
              switch.is_on: use_wake_word
            then:
              - voice_assistant.start_continuous:

esp32:
  board: m5stack-atom
  framework:
    type: esp-idf

logger:
api:
ota:

dashboard_import:
  package_import_url: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.1.43
  manual_ip:
    static_ip: 192.168.1.43
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  ap:

improv_serial:

i2s_audio:
  i2s_lrclk_pin: GPIO33
  i2s_bclk_pin: GPIO19

microphone:
  - platform: i2s_audio
    id: echo_microphone
    i2s_din_pin: GPIO23
    adc_type: external
    pdm: true

speaker:
  - platform: i2s_audio
    id: echo_speaker
    i2s_dout_pin: GPIO22
    dac_type: external
    mode: mono

voice_assistant:
  id: va
  microphone: echo_microphone
  speaker: echo_speaker
  noise_suppression_level: 2
  auto_gain: 31dBFS
  volume_multiplier: 2.0
  vad_threshold: 3
  on_listening:
    - light.turn_on:
        id: led
        blue: 100%
        red: 0%
        green: 0%
        brightness: 100%
        effect: pulse
  on_tts_start:
    - light.turn_on:
        id: led
        blue: 0%
        red: 0%
        green: 100%
        brightness: 100%
        effect: pulse
  on_end:
    - delay: 100ms
    - wait_until:
        not:
          speaker.is_playing:
    - script.execute: reset_led
  on_error:
    - light.turn_on:
        id: led
        blue: 0%
        red: 100%
        green: 0%
        brightness: 100%
        effect: none
    - delay: 1s
    - script.execute: reset_led
    - script.wait: reset_led
    - lambda: |-
        if (code == "wake-provider-missing" || code == "wake-engine-missing") {
          id(use_wake_word).turn_off();
        }

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO39
      inverted: true
    name: Button
    disabled_by_default: true
    entity_category: diagnostic
    id: echo_button
    on_click:
      - if:
          condition:
            switch.is_off: use_wake_word
          then:
            - if:
                condition: voice_assistant.is_running
                then:
                  - voice_assistant.stop:
                  - script.execute: reset_led
                else:
                  - voice_assistant.start:
          else:
            - voice_assistant.stop
            - delay: 1s
            - script.execute: reset_led
            - script.wait: reset_led
            - voice_assistant.start_continuous:

light:
  - platform: esp32_rmt_led_strip
    id: led
    name: None
    disabled_by_default: true
    entity_category: config
    pin: GPIO27
    default_transition_length: 0s
    chipset: SK6812
    num_leds: 1
    rgb_order: grb
    rmt_channel: 0
    effects:
      - pulse:
          transition_length: 250ms
          update_interval: 250ms

script:
  - id: reset_led
    then:
      - if:
          condition:
            - switch.is_on: use_wake_word
            - switch.is_on: use_listen_light
          then:
            - light.turn_on:
                id: led
                blue: 100%
                red: 100%
                green: 0%
                brightness: 100%
                effect: none
          else:
            - light.turn_off: led

switch:
  - platform: template
    name: Use wake word
    id: use_wake_word
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - lambda: id(va).set_use_wake_word(true);
      - if:
          condition:
            not:
              - voice_assistant.is_running
          then:
            - voice_assistant.start_continuous
      - script.execute: reset_led
    on_turn_off:
      - voice_assistant.stop
      - lambda: id(va).set_use_wake_word(false);
      - script.execute: reset_led
  - platform: template
    name: Use Listen Light
    id: use_listen_light
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    entity_category: config
    on_turn_on:
      - script.execute: reset_led
    on_turn_off:
      - script.execute: reset_led

external_components:
  - source: github://pr#5230
    components:
      - esp_adf
    refresh: 0s

esp_adf:
```

After doing these 2 things my atom echo has been rock solid.

smithimage, 2024-06-27:
I had a similar issue as this one, actually the exact same one as described in this issue https://github.com/esphome/issues/issues/5326

I have written my conclusion as a comment there : https://github.com/esphome/issues/issues/5326#issuecomment-2194652964

big-bigchen, 2024-07-20:
Hi,
After updating the ESPHome to 2024.7 nnd using the latest config from here (https://github.com/esphome/firmware/blob/main/voice-assistant/m5stack-atom-echo.yaml) my echo is not activated by default. Only if I tick on it (produce a loud input) the wake word listening is started. I tuned down these parameters already:
`voice_assistant:
  id: va
  microphone: echo_microphone
  speaker: echo_speaker
  noise_suppression_level: 0
  auto_gain: 0dBFS
  volume_multiplier: 4.0
  vad_threshold: 0
`
but no way to trigger the wake word detection without shouting at it... :-(
Before the update to latest ESPHome and older Echo Atom firmware it worked quite well...
Any ideas what to change?
Thanks!

skyw33, 2024-07-30:
After installing the yaml from the below link, my atom echo has been running for 3 days straight -- no restarts.
https://github.com/esphome/firmware/issues/227#issuecomment-2243440334

It uses the newly released micro wake word v2.

[1 more reply]

closed 2026-10-27 by rccoleman