esphome/issues #5039: hostByName(): DNS Failed with ethernet connection, esphome restarts after this error

zdenekchobot, 2023-10-26:
### The problem

<0x1b>[0;35m[C][http_request:014]: HTTP Request:<0x1b>[0m
<0x1b>[0;35m[C][http_request:015]:   Timeout: 10000ms<0x1b>[0m
<0x1b>[0;35m[C][http_request:016]:   User-Agent: esphome/wt32-eth01<0x1b>[0m
<0x1b>[0;35m[C][http_request:017]:   Follow Redirects: 1<0x1b>[0m
<0x1b>[0;35m[C][http_request:018]:   Redirect limit: 3<0x1b>[0m
**[  5932][E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for e9553n-2hakkm.tmep.cz
<0x1b>[0;36m[D][main:178]: Response status: -1, Durat
assert failed: xEventGroupSetBits event_groups.c:596 (xEventGroup)**

Backtrace:0x400836bd:0x3ffb35f00x4008c209:0x3ffb3610 0x400916cd:0x3ffb3630 0x4008c726:0x3ffb3760 0x40145a98:0x3ffb3780 0x400f747d:0x3ffb37a0 0x400f7539:0x3ffb37d0 0x400f7cc3:0x3ffb37f0 0x400fee69:0x3ffb3850 0x401020bf:0x3ffb38a0 0x401063e2:0x3ffb38d0 0x400f7179:0x3ffb38f0 

### Which version of ESPHome has the issue?

ESPHome 2023.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.10.5

### What platform are you using?

ESP32

### Board

wt32-eth01

### Component causing the issue

http_request

### Example YAML snippet

```yaml
esphome:
  name: wt32-eth01
  friendly_name: wt32-eth01

esp32:
  board: wt32-eth01
  framework:
    type: arduino

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:
  encryption:
    key: "W8apyx8y1zxepBWBO22U4s8xyX0OpjO/P7+2KuRpHqg="

ota:
  password: "179fde7cc36a876fd7651bc71e5c84b5"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

http_request:
  useragent: esphome/wt32-eth01
  timeout: 10s
  id: req001

interval:
  - interval: 10s
    then:
      - http_request.get:
          url: http://e9553n-2hakkm.tmep.cz/?temp=24.56
#          headers:
#            Content-Type: application/json
#          verify_ssl: false
          on_response:
            then:
              - logger.log:
                  format: 'Response status: %d, Duration: %u ms, Body: %s'
                  args:
                    - status_code
                    - duration_ms
                    - id(req001).get_string()
```

### Anything in the logs that might be useful for us?

```txt
<0x1b>[0;32m[I][logger:326]: Log initialized<0x1b>[0m
<0x1b>[0;35m[C][ota:473]: There have been 2 suspected unsuccessful boot attempts.<0x1b>[0m
<0x1b>[0;36m[D][esp32.preferences:114]: Saving 1 preferences to flash...<0x1b>[0m
<0x1b>[0;36m[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed<0x1b>[0m
<0x1b>[0;32m[I][app:029]: Running through setup()...<0x1b>[0m
<0x1b>[0;35m[C][ethernet:028]: Setting up Ethernet...<0x1b>[0m
<0x1b>[0;33m[W][http_request:037]: HTTP Request failed; Not connected to network<0x1b>[0m
<0x1b>[0;32m[I][ethernet:137]: Starting ethernet connection<0x1b>[0m
<0x1b>[0;32m[I][ethernet:148]: Connected via Ethernet!<0x1b>[0m
<0x1b>[0;35m[C][ethernet:353]:   IP Address: 0.0.0.0<0x1b>[0m
<0x1b>[0;35m[C][ethernet:354]:   Hostname: 'wt32-eth01'<0x1b>[0m
<0x1b>[0;35m[C][ethernet:355]:   Subnet: 0.0.0.0<0x1b>[0m
<0x1b>[0;35m[C][ethernet:356]:   Gateway: 0.0.0.0<0x1b>[0m
<0x1b>[0;35m[C][ethernet:361]:   DNS1: 192.168.254.5<0x1b>[0m
<0x1b>[0;35m[C][ethernet:362]:   DNS2: 0.0.0.0<0x1b>[0m
<0x1b>[0;35m[C][ethernet:382]:   MAC Address: 24:DC:C3:86:B6:E3<0x1b>[0m
<0x1b>[0;35m[C][ethernet:387]:   Is Full Duplex: YES<0x1b>[0m
<0x1b>[0;35m[C][ethernet:392]:   Link Speed: 100<0x1b>[0m
<0x1b>[0;35m[C][ota:097]: Over-The-Air Updates:<0x1b>[0m
<0x1b>[0;35m[C][ota:098]:   Address: wt32-eth01.local:3232<0x1b>[0m
<0x1b>[0;35m[C][ota:101]:   Using Password.<0x1b>[0m
<0x1b>[0;33m[W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts<0x1b>[0m
<0x1b>[0;35m[C][api:025]: Setting up Home Assistant API server...<0x1b>[0m
<0x1b>[0;32m[I][app:062]: setup() finished successfully!<0x1b>[0m
<0x1b>[0;32m[I][app:102]: ESPHome version 2023.10.3 compiled on Oct 26 2023, 22:40:54<0x1b>[0m
<0x1b>[0;35m[C][logger:416]: Logger:<0x1b>[0m
<0x1b>[0;35m[C][logger:417]:   Level: DEBUG<0x1b>[0m
<0x1b>[0;35m[C][logger:418]:   Log Baud Rate: 115200<0x1b>[0m
<0x1b>[0;35m[C][logger:420]:   Hardware UART: UART0<0x1b>[0m
<0x1b>[0;35m[C][ethernet:221]: Ethernet:<0x1b>[0m
<0x1b>[0;35m[C][ethernet:353]:   IP Address: 192.168.254.167<0x1b>[0m
<0x1b>[0;35m[C][ethernet:354]:   Hostname: 'wt32-eth01'<0x1b>[0m
<0x1b>[0;35m[C][ethernet:355]:   Subnet: 255.255.255.0<0x1b>[0m
<0x1b>[0;35m[C][ethernet:356]:   Gateway: 192.168.254.1<0x1b>[0m
<0x1b>[0;35m[C][ethernet:361]:   DNS1: 192.168.254.5<0x1b>[0m
<0x1b>[0;35m[C][ethernet:362]:   DNS2: 0.0.0.0<0x1b>[0m
<0x1b>[0;35m[C][ethernet:382]:   MAC Address: 24:DC:C3:86:B6:E3<0x1b>[0m
<0x1b>[0;35m[C][ethernet:387]:   Is Full Duplex: YES<0x1b>[0m
<0x1b>[0;35m[C][ethernet:392]:   Link Speed: 100<0x1b>[0m
<0x1b>[0;35m[C][ethernet:224]:   Power Pin: 16<0x1b>[0m
<0x1b>[0;35m[C][ethernet:226]:   MDC Pin: 23<0x1b>[0m
<0x1b>[0;35m[C][ethernet:227]:   MDIO Pin: 18<0x1b>[0m
<0x1b>[0;35m[C][ethernet:228]:   Type: LAN8720<0x1b>[0m
<0x1b>[0;35m[C][ethernet:229]:   PHY addr: 1<0x1b>[0m
<0x1b>[0;35m[C][mdns:115]: mDNS:<0x1b>[0m
<0x1b>[0;35m[C][mdns:116]:   Hostname: wt32-eth01<0x1b>[0m
<0x1b>[0;35m[C][ota:097]: Over-The-Air Updates:<0x1b>[0m
<0x1b>[0;35m[C][ota:098]:   Address: wt32-eth01.local:3232<0x1b>[0m
<0x1b>[0;35m[C][ota:101]:   Using Password.<0x1b>[0m
<0x1b>[0;33m[W][ota:107]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts<0x1b>[0m
<0x1b>[0;35m[C][api:138]: API Server:<0x1b>[0m
<0x1b>[0;35m[C][api:139]:   Address: wt32-eth01.local:6053<0x1b>[0m
<0x1b>[0;35m[C][api:141]:   Using noise encryption: YES<0x1b>[0m
<0x1b>[0;35m[C][http_request:014]: HTTP Request:<0x1b>[0m
<0x1b>[0;35m[C][http_request:015]:   Timeout: 10000ms<0x1b>[0m
<0x1b>[0;35m[C][http_request:016]:   User-Agent: esphome/wt32-eth01<0x1b>[0m
<0x1b>[0;35m[C][http_request:017]:   Follow Redirects: 1<0x1b>[0m
<0x1b>[0;35m[C][http_request:018]:   Redirect limit: 3<0x1b>[0m
[  5932][E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for e9553n-2hakkm.tmep.cz
<0x1b>[0;36m[D][main:178]: Response status: -1, Durat
assert failed: xEventGroupSetBits event_groups.c:596 (xEventGroup)

Backtrace:0x400836bd:0x3ffb35f00x4008c209:0x3ffb3610 0x400916cd:0x3ffb3630 0x4008c726:0x3ffb3760 0x40145a98:0x3ffb3780 0x400f747d:0x3ffb37a0 0x400f7539:0x3ffb37d0 0x400f7cc3:0x3ffb37f0 0x400fee69:0x3ffb3850 0x401020bf:0x3ffb38a0 0x401063e2:0x3ffb38d0 0x400f7179:0x3ffb38f0
```

### Additional information

DNS works ok. With wifi component/connection everything works perfectly.

ssieb, 2023-10-26:
The crash is most likely because you're trying to access `id(req001).get_string()` when the request failed.

zdenekchobot, 2023-10-27:
> The crash is most likely because you're trying to access `id(req001).get_string()` when the request failed.

The crash probably occurs earlier. After removing "id(req001).get_string()", the behavior is the same.

<0x1b>[0;35m[C][http_request:014]: HTTP Request:<0x1b>[0m
<0x1b>[0;35m[C][http_request:015]: Timeout: 10000ms<0x1b>[0m
<0x1b>[0;35m[C][http_request:016]: User-Agent: esphome/wt32-eth01<0x1b>[0m
<0x1b>[0;35m[C][http_request:017]: Follow Redirects: 1<0x1b>[0m
<0x1b>[0;35m[C][http_request:018]: Redirect limit: 3<0x1b>[0m
<0x1b>[0;36m[D][api:102]: Accepted 192.168.254.10<0x1b>[0m
<0x1b>[0;36m[D][api.connection:1032]: ESPHome Logs 2023.10.3 (192.168.254.10): Connected successfully<0x1b>[0m
[ 7996][E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for [e9553n-2hakkm.tmep.cz](http://e9553n-2hakkm.tmep.cz/)
<0x1b>[0;36m[D][main:176]: Response status: -1, Du
assert failed: xEventGroupSetBits event_groups.c:596 (xEventGroup)

Backtrace:0x400836bd:0x3ffb35f00x4008c209:0x3ffb3610 0x400916cd:0x3ffb3630 0x4008c726:0x3ffb3760 0x40145354:0x3ffb3780 0x400f6f1d:0x3ffb37a0 0x400f6fd9:0x3ffb37d0 0x400f7763:0x3ffb37f0 0x400fe909:0x3ffb3850 0x40101b5f:0x3ffb38a0 0x40105e82:0x3ffb38d0 0x400f6c19:0x3ffb38f0

ssieb, 2023-10-27:
What's the new `on_response:` section?
Can you connect this device directly to esphome so that the stack trace will get decoded?

zdenekchobot, 2023-10-27:
> What's the new `on_response:` section? 
          on_response:
            then:
              - logger.log:
                  format: 'Response status: %d, Duration: %u ms,'
                  args:
                    - status_code
                    - duration_ms

> Can you connect this device directly to esphome so that the stack trace will get decoded?
I can - in the evening. But what exactly do you mean? The board is connected via ethernet (cable) to the LAN and managed by the esphome plug-in in HA.

ssieb, 2023-10-27:
Somehow you are getting serial logs now.  But you need esphome to read those serial logs so that it can decode the stack trace.  We need that to find out what's crashing.

zdenekchobot, 2023-10-27:
> Somehow you are getting serial logs now. But you need esphome to read those serial logs so that it can decode the stack trace. We need that to find out what's crashing.

[image]
or
[image]
or something else?

ssieb, 2023-10-27:
The first one, but the ESP needs to be plugged into the system running esphome/HA.  You need to choose the option for "device plugged into the system running esphome".

zdenekchobot, 2023-10-27:
> The first one, but the ESP needs to be plugged into the system running esphome/HA. You need to choose the option for "device plugged into the system running esphome".

Thank you, I will send the log in the afternoon or evening

zdenekchobot, 2023-10-27:
> The first one, but the ESP needs to be plugged into the system running esphome/HA. You need to choose the option for "device plugged into the system running esphome".

Log attached
[logs_wt32-eth01.txt](https://github.com/esphome/issues/files/13187671/logs_wt32-eth01.txt)

ssieb, 2023-10-27:
It's all internal to lwip and freertos:
```
[11:13:51][  5369][E][WiFiGeneric.c
[11:13:51]assert failed: xEventGroupSetBits event_groups.c:596 (xEventGroup)
[11:13:51]
[11:13:51]
[11:13:51]Backtrace:0x400836bd:0x3ffb35f00x4008c209:0x3ffb3610 0x400916cd:0x3ffb3630 0x4008c726:0x3ffb3760 0x40148154:0x3ffb3780 0x400f9c19:0x3ffb37a0 0x400f9cd5:0x3ffb37d0 0x400fa45f:0x3ffb37f0 0x40101605:0x3ffb3850 0x4010485b:0x3ffb38a0
 0x40108b7e:0x3ffb38d0 0x400f9915:0x3ffb38f0 
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400836bd: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
WARNING Decoded 0x4008c209: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
WARNING Decoded 0x400916cd: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
WARNING Decoded 0x4008c726: xEventGroupSetBits at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/event_groups.c:597 (discriminator 1)
WARNING Decoded 0x40148154: wifi_dns_found_callback(char const*, ip_addr const*, void*) at /data/cache/platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1413
WARNING Decoded 0x400f9c19: dns_call_found at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/dns.c:999
WARNING Decoded 0x400f9cd5: dns_correct_response at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/dns.c:1197
WARNING Decoded 0x400fa45f: dns_recv at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/dns.c:1364
WARNING Decoded 0x40101605: udp_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/udp.c:404
WARNING Decoded 0x4010485b: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c:794
WARNING Decoded 0x40108b7e: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c:186
WARNING Decoded 0x400f9915: tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:180
 (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154
```

zdenekchobot, 2023-11-05:
> It's all internal to lwip and freertos:
>
> ```
> [11:13:51][  5369][E][WiFiGeneric.c
> [11:13:51]assert failed: xEventGroupSetBits event_groups.c:596 (xEventGroup)
> [11:13:51]
> [11:13:51] [...]

😒 How to fix it?

Tarik2142, 2024-10-17:
not stale, i have same error

J3bs, 2024-10-31:
+1 also experiencing this issue, same board/platform/result.

As a test, I tried switching to wifi rather than ethernet and that allowed http_reqyest to work as expected. Sadly not a viable solution for my use case tho

J3bs, 2024-10-31:
> Thanks, that adds weight to my growing suspicion it is relying on the WiFi libraries; I will see if I can code something in C++ to work around this for Ethernet...

Ha I started combing through the http_request componant last night with the same suspicion. Haven't found anything yet, but will post back if/when I do.

I'm a network admin, so I certainly wouldn't be able to fix this, but hoping if I can find the root cause perhaps someone with the right skillset would help? :)

J3bs, 2024-10-31:
> So looking through the libraries and ESPhome API docs lead me here: https://esphome.io/api/http__request__arduino_8h_source
> 
> This describes an object that inherits [http_container](https://esphome.io/api/classesphome_1_1http__request_1_1_http_container) and has a member of type HTTPClient .
> 
> When starting a request, the client ultimately appears to being with a string/char array url: `bool status = container->client_.begin(url.c_str());`
> 
> That's all I have time for tonight; I will keep digging further down the rabbit hole, hopefully the `begin()` method will highlight how it requests a DNS resolution. It also isn't clear to me yet how it passes this to the Ethernet/WiFi object (there is a global one created when ESPHome initialises the component)

Good work sir. I spent a bit of time on it today, mainly information gathering for the moment but wanted to put my findings here incase they are useful to someone more knoledgable than me....

So I did a regex match against all files in the http_request component's folder searching for 'ethernet|wifi' - as expected, I found more matches for wifi (14), then i did for ethernet (0). Matches w/ files & line numbers below

http_request\__init__.py, line 164:                 		cg.add_library("WiFiClientSecure", None)
http_request_arduino.cpp, line 42:       					std::unique_ptr<WiFiClient> stream_ptr;
http_request\http_request_arduino.cpp, line 45:         	ESP_LOGV(TAG, "ESP8266 HTTPS connection with WiFiClientSecure");
http_request\http_request_arduino.cpp, line 46:         	stream_ptr = std::make_unique<WiFiClientSecure>();
http_request\http_request_arduino.cpp, line 47:         	WiFiClientSecure *secure_client = static_cast<WiFiClientSecure *>(stream_ptr.get());
http_request\http_request_arduino.cpp, line 51:         	stream_ptr = std::make_unique<WiFiClient>();
http_request\http_request_arduino.cpp, line 54:       		ESP_LOGV(TAG, "ESP8266 HTTP connection with WiFiClient");
http_request\http_request_arduino.cpp, line 59:       		stream_ptr = std::make_unique<WiFiClient>();
http_request\http_request_arduino.cpp, line 133:       		WiFiClient *stream_ptr = this->client_.getStreamPtr();
http_request\http_request_arduino.h, line 12:     			#include <WiFiClientSecure.h>
http_request\http_request.h, line 112:       				float get_setup_priority() const override { return setup_priority::AFTER_WIFI; }
http_request\ota\ota_http_request.h, line 28:       		float get_setup_priority() const override { return setup_priority::AFTER_WIFI; }
http_request\update\http_request_update.h, line 25:       	float get_setup_priority() const override { return setup_priority::AFTER_WIFI; }

Tarik2142, 2024-11-08:
> So after a week playing with this, I have had zero luck; all libraries seem to use WiFi. I have confirmed mDNS is working over WiFi and Ethernet to my nodes via Windoze and Linux workstations and direct IP addresses work fine.
> 
> I will be using IP addresses instead of host names until this gets fixed.

The arduino core less than 3.0 has all network functions in the WiFi class, but this does not mean that these functions do not work over ethernet, it is more of a naming problem because WiFiClient can work with both WiFi and ethernet.
I use hostByName() in my code and it works fine with both WiFi and ethernet, so it's more of a bug in esphome.
Without decoding the crash dump, I don't think it will be possible to find the reason

Mywk, 2025-01-02:
Bump, still an issue.