esphome/issues #5042: Updating text_sensor from Lambda

jtricerolph, 2023-10-27:
### The problem

Unless im missing something stupid acording to docs for text_sensor you should be able to update/publish a new value from with lambda as below

```
publish_state(): Manually cause the sensor to push out a value.

// Within lambda, push a value of "Hello World"
id(my_sensor).publish_state("Hello World");
```

so using;
```
text_sensor:
  - platform: template
    name: "Gate RF Signal Received"
    id: rf_gate_signal
    on_value:
      lambda: !lambda |-
        id(gate_triggered_method).state_publish("RF");
```

but i get;

```
/config/esphome/esp32-bluetooth-proxy-7dac00.yaml:63:32: error: 'class esphome::template_::TemplateTextSensor' has no member named 'state_publish'
           id(gate_triggered_method).state_publish("RF");
```

### Which version of ESPHome has the issue?

Current version: 2023.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Home Assistant 2023.10.3 Supervisor 2023.10.1 Operating System 11.0 Frontend 20231005.0 - latest

### What platform are you using?

ESP32-IDF

### Board

wroom

### Component causing the issue

text_sensor

### Example YAML snippet

```yaml
text_sensor:
  - platform: template
    name: "Gate RF Signal Received"
    id: rf_gate_signal
    on_value:
      lambda: !lambda |-
        id(gate_triggered_method).state_publish("RF");
```

### Anything in the logs that might be useful for us?

```txt
/config/esphome/esp32-bluetooth-proxy-7dac00.yaml:63:32: error: 'class esphome::template_::TemplateTextSensor' has no member named 'state_publish'
           id(gate_triggered_method).state_publish("RF");
```

jtricerolph, 2023-10-27:
sorry if wrong place to query this.

kvj, 2023-10-29:
Why `state_publish()` and not `publish_state()`?

ssieb, 2023-10-29:
This is the wrong place and as discussed on discord, he did realize that he typed it wrong.

closed 2023-10-29 by ssieb