esphome/issues #5044: IPv4 struct compile error on older Arduino framework for FastLED (2023.10 regression)

TheJulianJES, 2023-10-28:
### The problem

See the example YAML below and compile error. This seems to be a regression in ESPHome 2023.10.3, as 2023.9.1 was still fine.

FastLED requires Arduino framework 2.7.4 on ESP8266, but it looks like this version is no longer compatible with the latest ESPHome release.

### Which version of ESPHome has the issue?

2023.10.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

NodeMCU

### Example YAML snippet

```yaml
substitutions:
  device_name: "led8"

esphome:
  name: ${device_name}
  comment: "LED Test"
  on_boot:
    - lambda: |-
        id(my_ota).set_auth_password("zAbdHVVrlHYzX5i4H60qwHvlMUk0GqKxEW/O+XckhZA=");

esp8266:
  board: nodemcuv2
  framework:
    version: 2.7.4 # Check https://esphome.io/components/light/fastled.html later

logger:

api:
  encryption:
    key: "lg6uJ/pcK7FJjTRGFgpUK2ghDg9j+BcCBqgkxQbuSp0="
  # reboot_timeout: 30min

ota:
  password: !secret ota_password
  id: my_ota

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: HIGH # for ESP8266 LOW/HIGH are mixed up https://github.com/esphome/issues/issues/1532

button:
  - platform: restart
    name: "${device_name} Restart"

light:
  - platform: fastled_clockless
    name: "${device_name} Light"
    id: leds
    chipset: WS2811
    pin: D3
    num_leds: 50
    rgb_order: BRG
    gamma_correct: 1.0
    effects:
      - random:
      - pulse:
      - strobe:
      - flicker:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
```

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs/led8/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:31:0,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/glue.h:53,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/arch/cc.h:61,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/arch.h:48,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwipopts.h:3608,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/ip_addr.h:40,
                 from src/esphome/components/network/ip_address.h:6,
                 from src/esphome/components/network/util.h:4,
                 from src/esphome/components/network/util.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/ipv4_addr.h:53:22: error: field 'ip' has incomplete type
     struct ipv4_addr ip;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/ipv4_addr.h:54:22: error: field 'netmask' has incomplete type
     struct ipv4_addr netmask;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/ipv4_addr.h:55:22: error: field 'gw' has incomplete type
     struct ipv4_addr gw;
                      ^
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/glue.h:53:0,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/arch/cc.h:61,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/arch.h:48,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/debug.h:40,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwipopts.h:3608,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/opt.h:51,
                 from /data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/lwip2/include/lwip/ip_addr.h:40,
                 from src/esphome/components/network/ip_address.h:6,
                 from src/esphome/components/network/util.h:4,
                 from src/esphome/components/network/util.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:363:22: error: field 'ip' has incomplete type
     struct ipv4_addr ip;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:368:22: error: field 'start_ip' has incomplete type
     struct ipv4_addr start_ip;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:369:22: error: field 'end_ip' has incomplete type
     struct ipv4_addr end_ip;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:532:22: error: field 'ip' has incomplete type
     struct ipv4_addr ip;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:533:22: error: field 'mask' has incomplete type
     struct ipv4_addr mask;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:534:22: error: field 'gw' has incomplete type
     struct ipv4_addr gw;
                      ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/tools/sdk/include/user_interface.h:544:22: error: field 'ip' has incomplete type
     struct ipv4_addr ip;
                      ^
Compiling .pioenvs/led8/src/esphome/components/ota/ota_backend_arduino_libretiny.cpp.o
Compiling .pioenvs/led8/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling .pioenvs/led8/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
Compiling .pioenvs/led8/src/esphome/components/ota/ota_component.cpp.o
*** [.pioenvs/led8/src/esphome/components/network/util.cpp.o] Error 1
========================== [FAILED] Took 6.21 seconds ==========================
```

### Additional information

Last confirmed working version: 2023.9.1 (and probably all other 2023.9.x versions too)

TheJulianJES, 2023-10-28:
Possibly related to @HeMan's "ip address representation" PR: https://github.com/esphome/esphome/pull/5252? (not yet confirmed though)

I know it's not optimal to use an outdated Arduino framework version, but as long as the docs mention that it's required (and okay to use) for FastLED, this should not throw an error probably.
Alternative would be to remove FastLED support for ESP8266 completely(?)

HeMan, 2023-10-28:
Seems like it is my PR, and it looks like it's a simple fix. Making a new PR now.

closed 2023-10-29 by jesserockz