esphome/issues #5048: Pico W takes installer but does not boot

allw1994, 2023-10-29:
### The problem

Using ESPHome addon within HomeAssistant, creating a new RPi device and moving the image over to the pico seems to work fine, but then the device never shows as connected to the WiFi.

If I go to examine the logs then the device shows up as plugged into the local PC but connecting gives a completely blank screen - downloading yields a file of 0B.

The Pico works fine with other firmware and has been connected to the same WiFi in the same location.

Thonny reports device busy or did not respond - seems to be the only "logging" I can get.

### Which version of ESPHome has the issue?

2023.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

RP2040

### Board

Pico W

### Component causing the issue

Unknown

### Example YAML snippet

```yaml
INFO ESPHome 2023.10.3
INFO Reading configuration /config/esphome/homenetwork-iot.yaml...
esphome:
  name: homenetwork-iot
  friendly_name: HOMENETWORK_IoT
  build_path: build/homenetwork-iot
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
  min_version: 2023.10.3
rp2040:
  board: rpipicow
  framework:
    platform_version: https://github.com/maxgerhardt/platform-raspberrypi.git
    version: 3.4.0
    source: https://github.com/earlephilhower/arduino-pico/releases/download/3.4.0/rp2040-3.4.0.zip
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: USB_CDC
  level: DEBUG
  logs: {}
api:
  encryption:
    key: <redacted>
  port: 6053
  password: ''
  reboot_timeout: 15min
ota:
  password: <redacted>
  safe_mode: true
  port: 2040
  reboot_timeout: 5min
  num_attempts: 10
wifi:
  ap:
    ssid: Homenetwork-Iot Fallback Hotspot
    password: tJaM6FLult7g
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  passive_scan: false
  enable_on_boot: true
  networks:
  - ssid: HOMENETWORK_IoT
    password: '<redacted>'
    priority: 0.0
  use_address: homenetwork-iot.local

INFO Configuration is valid!
```

### Anything in the logs that might be useful for us?

```txt
Logs are blank.
```

ssieb, 2023-10-29:
Try setting a static IP address.
Do you have a USB uart adapter?

allw1994, 2023-10-29:
> Try setting a static IP address.

Worked first time - You're a star!

Know where to start now with getting DHCP DNS working - thanks.

closed 2023-10-29 by allw1994

ssieb, 2023-10-29:
https://github.com/esphome/issues/issues/4110