esphome/issues #5049: Docker container cannot install ltchiptool and fails

blakadder, 2023-10-29:
### The problem

When trying to build a libretuya device from docker container it fails because it cannot install ltchiptool
```
INFO ESPHome 2023.11.0-dev
INFO Reading configuration /config/zg-205w.yaml...
INFO Detected timezone 'Europe/Zagreb'
INFO Generating C++ source...
INFO Compiling app...
Processing zg-205w (board: generic-bk7231n-qfn32-tuya; framework: arduino; platform: libretiny @ 1.4.1)
--------------------------------------------------------------------------------
Installing/updating ltchiptool
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
!!! Installing ltchiptool failed, or version outdated. Please install ltchiptool manually using pip. Cannot continue. <member 'name' of 'ImportError' objects>: No module named 'ltchiptool'
UnknownPlatform: Unknown development platform 'libretiny'
```

### Which version of ESPHome has the issue?

 2023.11.0-dev

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2023.12dev

### What platform are you using?

BK72XX

### Board

CBU

### Component causing the issue

ltchiptool

fightforlife, 2023-11-12:
I now also habe this problem on esp8285. Although I do not even have libretiny defined anywhere.
```
INFO ESPHome 2023.11.0b3
(board: esp8285; framework: arduino; platform: platformio/espressif8266@3.2.0)
No module named 'ltchiptool'
UnknownPlatform: Unknown development platform 'libretiny'
```

FARKIr, 2023-11-15:
me to 
[image]

fakuivan, 2023-11-16:
Delete the directory `.esphome`, that worked for me.

kuba2k2, 2023-11-16:
@blakadder @fightforlife @FARKIr 
This error is caused by old version of LibreTiny being installed.
Installing ltchiptool to a virtual environment has been addressed before v1.4.0 and before merging with ESPHome.
Please clear `/config/.esphome/` directory as mentioned above.