esphome/issues #5051: After upgrade to 10.4 display component don't compile

FrColin, 2023-10-30:
### The problem

esphome stop compiling when display component ssd1306_i2c present in yaml 
`In file included from src/esphome/components/display/display_buffer.h:6,
                 from src/esphome/components/display/image.cpp:4:
src/esphome/components/display/display.h:153:7: error: redefinition of 'class esphome::display::BaseImage'
 class BaseImage {
       ^~~~~~~~~
In file included from src/esphome/components/display/image.cpp:1:
src/esphome/components/display/image.h:40:7: note: previous definition of 'class esphome::display::BaseImage'
 class BaseImage {
       ^~~~~~~~~
*** [.pioenvs\linkytic\src\esphome\components\display\image.cpp.o] Error 1`

### Which version of ESPHome has the issue?

2023.10.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

wemos_d1_mini32

### Component causing the issue

display

### Example YAML snippet

```yaml
font:
  - file: "fonts/ComicSansMSRegular.ttf"
    id: arial
    size: 15
display:
  - platform: ssd1306_i2c
    id: my_display
    model: "SSD1306 128x64"
    address: 0x3C
    brightness: 100%
    pages:
      - id: page1
        lambda: |-
```

### Anything in the logs that might be useful for us?

```txt
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (5.4.0) > WEMOS D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
|-- ArduinoJson @ 6.18.5
|-- NeoPixelBus @ 2.7.3
|-- WiFiClientSecure @ 2.0.0
|-- HTTPClient @ 2.0.0
Building in release mode
Compiling .pioenvs\linkytic\src\esphome\components\display\image.cpp.o
Compiling .pioenvs\linkytic\src\esphome\components\font\font.cpp.o
Compiling .pioenvs\linkytic\src\esphome\components\gpio\switch\gpio_switch.cpp.o
Compiling .pioenvs\linkytic\src\esphome\components\homeassistant\sensor\homeassistant_sensor.cpp.o
Compiling .pioenvs\linkytic\src\esphome\components\homeassistant\time\homeassistant_time.cpp.o
Compiling .pioenvs\linkytic\src\esphome\components\http_request\http_request.cpp.o
In file included from src/esphome/components/display/display_buffer.h:6,
                 from src/esphome/components/display/image.cpp:4:
src/esphome/components/display/display.h:153:7: error: redefinition of 'class esphome::display::BaseImage'
 class BaseImage {
       ^~~~~~~~~
In file included from src/esphome/components/display/image.cpp:1:
src/esphome/components/display/image.h:40:7: note: previous definition of 'class esphome::display::BaseImage'
 class BaseImage {
       ^~~~~~~~~
*** [.pioenvs\linkytic\src\esphome\components\display\image.cpp.o] Error 1
================================================================== [FAILED] Took 6.57 seconds ==================================================================
PS D:\ESPHOME> esphome version
Version: 2023.10.4
PS D:\ESPHOME>
```

### Additional information

I try on esp8266 and esp32 with the same results

ssieb, 2023-10-31:
image is not in display any more, so something is wrong there.  Try a clean build.  Do you have any external components or a `custom_components` directory?

FrColin, 2023-10-31:
I allready try a clean build without success.
I have a custom components , but it as nothing to do with display.
I think two include file include the base display witch define the class.
I would try a compilé whitout thé custom component to see if something change.
I will send you black a message of my test

FrColin, 2023-10-31:
I tested whitout custom component with same error after a clean build and remove of src/esphome
`In file included from src\esphome\components\display\display_buffer.h:6,
                 from src\esphome\components\display\image.cpp:4:
src\esphome\components\display\display.h:153:7: error: redefinition of 'class esphome::display::BaseImage'
  153 | class BaseImage {
      |       ^~~~~~~~~
In file included from src\esphome\components\display\image.cpp:1:
src\esphome\components\display\image.h:40:7: note: previous definition of 'class esphome::display::BaseImage'
   40 | class BaseImage {
      |       ^~~~~~~~~
Compiling .pioenvs\oled\src\esphome\components\esp8266\preferences.cpp.o
*** [.pioenvs\oled\src\esphome\components\display\image.cpp.o] Error 1`

the class BaseImage is declared into two files image.h and display.h

FrColin, 2023-10-31:
image is still in display repository:
[image]

ssieb, 2023-10-31:
What is that screenshot from?
It's not there: https://github.com/esphome/esphome/tree/dev/esphome/components/display

FrColin, 2023-10-31:
The screen shoot is from the .pioenv/SRC/esphome...

ssieb, 2023-10-31:
Then you have old data somewhere because it's really not there.
If you need more help, come ask on the esphome discord server.

closed 2023-10-31 by ssieb

FrColin, 2023-11-01:
I remove the repository but the file is regenerated even if i remove image.h and image.cpp those file are recreated!