esphome/issues #5054: BME280 Humidity problem after update to 2023.10.4

pepsonEL, 2023-10-31:
### The problem

Hi. When use ESPHome 2023.10.3 all works ok and temp, pressure and humidity read no problem from BME280. Today after update to ESPHome 2023.10.4 no read humidity. Temp and pressure works.

### Which version of ESPHome has the issue?

2023.10.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

i2c bme280

### Example YAML snippet

```yaml
sensor:
  - platform: bme280
    temperature:
      name: "Temperatura"
      id: temperatura
      oversampling: 16x
    pressure:
      name: "Ciśnienie"
      id: ciscnienie
    humidity:
      name: "Wilgotność"
      id: wilgotnosc
    address: 0x76
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[08:50:28][C][bme280.sensor:197]: Humidity 'Wilgotność' 
[08:50:28][C][bme280.sensor:197]: Device Class: 'humidity' 
[08:50:28][C][bme280.sensor:197]: State Class: 'measurement' 
[08:50:28][C][bme280.sensor:197]: Unit of Measurement: '%' 
[08:50:28][C][bme280.sensor:197]: Accuracy Decimals: 1 
[08:50:28][C][bme280.sensor:198]: Oversampling: 16x

[08:50:29][D][sensor:093]: 'Temperatura': Sending state 15.65000 °C with 1 decimals of accuracy [08:50:29][D][sensor:093]: 'Ciśnienie': Sending state 975.21729 hPa with 1 decimals of accuracy [08:50:29][D][sensor:093]: 'Wilgotność': Sending state nan % with 1 decimals of accuracy
```

JamieR007, 2024-01-21:
I was seeing exactly the same behavior using 2023.12.5. Updating to 2023.12.8 seems to resolve the issue. :)

Hagakurje, 2024-02-14:
Same problem. After reboot humidity works, sometimes 1 hour, sometimes 1 week. Version 2023.12.9.

IvanBye, 2024-06-12:
Same issue after update ****