esphome/issues #5055: framework 2.7.4 (required for fastLED) fails compile

ldeffenb, 2023-10-31:
### The problem

I'm trying to use the fastLED package to support a 2812LED.  This package requires framework 2.7.4.   But when this framework is specified, various components (mqtt being the latest) fail to compile due to IP-related errors (see below).

### Which version of ESPHome has the issue?

2023.10.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP8266

### Board

Homebrew 8266-based

### Component causing the issue

mqtt

### Example YAML snippet

```yaml
esphome:
  name: "lp-temp1"
  friendly_name: lp-temp1

esp8266:
  board: esp01_1m
  framework:
    version: 2.7.4  # Required for FastLED support

# Enable Home Assistant API
#api:
#  encryption:
#    key: "iO/3hh8Z0p76ge1l6sz7Ctqka8FTW2l7O51w1amuaaU="

mqtt:
  broker: 192.168.10.4
  topic_prefix: "esphome/lp-temp1"
  discovery: true

# Example configuration entry
light:
  - platform: fastled_clockless
    chipset: WS2812B
    pin: GPIO16
    num_leds: 1
    rgb_order: GRB
    name: "FastLED WS2812 Light"
    
binary_sensor:
  - platform: status
    name: "LP Temp1 Status"
```

### Anything in the logs that might be useful for us?

```txt
A bunch of items compile cleanly and then:

Compiling .pioenvs/lp-temp1/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o
In file included from src/esphome/components/mqtt/mqtt_client.h:15:0,
                 from src/esphome/components/mqtt/custom_mqtt_device.h:7,
                 from src/esphome/components/mqtt/custom_mqtt_device.cpp:1:
src/esphome/components/mqtt/mqtt_backend_esp8266.h: In member function 'virtual void esphome::mqtt::MQTTBackendESP8266::set_server(esphome::network::IPAddress, uint16_t)':
src/esphome/components/mqtt/mqtt_backend_esp8266.h:22:100: error: call of overloaded 'IPAddress(esphome::network::IPAddress&)' is ambiguous
   void set_server(network::IPAddress ip, uint16_t port) final { mqtt_client_.setServer(IPAddress(ip), port); }
                                                                                                    ^
src/esphome/components/mqtt/mqtt_backend_esp8266.h:22:100: note: candidates are:
In file included from src/esphome/components/network/ip_address.h:14:0,
                 from src/esphome/components/mqtt/mqtt_client.h:11,
                 from src/esphome/components/mqtt/custom_mqtt_device.h:7,
                 from src/esphome/components/mqtt/custom_mqtt_device.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/IPAddress.h:163:9: note: IPAddress::IPAddress(const ip4_addr&)
         IPAddress(const ipv4_addr& fw_addr)   { setV4(); v4() = fw_addr.addr; }
         ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/IPAddress.h:79:9: note: IPAddress::IPAddress(const IPAddress&)
         IPAddress(const IPAddress& from);
         ^
In file included from src/esphome/components/mqtt/mqtt_client.h:15:0,
                 from src/esphome/components/mqtt/mqtt_component.h:11,
                 from src/esphome/components/mqtt/mqtt_binary_sensor.h:7,
                 from src/esphome/components/mqtt/mqtt_binary_sensor.cpp:1:
src/esphome/components/mqtt/mqtt_backend_esp8266.h: In member function 'virtual void esphome::mqtt::MQTTBackendESP8266::set_server(esphome::network::IPAddress, uint16_t)':
src/esphome/components/mqtt/mqtt_backend_esp8266.h:22:100: error: call of overloaded 'IPAddress(esphome::network::IPAddress&)' is ambiguous
   void set_server(network::IPAddress ip, uint16_t port) final { mqtt_client_.setServer(IPAddress(ip), port); }
                                                                                                    ^
src/esphome/components/mqtt/mqtt_backend_esp8266.h:22:100: note: candidates are:
In file included from src/esphome/components/network/ip_address.h:14:0,
                 from src/esphome/components/mqtt/mqtt_client.h:11,
                 from src/esphome/components/mqtt/mqtt_component.h:11,
                 from src/esphome/components/mqtt/mqtt_binary_sensor.h:7,
                 from src/esphome/components/mqtt/mqtt_binary_sensor.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/IPAddress.h:163:9: note: IPAddress::IPAddress(const ip4_addr&)
         IPAddress(const ipv4_addr& fw_addr)   { setV4(); v4() = fw_addr.addr; }
         ^
/data/cache/platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/IPAddress.h:79:9: note: IPAddress::IPAddress(const IPAddress&)
         IPAddress(const IPAddress& from);
         ^
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/lp-temp1/src/esphome/components/mqtt/custom_mqtt_device.cpp.o] Error 1
At global scope:
cc1plus: warning: unrecognized command line option "-Wno-nonnull-compare" [enabled by default]
*** [.pioenvs/lp-temp1/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o] Error 1
```

nielsnl68, 2023-10-31:
Please have a look at my pr: https://github.com/esphome/esphome/pull/4257
this should fix some of the issues, i hope.

ldeffenb, 2023-10-31:
@nielsnl68 Actually, I think https://github.com/esphome/esphome/pull/5639 will likely fix my particular compilation errors.   I changed from mqtt to api and it was clean, so my bust is actually in mqtt.

closed 2023-10-31 by jesserockz