esphome/issues #5057: website

ThomasFTL, 2023-10-31:
### The problem

Because the sensors from ESPhome are only visible in HA if they have a name, it would be very important that there is an indicator as to whether the sensor should be visible on the ESPhome website.
In addition, it would be nice if you could define the order of the sensors and empty lines so that the website remains clear and readable when using several auxiliary sensors.

### Which version of ESPHome has the issue?

2023.10.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.5

### What platform are you using?

ESP8266

### Board

ESP8266 f12

### Component causing the issue

web site

ssieb, 2023-10-31:
What do you mean by an "indicator"?  If the component is internal, it won't be shown unless the web server is configured to show internal components.  I believe the displayed order is alphabetical.

ThomasFTL, 2023-11-01:
I understood this to mean that internal components are not visible under HA either. But that's exactly where I need it.

Unfortunately, the order is alphabetical and not logically sorted and grouped.

In addition, each numerical sensor has a slider or an edit box.
Unfortunately, the “display only” status does not exist. I need the sensor function at this point. But I don't want to allow manual changes via the website.

ssieb, 2023-11-01:
I don't understand what you mean and this really isn't an issue.  You can file feature requests if you want, but you need to be more clear about what you're asking and how it would be done.
The website is not intended to be a fancy interface.  It's mainly a basic interaction that shows everything that's available.  You can make your own interface if you want to restrict things.

closed 2023-11-01 by ssieb