esphome/issues #5059: Waveshare 7.5inV2 is it supported in the most recent release

congestion247, 2023-10-31:
### The problem

I want to build an E-Paper Dashboard for Home-Assistant.

All of the Hardware is working and tested in other combinations.
I am able to flash the code to the ESP32 but whatever code parameters I use, I could not get anything displayed on the waveshare display

Hardware:   
- Waveshare 7.5 Inch E-Paper Display Hat Module V2 
- Waveshare e-Paper Driver Hat Version 2.**3**
- AZ-Delivery ESP32 Dev KitC V

Driver is set to Disbplay Mode B and Interface conifg 1

I tested:
different board-definitons (nodemcu-32s, az-delivery-devkit-v4, esp32dev), 
different display model configurations (7.50in-bV2, 7.50inV2, 7.50inV2alt) 
multiple GPIO combinations

The board and the driver are working because I can display something on it when it is connected to an Arduino.

I found some  blockposts with similar setups with this hardware.
The only difference is mostly that I have an e-Paper Driver hardware version 2.**3** .
All of the others of an hardware version 2.**2**

The only physical difference in this version is a new PWR Pin.
Based on the recommendations auf the vendor I connected it to 3.3V so this should not be a dealbreaker.

https://www.waveshare.com/wiki/E-Paper_Driver_HAT

I am very thankful for any help.

### Which version of ESPHome has the issue?

2023.10.4

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

AZ-Delivery ESP32 Dev KitC V2

### Component causing the issue

Waveshare 7.5 Inch E-Paper Display Hat Module V2

### Example YAML snippet

```yaml
esphome:
  name: display-7-5
  friendly_name: Display_7_5

esp32:
  board: esp32dev 
  framework:
    type: arduino

# Enable logging
logger:

font:
  - file: 'fonts/ConsolaMono-Bold.ttf'
    id: consolaMono_bold_15
[... 211 lines cut ...]
[22:32:29][C][spi:070]:   SDO Pin: GPIO14
[22:32:29][C][spi:072]:   Using HW SPI: SPI
[22:32:29][C][waveshare_epaper:1794]: Waveshare E-Paper
[22:32:29][C][waveshare_epaper:1794]:   Rotations: 0 °
[22:32:29][C][waveshare_epaper:1794]:   Dimensions: 800px x 480px
[22:32:29][C][waveshare_epaper:1795]:   Model: 7.5inV2
[22:32:29][C][waveshare_epaper:1796]:   Reset Pin: GPIO26
[22:32:29][C][waveshare_epaper:1797]:   DC Pin: GPIO27
[22:32:29][C][waveshare_epaper:1798]:   Busy Pin: GPIO25
[22:32:29][C][waveshare_epaper:1799]:   Update Interval: 30.0s
[22:32:54][I][waveshare_epaper:1628]: Power on the display and hat
[22:32:56][W][component:214]: Component waveshare_epaper.display took a long time for an operation (1.49 s).
[22:32:56][W][component:215]: Components should block for at most 20-30ms.
[22:33:24][I][waveshare_epaper:1628]: Power on the display and hat
[22:33:26][W][component:214]: Component waveshare_epaper.display took a long time for an operation (1.35 s).
[22:33:26][W][component:215]: Components should block for at most 20-30ms.
[22:33:54][I][waveshare_epaper:1628]: Power on the display and hat
[22:33:56][W][component:214]: Component waveshare_epaper.display took a long time for an operation (1.34 s).
[22:33:56][W][component:215]: Components should block for at most 20-30ms.
[22:34:24][I][waveshare_epaper:1628]: Power on the display and hat
[22:34:26][W][component:214]: Component waveshare_epaper.display took a long time for an operation (1.35 s).
[22:34:26][W][component:215]: Components should block for at most 20-30ms.

....
```

fukajan, 2023-11-27:
Same problem here when using this YAML configuration:

```
esphome:
  name: display-floor

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: Secret

ota:
  password: Secret

wifi:
  ssid: Secret
  password: Secret
  domain: Secret

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Screen Floor Fallback Hotspot"
    password: Secret

font:
  - file: 'fonts/PassionOne-Regular.ttf'
    id: font_small_book
    size: 18
    glyphs:
      ['H', 'E', 'L', 'O']

spi:
  clk_pin: GPIO12
  mosi_pin: GPIO14

display:
  - platform: waveshare_epaper
    id: eink_display
    cs_pin: GPIO15
    dc_pin: GPIO27
    busy_pin: GPIO25
    reset_pin: GPIO26
    model: 7.50inV2
    lambda: |-
      it.print(20, 20, id(font_small_book), "HELLO");

``` 

I am able to compile and upload both through Wifi and the serial connection. While the display does not show anything, the log output loops:

```
[14:01:50][I][waveshare_epaper:1628]: Power on the display and hat
[14:01:51][W][component:214]: Component waveshare_epaper.display took a long time for an operation (1.35 s).
[14:01:51][W][component:215]: Components should block for at most 20-30ms.
```

fukajan, 2023-11-27:
This seems to be a duplicate of #4628

briodan, 2024-06-12:
running into issue as well, seems mostly like its a contrast issue or something because the display is very faint. 
using the yaml from 
https://github.com/Madelena/esphome-weatherman-dashboard