esphome/issues #5065: voluptuous.error.Invalid: fatal: Needed a single revision Unable to find current revision in submodule path 'components/esp-adf-libs'

CleverCompiler, 2023-11-01:
### The problem

I'm trying to update my M5stack to 2023.10.5 and Esphome is failing to find the esp-adf-libs submodule.

`INFO ESPHome 2023.10.5
INFO Reading configuration /config/esphome/m5stack-atom-echo-82ad48.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/5230/head
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-adf@v2.5
INFO Updating submodules (components/esp-sr, components/esp-adf-libs) for https://github.com/espressif/esp-adf@v2.5
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 1036, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 454, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 190, in write_cpp
    return write_cpp_file()
  File "/esphome/esphome/__main__.py", line 208, in write_cpp_file
    writer.write_cpp(code_s)
  File "/esphome/esphome/writer.py", line 342, in write_cpp
    copy_src_tree()
  File "/esphome/esphome/writer.py", line 295, in copy_src_tree
    copy_files()
  File "/esphome/esphome/components/esp32/__init__.py", line 593, in copy_files
    repo_dir, _ = git.clone_or_update(
  File "/esphome/esphome/git.py", line 114, in clone_or_update
    run_git_command(
  File "/esphome/esphome/git.py", line 32, in run_git_command
    raise cv.Invalid(err_str)
voluptuous.error.Invalid: fatal: Needed a single revision
Unable to find current revision in submodule path 'components/esp-adf-libs'`

### Which version of ESPHome has the issue?

2023.10.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.11.0

### What platform are you using?

ESP8266

### Board

M5Stack

### Component causing the issue

esp-adf-libs

### Example YAML snippet

```yaml
substitutions:
  name: atom-echo2
  friendly_name: Atom Echo 2
packages:
  m5stack.atom-echo-voice-assistant: github://esphome/firmware/voice-assistant/m5stack-atom-echo.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: <Key Omitted>

wifi:
  ssid: !secret wifi_lan_ssid
  password: !secret wifi_lan_password
```

### Anything in the logs that might be useful for us?

```txt
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:230
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 2:54:54 PM (1 occurrences)
Last logged: 2:54:54 PM

[547331760192] Error compiling m5stack-atom-echo-82ad48.yaml; Try again in ESPHome dashboard for more information.
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 157, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 432, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/esphome/update.py", line 174, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error compiling m5stack-atom-echo-82ad48.yaml; Try again in ESPHome dashboard for more information.
```

### Additional information

It work earlier today on my other M5 atom-echo1.  Now I can't install to either of them.  I've tried to install wireless and over USB and they both fail which makes sense since it's a compile time error.

genebean, 2023-11-28:
This is likely the same as https://github.com/esphome/issues/issues/4952

jesserockz, 2023-11-28:
Duplicate of #4952

closed 2023-11-28 by jesserockz