esphome/issues #5066: Error "Checksum error in received SML data." since update to 2023.10.5

myxor, 2023-11-02:
### The problem

Since update to ESPHome 2023.10.5 SML can not read values from Holley DTZ541 energy meter anymore.

> Checksum error in received SML data.

It was working with 2023.10.4 - see logs below.

### Which version of ESPHome has the issue?

2023.10.5

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2023.10.1

### What platform are you using?

ESP32

### Component causing the issue

SML

### Example YAML snippet

```yaml
uart:
  id: uart_bus
  rx_pin: GPIO3
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1

sml:
  id: mysml
  uart_id: uart_bus

sensor:
  - platform: sml
    name: "Stromzähler Bezug WP + Haus (1.8.0)"
    sml_id: mysml
    server_id: "0a01484c5903000596cd"
    obis_code: "1-0:1.8.0"
    unit_of_measurement: kWh
    accuracy_decimals: 1
    device_class: energy
    state_class: total_increasing
    filters:
      - multiply: 0.0001
```

### Anything in the logs that might be useful for us?

```txt
INFO ESPHome 2023.10.4
INFO Reading configuration /config/esphome/esphome-web-e6654c.yaml...
INFO Starting log output from esphome-web-e6654c.local using esphome API
INFO Successfully connected to esphome-web-e6654c.local
[21:12:33][I][app:102]: ESPHome version 2023.10.4 compiled on Oct 31 2023, 20:33:19
[21:12:33][C][wifi:546]: WiFi:
...
[21:12:33][C][sml:092]: SML:
[21:12:33][C][sml_sensor:033]: SML 'Stromzähler Bezug WP + Haus (1.8.0)'
[21:12:33][C][sml_sensor:033]:   Device Class: 'energy'
[21:12:33][C][sml_sensor:033]:   State Class: 'total_increasing'
[21:12:33][C][sml_sensor:033]:   Unit of Measurement: 'kWh'
[21:12:33][C][sml_sensor:033]:   Accuracy Decimals: 1
[21:12:33][C][sml_sensor:035]:   Server ID: 0a01484c5903000596cd
[21:12:33][C][sml_sensor:037]:   OBIS Code: 1-0:1.8.0

INFO ESPHome 2023.10.5
INFO Reading configuration /config/esphome/esphome-web-e6654c.yaml...
INFO Starting log output from esphome-web-e6654c.local using esphome API
INFO Successfully connected to esphome-web-e6654c.local
[08:30:10][W][sml:116]: Checksum error in received SML data.
[08:30:12][W][sml:116]: Checksum error in received SML data.
[08:30:13][W][sml:116]: Checksum error in received SML data.
[08:30:14][W][sml:116]: Checksum error in received SML data.
```

AnanasAPod, 2023-11-02:
SML no longer works for me either. I also have a Holley DTZ541. 
The strange thing for me is that since today at 5:00 a.m. I only get “Checksum error in received SML data”. 
I had version 2023.8.3 on it. I've updated now, unfortunately no solution. 
I checked the data bus with the osiloscope. 
There will definitely be data. So I can rule out a hardware error.

Did it happen after an update? Or just like that.

myxor, 2023-11-02:
Sometimes it seems to work and sometimes not. I have not yet been able to recognize a clear pattern.

AnanasAPod, 2023-11-02:
Maybe the Holley is the problem. I found known problems about the Holley in a German description. That the transmitting LED no longer works after a while or only works sporadically. That could be suitable for my case.

https://shop.weidmann-elektronik.de/media/files_public/67303f78bbcd558d664d0095291562d3/BekannteProblememitZhlermodellen.pdf

[image]

closed 2024-03-28 by github-actions[bot]